Koenkk/zigbee2mqtt #13559: Gledopto GL-C-007 2ID respond wrong color in Webinterface

cradermacher, 2022-08-16:
### What happened?

The Device show the wrong color in the webinterface, after changing the color
[image]

If i user one of the predifined color Buttons or use the colorwidget the device changes the color correkt, but the colorwidget show everytime "orange".

The "color_rgb" is eveytime the same. Only "color" changes the values.
{
    "brightness": 140,
    "brightness_rgb": 113,
    "color": {
        "hue": 118,
        "saturation": 95,
        "x": 0.6307692307692307,
        "y": 0.3230769230769231
    },
    "color_mode": "xy",
    "color_mode_rgb": "color_temp",
    "color_rgb": {
        "x": 0.4599,
        "y": 0.4106
    },
    "color_temp": 319,
    "color_temp_rgb": 370,
    "linkquality": 48,
    "state": "ON",
    "state_rgb": "ON",
    "device": {
        "applicationVersion": 1,
        "dateCode": "20180920",
        "friendlyName": "2_4_Schlafzimmer_RGB",
        "hardwareVersion": 1,
        "ieeeAddr": "0x00124b00232cf9f4",
        "manufacturerID": 0,
        "manufacturerName": "GLEDOPTO",
        "model": "GL-C-007-2ID",
        "networkAddress": 18565,
        "powerSource": "Mains (single phase)",
        "softwareBuildID": "2.0.5",
        "stackVersion": 2,
        "type": "Router",
        "zclVersion": 1
    }
}

### What did you expect to happen?

I expect, that the color Widget show the correct color, an the the value "color_rgb" has the correct value.

### Zigbee2MQTT version

 1.26.0 commit: bc4ffc0

### Adapter firmware version

20210320

### Adapter

ZigStar Stick v4

Koenkk, 2022-08-16:
@nurikk if I understand correctly, the color picker value is based on the `color_xy` value, right?

cradermacher, 2022-08-16:
@Koenkk 
this changes if i change the color:
"color": {
        "hue": 118,
        "saturation": 95,
        "x": 0.14912280701754385,
        "y": 0.06140350877192982
    },

this is constant:
 "color_rgb": {
        "x": 0.4599,
        "y": 0.4106
    },

Regards
Christian

Burki24, 2022-08-17:
I think it`s a general problem with the colorpicker:
When I choose a color in the colorpicker, sometimes it takes a wrong color:
[image]
And also, there are differences between color_xy and color_hs when I choose es example a color with the colorpicker from color_hs:
[image]

Burki24, 2022-09-07:
@Koenkk , @nurikk 

Are there any news to this problem?

Thanks in advance.

Burkhard

closed 2022-10-16 by github-actions[bot]