Koenkk/zigbee2mqtt #13560: TuYa TV02-Zigbee _TZE200_hue3yfsn - Calibration shows wrong precision in

rotilho, 2022-08-16:
### What happened?

According to herdsman-converters the calibration precision should be 0.1 however the propagate precision to HA is 1

### What did you expect to happen?

I expected to Z2M propagate correct precision.

### How to reproduce it (minimal and precise)

Just pair one of the TuYa TV02 and check precision show in HA

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

rotilho, 2022-08-19:
@vladi1234 @serios @amaciuc tagging you guys because you probably have the TV02 too. Do you guys experience the same issue?

Serios, 2022-08-19:
@rotilho Yes I can confirm it. Thou didn't noticed until now when you mentioned it, since I'm using automation to set calibration temp value and not the slider in HA. But yes the step in Z2M is 0.1, but in HA is 1

rotilho, 2022-08-19:
I'm also setting it through an automation but I'm reading the step to know how much I can add/subtract. I ended up adding the option to override the step but it would be great have it properly configured.

amaciuc, 2022-08-19:
> @vladi1234 @Serios @amaciuc tagging you guys because you probably have the TV02 too. Do you guys experience the same issue?

I am using thermostats only as valve (on/off). I am harvesting temperature from other zigbee sensors and I have my own thermostat component created in hassio: input from sensors and output to this valve. So I cannot gave you a feedback related to the calibration.

closed 2022-09-26 by github-actions[bot]