Koenkk/zigbee2mqtt #13563: MAC channel access failure' (225)

draconight, 2022-08-16:
### What happened?

Hi,
I installed Home Assistant on a PI4 with Z2M some months ago, and quickly got it up and running.
A week later, all devices stopped responding, and I got the "MAC channel access failure' (225)" error.

I found that it was because of channel interference, so:
- I forced my Unifi APs to use channel 1 and 6 respectively
- Set the Z2M Channel to 25 in the configurations YAML
- Repaired all my devices

This has worked for some months, but after updating Z2M and HA a week ago, I now get the error again.

I have rechecked that the Z2M channel is indeed 25, and rechecked all my AP's as well, which are still set to 1 and 6.
I use an adapter cord to my Sonoff Zigbee 3.0 USB Dongle Plus.

I have tried restarting both the PI, HA, the Dongle, Z2M and the MQTT Broker, without success.
I have also updated the Sonoff Zigbee 3.0 USB Dongle Plus do the latest firmware version, with no success.

### Zigbee2MQTT version

1.27.0-dev commit: 36a4754

### Adapter firmware version

20220219

### Adapter

ZBDongle-P CC2652P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9350173/log.txt)

draconight, 2022-09-04:
Antenna

closed 2022-09-04 by draconight

pastukhov, 2022-10-26:
@draconight How do you fix it?