Koenkk/zigbee2mqtt #13566: Legrand 600086: no synchronization of the position when managing the shutter manually

Alindil, 2022-08-16:
### What happened?

The position does not synchronise when closing or opening the shutters manually. But wehn
[image]
But if I click on the position sync. The value is well synchronised

### What did you expect to happen?

I think that the synchronization should be done automatically after a given time

### How to reproduce it (minimal and precise)

Device: Legrand dooxie 600086 but when I associate, it is recognised as 067776
[image]

You just have to close and open the flaps manually to see that the position is not updated
[image]

### Zigbee2MQTT version

1.27.0 commit: a9b8808

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB

Koenkk, 2022-08-16:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [' Shutter switch with neutral\u0000\u0000\u0000'],
    model: '067776',
    vendor: 'Legrand',
    description: 'Netatmo wired shutter switch',
    // the physical LED will be green when permit join is true, off otherwise and red when not linked
    fromZigbee: [
        // Devices can send an identify message when the configuration button is pressed
        // (behind the physical buttons)
        // Used on the official gateway to send to every devices an identify command (green)
        fz.identify, fz.ignore_basic_report,
        // support binary report on moving state (supposed)
        fz.legrand_binary_input_moving, fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.legrand_identify, tz.legrand_settingAlwaysEnableLed],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Alindil, 2022-08-17:
Hello,
the correction works perfectly. 
I would like to take advantage of this issue to suggest another point. 
When you click on the close, open and stop buttons in zigbee2mqtt, the position is not updated either. It might be better if open = 100 (if not inverted), close = 0 (if not inverted) and stop = 50
[image]

Alindil, 2022-08-18:
@Koenkk 
I tested under docker on the image koenkk/zigbee2mqtt:latest-dev and it does not work. 
Initial status with the shutter open:
[image]

When I click on the closed button:
[image]

When I click on the synchronization button:
[image]

Koenkk, 2022-08-18:
Does the `state` updated when you change the shutter position manually? Position 0 should show `CLOSE`, anything else `OPEN`.

Alindil, 2022-08-18:
Yes, manually it works. And when I ask for the action to close for mqtt (not using the position 0), the shutter closes well but the status is still open and the position is at 100And when I ask for the action to close (not using the action position= 0), the shutter closes well but the status is still open and the position is 100 (or 50 if it was between open)

M-Gregoire, 2022-09-19:
I believe this issue is not stale as the problem described is still existing.
I'd be happy to help in the resolution if I can.

closed 2022-10-28 by github-actions[bot]

novisys, 2022-11-15:
same issue here !
Can reopen this issue or we need to open new one ?
Thanks