Koenkk/zigbee2mqtt #13570: Brightness ist set to 1 when turning light on again

gucki, 2022-08-16:
### What happened?

A few months ago, when I turned off a light and back on again the brightness was the same as before. Now when I turn off a light and back on again the brightness is always set to 1.

### What did you expect to happen?

When turning a light back on again, the brightness should be set to the previous value, before the light was turned off.

I debugged the mqtt messages sent and it seems zigbee2mqtt is (surprisingly) sending two mqtt messages when a light is turned on. The first message still contains the previous brightness and state off, the second brightness 1 and state on. So this really looks like a bug too me.

Turning light off:
Zigbee2MQTT:info  2022-08-16 18:49:46: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":254,"color_mode":"color_temp","color_temp":253,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'

Turning light back on again:
Zigbee2MQTT:info  2022-08-16 18:49:59: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":254,"color_mode":"color_temp","color_temp":253,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-08-16 18:49:59: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":253,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'

### How to reproduce it (minimal and precise)

Turn off/on a light using the zigbee2mqtt UI.

### Zigbee2MQTT version

1.27.0 commit: a9b8808

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

Koenkk, 2022-08-17:
Can you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

gucki, 2022-08-17:
@Koenkk Sure, here it is:

Turning light off:

    2022-08-17T16:54:23.199Z zigbee-herdsman:controller:endpoint Command 0xf0d1b8000011becd/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
    2022-08-17T16:54:23.201Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xc722 EP:1 SeqNr. 38 request id: 8
    2022-08-17T16:54:23.211Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 8 status: 0
    2022-08-17T16:54:23.212Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T16:54:23.212Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
    2022-08-17T16:54:23.212Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T16:54:23.217Z zigbee-herdsman:deconz:driver query aps data confirm
    2022-08-17T16:54:23.221Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 39
    2022-08-17T16:54:23.224Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xc722 request id: 8 confirm status: 0
    2022-08-17T16:54:23.224Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T16:54:23.224Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 5
    2022-08-17T16:54:23.224Z zigbee-herdsman:deconz:adapter false, false, false
    2022-08-17T16:54:23.224Z zigbee-herdsman:deconz:adapter resolve request
    2022-08-17T16:54:23.259Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
    2022-08-17T16:54:23.259Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T16:54:23.262Z zigbee-herdsman:deconz:driver query aps data indication
    2022-08-17T16:54:23.267Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 40
    2022-08-17T16:54:23.270Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 40 srcAddr: 0xc722 destAddr: 0x0 profile id: 0x104 cluster id: 0x8 lqi: 255
    2022-08-17T16:54:23.270Z zigbee-herdsman:deconz:frameParser response payload: 8,5,11,4,0
    2022-08-17T16:54:23.270Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 5
    2022-08-17T16:54:23.273Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":4,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":50978,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
    2022-08-17T16:54:23.274Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T16:54:23.274Z zigbee-herdsman:deconz:adapter response received
    Zigbee2MQTT:info  2022-08-17 16:54:23: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":169,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'

Turning light on again:

    2022-08-17T16:54:27.692Z zigbee-herdsman:controller:endpoint Read 0xf0d1b8000011becd/1 genLevelCtrl(["onLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
    2022-08-17T16:54:27.694Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xc722 EP:1 SeqNr. 41 request id: 9
    2022-08-17T16:54:27.706Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 9 status: 0
    2022-08-17T16:54:27.706Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T16:54:27.707Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
    2022-08-17T16:54:27.707Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T16:54:27.710Z zigbee-herdsman:deconz:driver query aps data confirm
    2022-08-17T16:54:27.716Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 42
    2022-08-17T16:54:27.718Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xc722 request id: 9 confirm status: 0
[... cut ...]
    Zigbee2MQTT:info  2022-08-17 16:54:27: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":248,"state":"ON","update":{"state":"idle"},"update_available":false}'

Koenkk, 2022-08-17:
I also need the received MQTT commands, can you provide the herdsman debug log of the following:
- Start zigbee2mqtt
- Turn off
- Turn on (light turns on @ brightness 1)
- provide log

gucki, 2022-08-17:
I have many connected devices, so the full log would be quite big. As I use the zigbee2mqtt UI for toggling the light I don't think there will be any received MQTT commands, or is it also using MQTT internally?

gucki, 2022-08-17:
I turned on debug log-level and now I see the received MQTT commands as well. Here are the relevant log entries:

Initial ON:

    Zigbee2MQTT:debug 2022-08-17 20:23:32: Received MQTT message on 'zigbee2mqtt/light1/set' with data '{"brightness":131}'
    Zigbee2MQTT:debug 2022-08-17 20:23:32: Publishing 'set' 'brightness' to 'light1'
    2022-08-17T20:23:32.263Z zigbee-herdsman:controller:endpoint Command 0xf0d1b80000117e2d/1 genLevelCtrl.moveToLevelWithOnOff({"level":131,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
    2022-08-17T20:23:32.264Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xb587 EP:1 SeqNr. 244 request id: 64
    2022-08-17T20:23:32.274Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 64 status: 0
    2022-08-17T20:23:32.274Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T20:23:32.274Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
    2022-08-17T20:23:32.274Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T20:23:32.276Z zigbee-herdsman:deconz:driver query aps data confirm
    2022-08-17T20:23:32.281Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 245
    2022-08-17T20:23:32.283Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xb587 request id: 64 confirm status: 0
    2022-08-17T20:23:32.283Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T20:23:32.283Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 40
    2022-08-17T20:23:32.283Z zigbee-herdsman:deconz:adapter false, false, false
    2022-08-17T20:23:32.283Z zigbee-herdsman:deconz:adapter resolve request
    2022-08-17T20:23:32.300Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
    2022-08-17T20:23:32.300Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T20:23:32.303Z zigbee-herdsman:deconz:driver query aps data indication
    2022-08-17T20:23:32.308Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 246
    2022-08-17T20:23:32.310Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 246 srcAddr: 0xb587 destAddr: 0x0 profile id: 0x104 cluster id: 0x8 lqi: 255
    2022-08-17T20:23:32.310Z zigbee-herdsman:deconz:frameParser response payload: 8,40,11,4,0
    2022-08-17T20:23:32.311Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 40
    2022-08-17T20:23:32.312Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":40,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":4,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":46471,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
    2022-08-17T20:23:32.313Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-17T20:23:32.314Z zigbee-herdsman:deconz:adapter response received
    Zigbee2MQTT:info  2022-08-17 20:23:32: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":131,"color_mode":"color_temp","color_temp":153,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b80000117e2d","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":46471,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'

Setting to OFF:

    Zigbee2MQTT:debug 2022-08-17 20:23:45: Received MQTT message on 'zigbee2mqtt/light1/set' with data '{"state":"OFF"}'
    Zigbee2MQTT:debug 2022-08-17 20:23:45: Publishing 'set' 'state' to 'light1'
    2022-08-17T20:23:45.979Z zigbee-herdsman:controller:endpoint Command 0xf0d1b80000117e2d/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
    2022-08-17T20:23:45.981Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xb587 EP:1 SeqNr. 248 request id: 65
    2022-08-17T20:23:45.991Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 65 status: 0
    2022-08-17T20:23:45.991Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...] 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b80000117e2d","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":46471,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'

Please let me know if you need anything else. Thank you.

Koenkk, 2022-08-18:
@pyos I think this is caused by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4211 . 

The `onLevel` read [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/b0692791cad6c8810d0dc16484f3f6058a4174b7/converters/toZigbee.js#L895) return 1 causing the brightness to be set to 1 instead of 131. Why is this logic needed?

```
Zigbee2MQTT:debug 2022-08-17 20:24:47: Received Zigbee message from 'light1', type 'readResponse', cluster 'genLevelCtrl', data '{"onLevel":1}' from endpoint 1 with groupID null
```

pyos, 2022-08-18:
Because that's what onLevel does. It sets the brightness to which an "on" command should turn the light on.

gucki, 2022-08-22:
Hello @pyos,

thank you for your input. I'm not sure if your reply was party for me, but nothing happens if I try:

    mosquitto_pub -t 'zigbee2mqtt/light1' -m '{"level_config": {"on_level": "previous"}}'

If I add "/set" to the topic I see the brightness changing to the previous value in the UI, but the light stays off:

    mosquitto_pub -t 'zigbee2mqtt/light1/set' -m '{"level_config": {"on_level": "previous"}}'

    Zigbee2MQTT:debug 2022-08-22 08:47:36: Received MQTT message on 'zigbee2mqtt/light1/set' with data '{"level_config": {"on_level": "previous"}}'
    Zigbee2MQTT:debug 2022-08-22 08:47:36: Publishing 'set' 'level_config' to 'light1'
    2022-08-22T08:47:36.767Z zigbee-herdsman:controller:endpoint Write 0xf0d1b80000117e2d/1 genLevelCtrl({"onLevel":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
    2022-08-22T08:47:36.769Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xb587 EP:1 SeqNr. 214 request id: 63
    2022-08-22T08:47:36.782Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 63 status: 0
    2022-08-22T08:47:36.782Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-22T08:47:36.782Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
    2022-08-22T08:47:36.782Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-22T08:47:36.783Z zigbee-herdsman:deconz:driver query aps data confirm
    2022-08-22T08:47:36.785Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 215
    2022-08-22T08:47:36.786Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xb587 request id: 63 confirm status: 0
    2022-08-22T08:47:36.786Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-22T08:47:36.787Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 61
    2022-08-22T08:47:36.787Z zigbee-herdsman:deconz:adapter true, true, false
    2022-08-22T08:47:36.787Z zigbee-herdsman:deconz:adapter resolve request
    2022-08-22T08:47:36.801Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
    2022-08-22T08:47:36.801Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-22T08:47:36.804Z zigbee-herdsman:deconz:driver query aps data indication
    2022-08-22T08:47:36.804Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 216
    2022-08-22T08:47:36.806Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 216 srcAddr: 0xb587 destAddr: 0x0 profile id: 0x104 cluster id: 0x8 lqi: 255
    2022-08-22T08:47:36.807Z zigbee-herdsman:deconz:frameParser response payload: 24,61,4,0
    2022-08-22T08:47:36.807Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 61
    2022-08-22T08:47:36.808Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":61,"manufacturerCode":null,"commandIdentifier":4},"Payload":[{"status":0}],"Command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}},"address":46471,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
    2022-08-22T08:47:36.809Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-08-22T08:47:36.810Z zigbee-herdsman:deconz:adapter response received
    Zigbee2MQTT:info  2022-08-22 08:47:36: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":206,"color_mode":"color_temp","color_temp":153,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b80000117e2d","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":46471,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":"previous"},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'

If I change the payload to include the state it turns on, but the brightness is again set to 1:

    mosquitto_pub -t 'zigbee2mqtt/light1/set' -m '{"state": "ON", "level_config": {"on_level": "previous"}}'

    Zigbee2MQTT:debug 2022-08-22 09:02:04: Received MQTT message on 'zigbee2mqtt/light1/set' with data '{"state": "ON", "level_config": {"on_level": "previous"}}'
    Zigbee2MQTT:debug 2022-08-22 09:02:04: Publishing 'set' 'state' to 'light1'
    2022-08-22T09:02:04.115Z zigbee-herdsman:controller:endpoint Read 0xf0d1b80000117e2d/1 genLevelCtrl(["onLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionS [... cut ...]
    Zigbee2MQTT:info  2022-08-22 09:00:51: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":166,"color_mode":"color_temp","color_temp":153,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b80000117e2d","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":46471,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":"previous"},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'

Hope it helps it getting this fixed :)

pyos, 2022-08-22:
Ok I guess I didn't explain clearly.

This sentence:

>When turning a light back on again, the brightness should be set to the previous value, before the light was turned off.

is no correct. The light has a configuration parameter that tells it which brightness to use when turned on. If it's set to "previous" (or 255), then it'll be the brightness when it was turned off; otherwise it'll be some other value. You can change that configuration parameter by sending `{"level_config": {"on_level": new value}}`; that *only* affects the configuration but does not immediately turn the light on.

So the behavior you're seeing is 100% correct - when on_level is set to 1, `{"state": "ON"}` sets brightness to 1 regardless of what it was before, and when on_level is set to "previous", `{"state": "ON"}` restores the old value.

What I don't understand though are the timestamps in your logs. At 08:58:51, you set on_level to "previous", then at 09:00:51 that seems to have worked fine - but at 09:02:04, on_level reverted back to 1 by itself?

gucki, 2022-08-22:
@pyos Yes, it's quite strange. I did some further tests and it seems setting the state to on always somehow generates another message (at least according to the zigbee2mqtt logs). Do you have any idea where this is coming from/ why it is generated?

    mosquitto_pub -t 'zigbee2mqtt/light1/set' -m '{"state": "ON"}'

    Zigbee2MQTT:info  2022-08-22 12:47:56: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
    Zigbee2MQTT:info  2022-08-22 12:47:56: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'

It seems setting the state to off resets the level_config "somewhere". I write "somewhere" because the correct on_level is still shown as "previous" in any subsequent logs. For example, if I publish turn off three times, the last two messages still show the on_level as previous.  But as soon as I set the state to on, the on_level is set/ shown as 1 in the weird additional message generated and also the brightness is that to that level:

    mosquitto_pub -t 'zigbee2mqtt/light1/set' -m '{"state": "OFF"}'
    mosquitto_pub -t 'zigbee2mqtt/light1/set' -m '{"state": "ON", "brightness":157}'
    mosquitto_pub -t 'zigbee2mqtt/light1/set' -m '{"state": "OFF"}'
    mosquitto_pub -t 'zigbee2mqtt/light1/set' -m '{"state": "ON"}'

    Zigbee2MQTT:info  2022-08-22 12:40:04: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":157,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":"previous"},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'
    Zigbee2MQTT:info  2022-08-22 12:40:07: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":157,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":"previous"},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
    Zigbee2MQTT:info  2022-08-22 12:40:10: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":157,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":"previous"},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'
    Zigbee2MQTT:info  2022-08-22 12:40:13: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":157,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'
    Zigbee2MQTT:info  2022-08-22 12:40:13: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_confi [... cut ...] u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":"previous"},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'

If you like we could also schedule a quick screensharing session...

pyos, 2022-08-22:
The extra publish when you send `{"state": "ON"}` is because zigbee2mqtt re-reads on_level from the light and updates its own state from that, and only then tries to turn the light on.

>It seems setting the state to off resets the level_config "somewhere". I write "somewhere" because the correct on_level is still shown as "previous" in any subsequent logs.

My guess is that the light does that by itself; zigbee2mqtt will not notice the change until it actually tries to read the value again. I think you should be able to force that by sending `{"level_config": null}` to `zigbee2mqtt/light/get` - if you do that after sending `{"state": "OFF"}` and zigbee2mqtt publishes a new state with `"on_level": 1` in it, then the light itself is buggy and unless there's an OTA update available, its on_level simply cannot be used.

gucki, 2022-08-22:
Ok, that makes sense. And yes, it seems the light resets the configuration when the state is set to off:

    mosquitto_pub -t 'zigbee2mqtt/light1/get' -m '{"level_config": null}'
    Zigbee2MQTT:info  2022-08-22 13:44:35: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"execute_if_off":false,"on_level":"previous"},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
    Zigbee2MQTT:info  2022-08-22 13:44:35: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"execute_if_off":false,"on_level":"previous"},"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'

    mosquitto_pub -t 'zigbee2mqtt/light1/set' -m '{"state": "OFF"}'
    Zigbee2MQTT:info  2022-08-22 13:45:04: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"execute_if_off":false,"on_level":"previous"},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'

    mosquitto_pub -t 'zigbee2mqtt/light1/get' -m '{"level_config": null}'
    Zigbee2MQTT:info  2022-08-22 13:45:09: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"execute_if_off":false,"on_level":1},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'
    Zigbee2MQTT:info  2022-08-22 13:45:09: MQTT publish: topic 'zigbee2mqtt/light1', payload '{"brightness":1,"color_mode":"color_temp","color_temp":303,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"light1","hardwareVersion":1,"ieeeAddr":"0xf0d1b8000011becd","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":50978,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"execute_if_off":false,"on_level":1},"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'

Imo it's strange that the get command also generates two messages. 

Unfortunately, no OTA is available (checked https://update.ledvance.com/firmware-overview). So I assume a workaround for this device is needed? Why is it needed to re-read the values to turn the light on?

pyos, 2022-08-22:
>Imo it's strange that the get command also generates two messages.

That's because level_config has two suboptions (on_level and execute_if_off), and each triggers a separate update.

>So I assume a workaround for this device is needed? Why is it needed to re-read the values to turn the light on?

Yes, one option is to just not re-read on_level from the device if it's already present in the state and assume that it shouldn't change otherwise. For now you should be able to fix this with an extension: https://gist.github.com/pyos/2e2a768dd1eba25bb8bc8396368463ca - add that, send `{"level_config": {"on_level": "previous"}}`, and then it should work.

gucki, 2022-08-22:
Thank you very much @pyos, your patch works perfectly. Is there an easy way to set the on_level for all devices? Perhaps using a startup script? Also, why isn't the on_level property exposed in the UI (or am I just missing it?)?

Wireheadbe, 2022-08-27:
> Because that's what onLevel does. As described in ZCLv7 chapter 3.10.2.1.1, it sets the brightness to which an "on" command should turn the light on. If you want to keep the old brightness, it should be 255, not 1. Try publishing `{"level_config": {"on_level": "previous"}}` to `zigbee2mqtt/light1/set` (note how the current published state already has `"level_config": {"on_level": 1}` in it). If that doesn't help, the device's firmware is buggy and needs workarounds.

This actually helped me for adaptive lighting, to not get blinded at turn-on. Irrelevant to the op's issue, but thanks nonetheless.

dhanukapulkit, 2022-08-29:
Hi @gucki, Thanks for your concern. Here, as I see that you have already received the support from the community as well.
Regarding the Lamp itself, there is no firmware update that is planned for this product, as of now.
We did some tests with the Conbee Stick and our partner gateways and here we found that the "previous" state was always "remembered" after the lamp was turned on, after removing the power supply. 
Have a nice day!
Best regards,
Your Ledvance Team

gucki, 2022-08-30:
@dhanukapulkit So you say it's a bug in zigbee2mqtt?

dhanukapulkit, 2022-08-30:
Hi @gucki , when we are testing the lamps with our partner gateways than the "previous" function is working.
But since we didn't have any test results regarding its behaviour with MQTT, that's why, we cannot be sure whether it is due to MQTT or some other possible reason.

kilimnik, 2022-09-02:
I had the same bug with zigbee2mqtt. The interesting part was, that on first installation it worked fine. Only after I changed the transition to 0, it stopped working properly.

gucki, 2022-10-04:
Is there anything I can do to get this issue resolved? Will your patch, @pyos, be merged?

closed 2022-11-12 by github-actions[bot]

gucki, 2022-11-12:
@Koenkk Can you please reopen it because it's not really fixed? How to push this forward? Thank you.

Koenkk, 2022-11-13:
The patch from @pyos got merged (https://github.com/Koenkk/zigbee-herdsman-converters/pull/4596), but if I understand correctly this did not fix the issue?  (z2m 1.28.2)

pyos, 2022-11-13:
You reverted it, then I forgot to fix the thing you mentioned in the comment.

Koenkk, 2022-11-13:
I see, any chance you could take a look at it?

Koenkk, 2022-11-15:
Fix from @pyos is merged now, @gucki can you check again?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gucki, 2022-11-16:
Hello @Koenkk , I just tried (using docker latest-dev tag, 1.28.2-dev (commit #50cb421)), but unfortunately, it doesn't seem to work. After turning the light on again, the brightness is not reset to the previous value. Here's the log output:

    2022-11-16T09:25:21.281Z zigbee-herdsman:controller:endpoint Command 0xf0d1b80000117961/1 genLevelCtrl.moveToLevelWithOnOff({"level":231,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
    2022-11-16T09:25:21.282Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfc05 EP:1 SeqNr. 128 request id: 36
    2022-11-16T09:25:21.293Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 36 status: 0
    2022-11-16T09:25:21.293Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-11-16T09:25:21.293Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
    2022-11-16T09:25:21.293Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-11-16T09:25:21.295Z zigbee-herdsman:deconz:driver query aps data confirm
    2022-11-16T09:25:21.300Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 129
    2022-11-16T09:25:21.302Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfc05 request id: 36 confirm status: 0
    2022-11-16T09:25:21.302Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-11-16T09:25:21.302Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 32
    2022-11-16T09:25:21.302Z zigbee-herdsman:deconz:adapter false, false, false
    2022-11-16T09:25:21.303Z zigbee-herdsman:deconz:adapter resolve request
    2022-11-16T09:25:21.340Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
    2022-11-16T09:25:21.340Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
    2022-11-16T09:25:21.341Z zigbee-herdsman:deconz:driver query aps data indication
    2022-11-16T09:25:21.347Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 130
    2022-11-16T09:25:21.350Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 130 srcAddr: 0xfc05 destAddr: 0x0 profile id: 0x104 cluster id: 0x8 lqi: 140
    2022-11-16T09:25:21.350Z zigbee-herdsman:deconz:frameParser response payload: 8,32,11,4,0
    2022-11-16T09:25:21.351Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 32
    2022-11-16T09:25:21.352Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":32,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":4,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":64517,"endpoint":1,"linkquality":140,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
    2022-11-16T09:25:21.353Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-11-16T09:25:21.354Z zigbee-herdsman:deconz:adapter response received
    Zigbee2MQTT:info  2022-11-16 09:25:21: MQTT publish: topic 'zigbee2mqtt/test', payload '{"brightness":231,"color_mode":"color_temp","color_temp":323,"device":{"applicationVersion":1,"dateCode":"Sep 29 2021\u0000\u0000\u0000\u0000\u0000","friendlyName":"test","hardwareVersion":1,"ieeeAddr":"0xf0d1b80000117961","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075148338","networkAddress":64517,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":140,"power_on_behavior":null,"state":"ON","update":{"installed_version":"01056400","state":"idle"},"update_available":false}'
    2022-11-16T09:25:23.185Z zigbee-herdsman:controller:endpoint Command 0xf0d1b80000117961/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
    2022-11-16T09:25:23.186Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfc05 EP:1 SeqNr. 131 request id: 37
    2022-11-16T09:25:23.196Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 37 status: 0
    2022-11-16T09:25:23.196Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-11-16T09:25:23.196Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
    2022-11-16T09:25:23.196Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
    2022-11-16T09:25:23.200Z zigbee-herdsman:deconz:driver query aps data confirm
[... cut ...] e":"LEDVANCE","model":"4058075148338","networkAddress":64517,"powerSource":"Mains (single phase)","softwareBuildID":"01056400","stackVersion":2,"type":"Router","zclVersion":2},"level_config":{"on_level":1},"linkquality":100,"power_on_behavior":null,"state":"ON","update":{"installed_version":"01056400","state":"idle"},"update_available":false}'

I didn't manually set the brightness as mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/13570#issuecomment-1222552903. Is this still necessary?

Koenkk, 2022-11-17:
@pyos could you check what goes wrong here?

pyos, 2022-11-17:
>Is this still necessary?

Yes, you need to manually change on_level one last time - zigbee2mqtt has no way to determine that the value that's already in the state is wrong

gucki, 2022-11-17:
@pyos Ok, I'll try this tomorrow then. But it means I'll not only have to do this for all existing lights but also all new lights, I might add in the future?

pyos, 2022-11-17:
I don't know, that depends on what settings the bulbs will report when at factory state.

gucki, 2023-01-07:
Sorry, it took a bit longer but I can confirm now that it works as expected after manually setting the brightness as mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/13570#issuecomment-1222552903.