Koenkk/zigbee2mqtt #13574: Third Reality 3RSB015BZ not reporting Battery Percentage

darkxst, 2022-08-17:
### What happened?

I am testing a Third Reality Smart Blind 3RSB015BZ and it is reporting that it is connected to mains power.

The following call is failing in the configure phase:
```await reporting.batteryPercentageRemaining(endpoint);```

I tested with an external converter and removing that line appears to resolve the issue and the battery percentage is reported correctly.

databse.db contents
```{"id":6,"type":"EndDevice","ieeeAddr":"0x282c02bfffea084d","nwkAddr":12324,"manufId":4659,"manufName":"Third Reality, Inc","powerSource":"Mains (single phase)","modelId":"3RSB015BZ","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,1,6,258],"outClusterList":[6,258,25],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":62,"batteryPercentageRemaining":200}},"genBasic":{"attributes":{"modelId":"3RSB015BZ","manufacturerName":"Third Reality, Inc","powerSource":1,"zclVersion":3,"appVersion":52,"stackVersion":2,"hwVersion":2,"swBuildId":"1.00.52"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffefcdbff","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffefcdbff","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":52,"stackVersion":2,"hwVersion":2,"swBuildId":"1.00.52","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1660682932737,"defaultSendRequestWhen":"immediate"}```

### What did you expect to happen?

This is a battery powered device and should report the battery status.

### How to reproduce it (minimal and precise)

On Joining the blind to Zigbee2MQTT there is a configure error.

### Zigbee2MQTT version

1.27.0-dev commit: 36a4754

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

ZBDongle-E

### Debug log

2022-08-16 19:49:10Failed to configure '0x282c02bfffea084d', attempt 1 (Error: ConfigureReporting 0x282c02bfffea084d/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22) at Object.batteryPercentageRemaining (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:61:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/third_reality.js:119:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))

darkxst, 2022-08-17:
This is my external converter code that I am testing with and appears to be working:
```const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;

const definition = {
        zigbeeModel: ['3RSB015BZ'],
        model: '3RSB015BZ-Custom',
        vendor: 'Third Reality',
        description: 'Roller shade',
        fromZigbee: [fz.cover_position_tilt, fz.battery],
        toZigbee: [tz.cover_state, tz.cover_position_tilt],
        meta: {battery: {dontDividePercentage: false}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.currentPositionLiftPercentage(endpoint);
            try {
                await reporting.batteryPercentageRemaining(endpoint);
            } catch (error) {/* Fails for some*/}
        },
        exposes: [e.cover_position(), e.battery()],
}

module.exports = definition;
```

Koenkk, 2022-08-17:
Pushed your fix, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-17 by Koenkk