Koenkk/zigbee2mqtt #13582: No converter available for 'system_mode' ("off") for supported device     Schneider Electric WV704R0A0902

scaiandre, 2022-08-17:
### What happened?

In homeassistant when turning on the radiator thermostat, zigbee2mqtt logs this error:
error 2022-08-17 16:18:00No converter available for 'system_mode' ("heat")

When turning off the radiator thermostat, zigbee2mqtt logs this error:
error 2022-08-17 16:19:00No converter available for 'system_mode' ("off")

I tried both using the "Zigbee2MQTT" addon and the "Zigbee2MQTT Edge" addon.

### What did you expect to happen?

Radiator thermostat turning off and on respectively

### How to reproduce it (minimal and precise)

Add the radiator thermostat and turn it on/off from the frontend,.

### Zigbee2MQTT version

1.27.0-dev commit: 040d2a1

### Adapter firmware version

0x26580700

### Adapter

usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2459507

### Debug log

2022-08-17T14:31:13.435Z mqttjs:client _handlePacket :: emitting packetreceive
2022-08-17T14:31:13.435Z mqttjs:client _handlePublish: packet Packet { cmd: 'publish', retain: false, qos: 0, dup: false, length: 51, topic: 'zigbee2mqtt/0x086bd7fffece4c8d/set/system_mode', payload: <Buffer 6f 66 66> }
2022-08-17T14:31:13.435Z mqttjs:client _handlePublish: qos 0
2022-08-17T14:31:13.458Z mqttjs:client publish :: message `{"level":"error","message":"No converter available for 'system_mode' (\"off\")"}` to topic `zigbee2mqtt/bridge/logging`
2022-08-17T14:31:13.462Z mqttjs:client publish :: qos 0
2022-08-17T14:31:13.463Z mqttjs:client MqttClient:publish: packet cmd: publish
2022-08-17T14:31:13.464Z mqttjs:client _sendPacket :: (mqttjs_f5c2a67f) ::  start
2022-08-17T14:31:13.464Z mqttjs:client sendPacket :: packet: {
  cmd: 'publish',
  topic: 'zigbee2mqtt/bridge/logging',
  payload: `{"level":"error","message":"No converter available for 'system_mode' (\\"off\\")"}`,
  qos: 0,
  retain: false,
  messageId: 0,
  dup: false
}
2022-08-17T14:31:13.465Z mqttjs:client sendPacket :: emitting `packetsend`
2022-08-17T14:31:13.465Z mqttjs:client sendPacket :: writing to stream
2022-08-17T14:31:13.466Z mqttjs:client sendPacket :: writeToStream result true
2022-08-17T14:31:13.467Z mqttjs:client sendPacket :: invoking cb
2022-08-17T14:31:13.467Z mqttjs:client _sendPacket :: (mqttjs_f5c2a67f) ::  end

Koenkk, 2022-08-17:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['iTRV'],
    model: 'WV704R0A0902',
    vendor: 'Schneider Electric',
    description: 'Wiser radiator thermostat',
    fromZigbee: [fz.ignore_basic_report, fz.ignore_haDiagnostic, fz.ignore_genOta, fz.ignore_zclversion_read,
        fz.legacy.wiser_thermostat, fz.legacy.wiser_itrv_battery, fz.hvac_user_interface, fz.wiser_device_info],
    toZigbee: [tz.thermostat_occupied_heating_setpoint, tz.thermostat_keypad_lockout, tz.thermostat_system_mode],
    exposes: [exposes.climate().withSetpoint('occupied_heating_setpoint', 7, 30, 1).withLocalTemperature(ea.STATE)
        .withSystemMode(['off', 'auto', 'heat'], ea.STATE).withRunningState(['idle', 'heat'], ea.STATE).withPiHeatingDemand()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['genBasic', 'genPowerCfg', 'hvacThermostat', 'haDiagnostic'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.batteryVoltage(endpoint);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatPIHeatingDemand(endpoint);
        // bind of hvacUserInterfaceCfg fails with 'Table Full', does this have any effect?
        await endpoint.configureReporting('hvacUserInterfaceCfg', [{attribute: 'keypadLockout', reportableChange: 1,
            minimumReportInterval: constants.repInterval.MINUTE, maximumReportInterval: constants.repInterval.HOUR}]);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

scaiandre, 2022-08-17:
This puts me into a restart loop with this:
`
[19:20:18] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0-dev start
> node index.js
/app/dist/util/externally-loaded.js:19
        fz.legacy.wiser_thermostat, fz.legacy.wiser_itrv_battery, fz.hvac_user_interface, fz.wiser_device_info],
                  ^
TypeError: Cannot read properties of undefined (reading 'wiser_thermostat')
    at /app/dist/util/externally-loaded.js:19:19
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

`

Koenkk, 2022-08-17:
I've updated https://github.com/Koenkk/zigbee2mqtt/issues/13582#issuecomment-1218234396

scaiandre, 2022-08-17:
This give me this `
[19:56:02] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0-dev start
> node index.js
node:internal/modules/cjs/loader:936
  throw err;
  ^
Error: Cannot find module '../converters/fromZigbee'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at /app/dist/util/externally-loaded.js:1:16
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)

`

scaiandre, 2022-08-17:
Now I get this error `
Zigbee2MQTT:error 2022-08-17 20:00:19: Failed to configure '0x086bd7fffece4c8d', attempt 2 (Error: Bind 0x086bd7fffece4c8d/1 genPowerCfg from '0x00212effff07a3a2/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:764:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9)
    at Object.configure (/app/dist/util/externally-loaded.js:24:9)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))

`

In the UI it seems as if the thermostat can be turned on now, but the thermostat itself is not yet reacting: `
Zigbee2MQTT:error 2022-08-17 20:02:20: Failed to configure '0x086bd7fffece4c8d', attempt 3 (Error: Bind 0x086bd7fffece4c8d/1 haDiagnostic from '0x00212effff07a3a2/1' failed (waiting for response TIMEOUT)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:764:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9)
    at Object.configure (/app/dist/util/externally-loaded.js:24:9)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))

`

Koenkk, 2022-08-17:
Does controlling the temperature setpoint still work?

scaiandre, 2022-08-18:
Yes, when I change it from let's say 20°C to e.g. 15°C it in the homeassistant frontend and then look into the device details in zigbee2mqtt, it shows the 15°C, too.

scaiandre, 2022-08-18:
Despite the "failed to configure" message, some things seem to work on the device. From both the homeassistant frontend and the details screen of the device in z2m, I can control the heating point (occupied_heating_setpoint).

When I set the occupied_heating_setpoint to 30°C, the device opens the valve and I can hear the water streaming into the radiator. I set the occupied_heating_setpoint back to 20°C and the valve closes. 

That is good.

In the homeassistant frontend there are also three other buttons on the bottom of the device control ui, labelled "Automatic", "Heat", "Off" (in German it is "Automatisch", "Heizen", "Aus").

I do not find a similar control option in the device details in z2m ui.

Nevertheless, when I set the device in the homeassistant frontend to "heat" (the flame icon), while the valve is closed, I would expect the device to open the valve and start heating.
In the device details in z2m, the system_mode property changes to "heat" and running_state is "Null". But the valve does not open.

When I click on the "off" icon in the homeassistant frontend, the device details in the z2m show that system_mode changes to "off", but nothing else happens, which is OK, since the valve is already closed.

When I click on the "auto" icon in the homeassistant frontend, the device details in the z2m show that system_mode changes to "auto", but nothing else happens, which is also OK.

Sometimes running_state changes to "N/A" occasionally, but resets back to "Null" after a while.

Another combination of state I tried, was setting the occupied_heating_setpoint to 30°C, which causes the device to open the valve, which is fine. When I then set the device to "off" from the homeassistant ui, again the system_mode property in the device details in z2m changes to "off", which is good. But the valve does not close.

So, communication seems to work, but manually controlling the valve does not seem to work. Is that an issue in the device?

Koenkk, 2022-08-18:
@sjorge do you know if this thermostat supports `system_mode`? Looks that it only supports the `occupied_heating_setpoint`.

sjorge, 2022-08-18:
The wiser never really worked well, it would indeed take a heating setpoint but that was about it.
THere was a lot of info in the ADG and ALG (from memory) message but we never managed to decode it.

I believe someone else got a tiny bit further with it than I did.

scaiandre, 2022-08-19:
Thank you for all the info and help on this. Sad, that it does not seem to work completely. I explicitly bought this device as it is listed as a supported device here: https://www.zigbee2mqtt.io/devices/WV704R0A0902.html

Any suggestions on a fully supported TRV?

sjorge, 2022-08-19:
> Thank you for all the info and help on this. Sad, that it does not seem to work completely. I explicitly bought this device as it is listed as a supported device here: https://www.zigbee2mqtt.io/devices/WV704R0A0902.html
> 
> Any suggestions on a fully supported TRV?

There are 2 that I know have that have pretty good support:
- https://www.zigbee2mqtt.io/devices/014G2461.html
- https://www.zigbee2mqtt.io/devices/H1.html

The Danfoss one is pretty well supported, I had one for a few weeks to play with. But it’s a lot of manufacturer specific clusters.

Currently I am using the relatively new Ubisys H1, and I am personally very satisfied with it. The only thing currently not working is setting the vacation_mode. The manufacturer is also very friendly to work with and provide technical documentation, for the H1 it is currently not available yet on there site but via e-mail they have answered the questions I asked. OTA is also working well on that one. I have found 1 small but in the device, but they acknowledged it is indeed a bug and plan to fix it in the future. (The running_mode is not correctly triggering an update via reporting at the moment) other that than that is is working very well for me and I plan on getting more come winter for my other radiators.

I’ve been looking for a while for a decent TRV I’m happy with and so far this is the only one for me personally. (I’ve gone through a lot, returned most.)

scaiandre, 2022-08-19:
Thank you very much, I order an H1 just right now and will check it out!

Koenkk, 2022-08-19:
Thanks @sjorge , I've removed the `system_mode` expose.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-19 by Koenkk

GSzabados, 2022-11-28:
> Edit: I think we also managed to get the device to enter boost mode, but it's been ages since I poked it and no longer have this TRV as I was not satisfied with it (So far only the latest one I've been toying with was been decent IMHO, pending some more docs from them manufacturer who is at least willing to help, unlike SE were not friendly on there forums)

@sjorge, I also tried to approach Wiser a few years back, probably the same time when support was added originally to Z2M, when I was developing a device handler for SmartThings for the TRVs. The answer was a hard NO from their support.

I figured out that ADC was something to do with the measurements on the device. Battery voltage, temperate, multiple times, and maybe valve position. I think the Hub uses these values for the heating model, and to compensate for environmental changes.

Do you remember by any chance how the boost was triggered?

sjorge, 2022-11-28:
I had a look but doesnt look i have the old testing branch anymore, I vaguely rember it was one of the values in ADC that would get set to 0, 1 or 2 depending on boost up/down or no boost. ~ sjorge