Koenkk/zigbee2mqtt #13587: error message "Not connected to MQTT server!" although broker is available and alive

stachelbeere80, 2022-08-17:
### What happened?

pi@raspi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.27.0 start
> node index.js

Zigbee2MQTT:info  2022-08-17 21:54:52: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-08-17.21-54-52' filename: log.txt
Zigbee2MQTT:info  2022-08-17 21:54:52: Starting Zigbee2MQTT version 1.27.0 (commit #a9b8808)
Zigbee2MQTT:info  2022-08-17 21:54:52: Starting zigbee-herdsman (0.14.46)
**Zigbee2MQTT:error 2022-08-17 21:55:06: Not connected to MQTT server!
Zigbee2MQTT:error 2022-08-17 21:55:06: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2022-08-17 21:55:06: Disconnecting from MQTT server**

/opt/zigbee2mqtt/lib/mqtt.ts:127
        this.client.end();
                    ^
TypeError: Cannot read properties of undefined (reading 'end')
    at MQTT.disconnect (/opt/zigbee2mqtt/lib/mqtt.ts:127:21)
    at Controller.stop (/opt/zigbee2mqtt/lib/controller.ts:197:9)
    at stop (/opt/zigbee2mqtt/index.js:113:5)
    at process.handleQuit (/opt/zigbee2mqtt/index.js:119:9)

configuration.yaml

homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost0
serial:
  port: /dev/ttyUSB0
advanced:
  network_key:
    - 172
    - 91
    - 39
    - 4
    - 85
    - 146
    - 104
    - 172
    - 88
    - 247
    - 188
    - 215
    - 132
    - 48
    - 226
    - 179

### What did you expect to happen?

zigbee2mqtt connect to the mqtt broker

I already tried all various combinations of configuring the server ip

e.g.
server: mqtt://localhost
server: mqtt://localhost:1883
server: mqtt://127.0.0.1
server: mqtt://127.0.0.1:1883
dedicated ip address, with ' and without ...

when I publish a topic to the broker manually it worjs fine

also when I check port 1883 from a remote client with "nc -zv 10.0.40.111 1883 -t" it is shown as open and available

### How to reproduce it (minimal and precise)

try to start zigbee2mqtt

### Zigbee2MQTT version

latest

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219

### Adapter

sonoff zigbee stick usb 3.0

### Debug log

  GNU nano 5.4                                                           log.txt *                                                                   
#info  2022-08-17 21:54:52: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-08-17.21-54-52' filename: log.txt
info  2022-08-17 21:54:52: Starting Zigbee2MQTT version 1.27.0 (commit #a9b8808)
info  2022-08-17 21:54:52: Starting zigbee-herdsman (0.14.46)
error 2022-08-17 21:55:06: Not connected to MQTT server!

aidbish, 2022-08-23:
i seem to be having the same issue using the latest-dev image, using tag latest works without issue

enabled herdmen debug and recieved the following

Using '/app/data' as data directory
Zigbee2MQTT:info  2022-08-23 10:07:03: Logging to console and directory: '/app/data/log/2022-08-23.10-07-03' filename: log.txt
Zigbee2MQTT:info  2022-08-23 10:07:03: Starting Zigbee2MQTT version 1.27.0-dev (commit #9e6b408)
Zigbee2MQTT:info  2022-08-23 10:07:03: Starting zigbee-herdsman (0.14.48)
2022-08-23T00:07:04.241Z zigbee-herdsman:adapter Path '/dev/ttyUSB0' is valid for 'EZSPAdapter'
2022-08-23T00:07:04.241Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[24]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-08-23T00:07:04.242Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-08-23T00:07:04.245Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2022-08-23T00:07:04.245Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-08-23T00:07:04.245Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-08-23T00:07:04.245Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
Zigbee2MQTT:error 2022-08-23 10:07:15: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:59:39)
    at Controller.callExtensions (/app/lib/controller.ts:315:40)
    at Controller.stop (/app/lib/controller.ts:192:9)
    at stop (/app/index.js:113:5)
    at process.handleQuit (/app/index.js:119:9))
Zigbee2MQTT:error 2022-08-23 10:07:15: Not connected to MQTT server!
Zigbee2MQTT:error 2022-08-23 10:07:15: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: '{"state":"offline"}
Zigbee2MQTT:info  2022-08-23 10:07:15: Disconnecting from MQTT server
TypeError: Cannot read properties of undefined (reading 'end')
    at MQTT.disconnect (/app/lib/mqtt.ts:126:21)
    at Controller.stop (/app/lib/controller.ts:197:9)
    at stop (/app/index.js:113:5)
    at process.handleQuit (/app/index.js:119:9)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-08-23 10:08:08: Logging to console and directory: '/app/data/log/2022-08-23.10-08-08' filename: log.txt
Zigbee2MQTT:info  2022-08-23 10:08:08: Starting Zigbee2MQTT version 1.27.0-dev (commit #9e6b408)
Zigbee2MQTT:info  2022-08-23 10:08:08: Starting zigbee-herdsman (0.14.48)
2022-08-23T00:08:08.753Z zigbee-herdsman:adapter Path '/dev/ttyUSB0' is valid for 'EZSPAdapter'
2022-08-23T00:08:08.753Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[24]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-08-23T00:08:08.754Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-08-23T00:08:08.757Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2022-08-23T00:08:08.757Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-08-23T00:08:08.757Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-08-23T00:08:08.758Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]

Koenkk, 2022-08-23:
Can you start z2m with `DEBUG=mqttjs* npm start` and provide me the log here?

stachelbeere80, 2022-08-25:
@Koenkk 

pi@automator:/opt/zigbee2mqtt $ DEBUG=mqttjs* npm start

> zigbee2mqtt@1.27.0 start
> node index.js

Zigbee2MQTT:debug 2022-08-25 13:39:24: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-08-25 13:39:24: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-08-25.13-39-24' filename: log.txt
Zigbee2MQTT:info  2022-08-25 13:39:24: Starting Zigbee2MQTT version 1.27.0 (commit #e0a9323)
Zigbee2MQTT:info  2022-08-25 13:39:24: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:debug 2022-08-25 13:39:24: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:debug 2022-08-25 13:39:31: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:error 2022-08-25 13:39:31: Not connected to MQTT server!
Zigbee2MQTT:error 2022-08-25 13:39:31: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2022-08-25 13:39:31: Disconnecting from MQTT server

/opt/zigbee2mqtt/lib/mqtt.ts:127
        this.client.end();
                    ^
TypeError: Cannot read properties of undefined (reading 'end')
    at MQTT.disconnect (/opt/zigbee2mqtt/lib/mqtt.ts:127:21)
    at Controller.stop (/opt/zigbee2mqtt/lib/controller.ts:197:9)
    at stop (/opt/zigbee2mqtt/index.js:113:5)
    at process.handleQuit (/opt/zigbee2mqtt/index.js:119:9)

pi@automator:/opt/zigbee2mqtt $

Koenkk, 2022-08-25:
Can you modify the `/opt/zigbee2mqtt/index.js` file and change `handleQuit()` like below:

```js
async function handleQuit() {
    console.log("GOT A SIGNAL TO STOP"); // <-- add this line
    if (!stopping && controller) {
        stopping = true;
        await stop(false);
    }
}
```

then provide me the log again.

closed 2022-10-02 by github-actions[bot]

thejacer, 2022-12-07:
`ebug 2022-12-07 07:22:32: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2022-12-07 07:22:32: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-12-07.07-22-32' filename: log.txt
info  2022-12-07 07:22:32: Starting Zigbee2MQTT version 1.28.4 (commit #52e545f)
info  2022-12-07 07:22:32: Starting zigbee-herdsman (0.14.76)
debug 2022-12-07 07:22:32: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/dat>
error 2022-12-07 07:23:34: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/opt/zigbee2mqtt/lib/extension/frontend.ts:59:39)
    at Controller.callExtensions (/opt/zigbee2mqtt/lib/controller.ts:316:40)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Controller.stop (/opt/zigbee2mqtt/lib/controller.ts:193:9)
    at stop (/opt/zigbee2mqtt/index.js:111:5)
    at process.handleQuit (/opt/zigbee2mqtt/index.js:118:9))
`

I added the line in index.js but I don't see where it did anything here. Maybe I did something wrong.