Koenkk/zigbee2mqtt #13589: Schneider Electric CCT5010-0001: No converter available for ‘brightness_move’ (40)

andersklenke, 2022-08-18:
### What happened?

When I'm trying to publish an event to the topic, I get the error: `No converter available for ‘brightness_move’ (40)`

I'm not familiar with the codebase, but it looks like https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/schneider_electric.js#L254-L273 is not exposing the action like the LK Dimmer does here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/schneider_electric.js#L511 Is that correct?

### What did you expect to happen?

That the light dims up. It's described in the documentation [here](https://www.zigbee2mqtt.io/devices/CCT5010-0001.html#moving-stepping).

### How to reproduce it (minimal and precise)

Publish the following event:
```
service: mqtt.publish
data:
  topic: "{{topic}}"
  payload: "{\"brightness_move\": 40}"
```

### Zigbee2MQTT version

 1.27.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus model ZBDongle-E

Koenkk, 2022-08-18:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PUCK/DIMMER/1'],
    model: 'CCT5010-0001',
    vendor: 'Schneider Electric',
    description: 'Micro module dimmer',
    fromZigbee: [...e.light_brightness().withLevelConfig().fromZigbee, fz.wiser_lighting_ballast_configuration],
    toZigbee: [...e.light_brightness().withLevelConfig().toZigbee, tz.ballast_config, tz.wiser_dimmer_mode],
    exposes: [e.light_brightness().withLevelConfig(),
        exposes.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        exposes.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        exposes.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)')],
    whiteLabel: [{vendor: 'Elko', model: 'EKO07090'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(3);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

andersklenke, 2022-08-18:
With the external config it throws this error when starting Zigbee2mqtt:
```
TypeError: e.light_brightness.withLevelConfig.fromZigbee is not iterable
    at /app/dist/util/externally-loaded.js:16:59
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
[22:48:35] INFO: Handing over control to Zigbee2MQTT Core ...
```

Koenkk, 2022-08-19:
Did you also copy the `()`? They are missing from `e.light_brightness.withLevelConfig.fromZigbee`, I expect `...e.light_brightness().withLevelConfig().fromZigbee

andersklenke, 2022-08-19:
Yes, this is the entire log:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/socat.sh
[22:40:12] INFO: Socat not enabled, marking service as down
cont-init: info: /etc/cont-init.d/socat.sh exited 0
cont-init: info: running /etc/cont-init.d/zigbee2mqtt.sh
cont-init: info: /etc/cont-init.d/zigbee2mqtt.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun socat (no readiness notification)
services-up: info: copying legacy longrun zigbee2mqtt (no readiness notification)
[22:40:14] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
/app/dist/util/externally-loaded.js:17
    ...e.light_brightness().withLevelConfig().fromZigbee,
                                             ^
TypeError: e.light_brightness.withLevelConfig.fromZigbee is not iterable
    at /app/dist/util/externally-loaded.js:17:46
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

Koenkk, 2022-08-20:
I've updated https://github.com/Koenkk/zigbee2mqtt/issues/13589#issuecomment-1219709553 , can you check again?

andersklenke, 2022-08-20:
Thanks for your help! Unfortunately still same error. Here's the script:
```js
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const ota = require("zigbee-herdsman-converters/lib/ota");
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  zigbeeModel: ["PUCK/DIMMER/1"],
  model: "CCT5010-0001",
  vendor: "Schneider Electric",
  description: "Micro module dimmer",
  fromZigbee: [
    ...e.light_brightness().withLevelConfig().fromZigbee,
    fz.wiser_lighting_ballast_configuration,
  ],
  toZigbee: [
    ...e.light_brightness().toZigbee,
    tz.ballast_config,
    tz.wiser_dimmer_mode,
  ],
  exposes: [
    e.light_brightness().withLevelConfig(),
    exposes
      .numeric("ballast_minimum_level", ea.ALL)
      .withValueMin(1)
      .withValueMax(254)
      .withDescription("Specifies the minimum light output of the ballast"),
    exposes
      .numeric("ballast_maximum_level", ea.ALL)
      .withValueMin(1)
      .withValueMax(254)
      .withDescription("Specifies the maximum light output of the ballast"),
    exposes
      .enum("dimmer_mode", ea.ALL, ["auto", "rc", "rl", "rl_led"])
      .withDescription(
        "Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)"
      ),
  ],
  whiteLabel: [{ vendor: "Elko", model: "EKO07090" }],
  configure: async (device, coordinatorEndpoint, logger) => {
    await extend
      .light_onoff_brightness()
      .configure(device, coordinatorEndpoint, logger);
    const endpoint = device.getEndpoint(3);
    await reporting.bind(endpoint, coordinatorEndpoint, [
      "genOnOff",
      "genLevelCtrl",
      "lightingBallastCfg",
    ]);
    await reporting.onOff(endpoint);
    await reporting.brightness(endpoint);
  },
};

module.exports = definition;
```

And here's the error:
```
> zigbee2mqtt@1.27.0 start
> node index.js
/app/dist/util/externally-loaded.js:17
    ...e.light_brightness().withLevelConfig().fromZigbee,
                                             ^
TypeError: e.light_brightness.withLevelConfig.fromZigbee is not iterable
    at /app/dist/util/externally-loaded.js:17:46
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

Koenkk, 2022-08-21:
I still see `e.light_brightness().withLevelConfig().fromZigbee` in your script while it should be `e.light_brightness().fromZigbee` as I posted in https://github.com/Koenkk/zigbee2mqtt/issues/13589#issuecomment-1219709553

andersklenke, 2022-08-21:
Okay sorry about that - I misunderstood you there 😊 Although it's still the same error:
```
[10:51:48] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
/app/dist/util/externally-loaded.js:17
    ...e.light_brightness().fromZigbee,
                           ^
TypeError: e.light_brightness.fromZigbee is not iterable
    at /app/dist/util/externally-loaded.js:17:28
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
[10:51:57] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
/app/dist/util/externally-loaded.js:17
    ...e.light_brightness().fromZigbee
```

Full script:
```js
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const ota = require("zigbee-herdsman-converters/lib/ota");
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  zigbeeModel: ["PUCK/DIMMER/1"],
  model: "CCT5010-0001",
  vendor: "Schneider Electric",
  description: "Micro module dimmer",
  fromZigbee: [
    ...e.light_brightness().fromZigbee,
    fz.wiser_lighting_ballast_configuration,
  ],
  toZigbee: [
    ...e.light_brightness().toZigbee,
    tz.ballast_config,
    tz.wiser_dimmer_mode,
  ],
  exposes: [
    e.light_brightness().withLevelConfig(),
    exposes
      .numeric("ballast_minimum_level", ea.ALL)
      .withValueMin(1)
      .withValueMax(254)
      .withDescription("Specifies the minimum light output of the ballast"),
    exposes
      .numeric("ballast_maximum_level", ea.ALL)
      .withValueMin(1)
      .withValueMax(254)
      .withDescription("Specifies the maximum light output of the ballast"),
    exposes
      .enum("dimmer_mode", ea.ALL, ["auto", "rc", "rl", "rl_led"])
      .withDescription(
        "Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)"
      ),
  ],
  whiteLabel: [{ vendor: "Elko", model: "EKO07090" }],
  configure: async (device, coordinatorEndpoint, logger) => {
    await extend
      .light_onoff_brightness()
      .configure(device, coordinatorEndpoint, logger);
    const endpoint = device.getEndpoint(3);
    await reporting.bind(endpoint, coordinatorEndpoint, [
      "genOnOff",
      "genLevelCtrl",
      "lightingBallastCfg",
    ]);
    await reporting.onOff(endpoint);
    await reporting.brightness(endpoint);
  },
};

module.exports = definition;

```

Koenkk, 2022-08-21:
Made a stupid mistake, sorry! this should do it:

```js
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const ota = require("zigbee-herdsman-converters/lib/ota");
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  zigbeeModel: ["PUCK/DIMMER/1"],
  model: "CCT5010-0001",
  vendor: "Schneider Electric",
  description: "Micro module dimmer",
  fromZigbee: [
    ...extend.light_onoff_brightness().fromZigbee,
    fz.wiser_lighting_ballast_configuration,
  ],
  toZigbee: [
    ...extend.light_onoff_brightness().toZigbee,
    tz.ballast_config,
    tz.wiser_dimmer_mode,
  ],
  exposes: [
    e.light_brightness().withLevelConfig(),
    exposes
      .numeric("ballast_minimum_level", ea.ALL)
      .withValueMin(1)
      .withValueMax(254)
      .withDescription("Specifies the minimum light output of the ballast"),
    exposes
      .numeric("ballast_maximum_level", ea.ALL)
      .withValueMin(1)
      .withValueMax(254)
      .withDescription("Specifies the maximum light output of the ballast"),
    exposes
      .enum("dimmer_mode", ea.ALL, ["auto", "rc", "rl", "rl_led"])
      .withDescription(
        "Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)"
      ),
  ],
  whiteLabel: [{ vendor: "Elko", model: "EKO07090" }],
  configure: async (device, coordinatorEndpoint, logger) => {
    await extend
      .light_onoff_brightness()
      .configure(device, coordinatorEndpoint, logger);
    const endpoint = device.getEndpoint(3);
    await reporting.bind(endpoint, coordinatorEndpoint, [
      "genOnOff",
      "genLevelCtrl",
      "lightingBallastCfg",
    ]);
    await reporting.onOff(endpoint);
    await reporting.brightness(endpoint);
  },
};

module.exports = definition;
```

andersklenke, 2022-08-21:
Hehe, no problem! This was it. Tested on the device and it works now 👌🏼 Thanks a lot for your help!  Will you do a PR with the fix, so we avoid more people having this issue?

Koenkk, 2022-08-22:
Integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-22 by Koenkk

joaquinvacas, 2022-11-01:
https://github.com/Koenkk/zigbee2mqtt/issues/14749

Maybe this ↑↑↑↑ is related to this issue? Maybe *-0002W variant is different (also, using Install Codes as I wrote on the issue)