Koenkk/zigbee2mqtt #13591: Failed to pair device after change Zigbee Coordinator. Device left the network

matfil92, 2022-08-18:
### What happened?

Good morning to all,
I am writing because I am going crazy with the configuration of Zigbee2MQTT. 

I installed Zigbee2MQTT a week ago, initially I was using it with Sonoff Zigbee Bridge perfectly working. Now I deleted the 3 test devices I had associated to configure the Sonoff Zigbee USB 3.0 usb stick with the correct firmware. The stick is correctly seen by the system but every time I press on Pairing and try to associate one or more devices I always get the same problem. It finds them and after 2 seconds they appear the message "device left the network" appears. 
I have already followed the FAQ page and tested everything i.e. put usb extension cable and put the device 2 centimeters closer while pairing but always the same problem. I also tried rebooting and changing USB port but nothing, nothing changes.
Can you help me understand what can it be? In desperation I also tried uninstalling and reinstalling Zigbee2MQTT but still the same problem. 

Below is the log set in debug mode 

```
Debug 2022-08-18 11:27:09Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"permit_join":false},"transaction":"fllvx-1"}'
Info 2022-08-18 11:27:09Zigbee: disabling joining new devices.
Info 2022-08-18 11:27:09Succesfully changed options
Info 2022-08-18 11:27:09MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"fllvx-1"}'
Debug 2022-08-18 11:27:17Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"fllvx-2","value":true}'
Info 2022-08-18 11:27:17Zigbee: allowing new devices to join.
Info 2022-08-18 11:27:17MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"fllvx-2"}'
Info 2022-08-18 11:27:41Device '0x001788011032ca4d' joined
Info 2022-08-18 11:27:41MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788011032ca4d","ieee_address":"0x001788011032ca4d"},"type":"device_joined"}'
Info 2022-08-18 11:27:41MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788011032ca4d"},"type":"device_connected"}'
Info 2022-08-18 11:27:41Starting interview of '0x001788011032ca4d'
Info 2022-08-18 11:27:41MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788011032ca4d","ieee_address":"0x001788011032ca4d","status":"started"},"type":"device_interview"}'
Info 2022-08-18 11:27:41MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788011032ca4d"},"type":"pairing"}'
Warning 2022-08-18 11:27:51Device '0x001788011032ca4d' left the network
Info 2022-08-18 11:27:51MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788011032ca4d","ieee_address":"0x001788011032ca4d"},"type":"device_leave"}'
Info 2022-08-18 11:27:51MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x001788011032ca4d"},"type":"device_removed"}'
Info 2022-08-18 11:28:01Device '0x001788011032ca4d' joined
Info 2022-08-18 11:28:01MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788011032ca4d","ieee_address":"0x001788011032ca4d"},"type":"device_joined"}'
Info 2022-08-18 11:28:01MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788011032ca4d"},"type":"device_connected"}'
Info 2022-08-18 11:28:01Starting interview of '0x001788011032ca4d'
Info 2022-08-18 11:28:01MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788011032ca4d","ieee_address":"0x001788011032ca4d","status":"started"},"type":"device_interview"}'
Info 2022-08-18 11:28:01MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788011032ca4d"},"type":"pairing"}'
Warning 2022-08-18 11:28:02Device '0x001788011032ca4d' left the network
Info 2022-08-18 11:28:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788011032ca4d","ieee_address":"0x001788011032ca4d"},"type":"device_leave"}'
Info 2022-08-18 11:28:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x001788011032ca4d"},"type":"device_removed"}'
Info 2022-08-18 11:28:02Device '0x001788011032ca4d' joined
Info 2022-08-18 11:28:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788011032ca4d","ieee_address":"0x001788011032ca4d"},"type":"device_joined"}'
Info 2022-08-18 11:28:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788011032ca4d"},"type":"device_connected"}'
Info 2022-08-18 11:28:02Starting interview of '0x001788011032ca4d'
Info 2022-08-18 11:28:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788011032ca4d","ieee_address":"0x001788011032ca4d","status":"started"},"type":"device_interview"}'
Info 2022-08-18 11:28:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788011032ca4d"},"type":"pairing"}'
```

### What did you expect to happen?

Device found and paired successfully 

### How to reproduce it (minimal and precise)

Try to reset device and repairing always happen the same problem, also with different devices supported by Zigbee2MQTT.

In the log above I've tried with HUE Dimmer Switch

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee USB 3.0 E

kirovilya, 2022-08-18:
Hi. I also have a similar problem with one of the devices - goes offline a few seconds after connecting. looking for a solution. I'll post here if I fix it.

alexberdnik, 2022-08-19:
Same issue for me. I guess the "E" is not fully supported yet.

matfil92, 2022-08-20:
Hi,
I confirm that I've solved the issue. the dongle -E is not supported. I've purchased the -E because in Amazon was wrote that is compatible with Home Assistant but in reality is not true. Also the firmware upgrade guide and procedure is not compatible with the -E version.

Today is arrived the -P version and works well!

closed 2022-09-27 by github-actions[bot]