Koenkk/zigbee2mqtt #13592: OSRAM Switch Mini - holding up not recognised?

vaderag, 2022-08-18:
### What happened?

I have the OSRAM Switch Mini (https://www.zigbee2mqtt.io/devices/AC0251100NJ_AC0251600NJ_AC0251700NJ.html) and have set this up in Zigbee2MQTT with Home Assistant

I've tried both Legacy and non-legacy but for some reason a hold on the top button doesn't seem to report anything

I'm assuming it should report the opposite of the down button and therefore
brightness_move_up = up long press brightness_stop = up long press release

For reference, I've tested out what is reported and I get the following overall:

on = Up (short)
off = Down (short)

brightness_move_to_level = middle (short)
color_temperature_move = middle 
    nb. both trigger (colour slightly after)

move_to_saturation = middle (long press)
    hue_move = middle long press hold
    hue_stop = middle long press release

brightness_move_down = down long press 
    brightness_stop = down long press release

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle 3.0

vaderag, 2022-09-18:
Unstaleing

closed 2022-10-27 by github-actions[bot]

PulsarFX, 2024-04-05:
I have 2 of those switches. Both behave the same: Holding up works for me, but not holding down.
For up button I get `on`, `brightness_move_up` and `brightness_stop` on release.
For down button I get nothing on long press. Only `off` for normal press.

```
Firmware-Date    201711162225070
Firmware-Version    e.1.11.0M
```