Koenkk/zigbee2mqtt #13595: set color HSV Value does not set brightness

NickWaterton, 2022-08-18:
### What happened?

Set the hsb (or hsv) value for color. Color of bulb (Hue Bloom) changes, but brightness does not.

Looks like the issue is in zigbee-herdsman-converters, file: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js between line 1064 and 1086 where the 'v' value (from hsv conversion) should be set as `value.brightness` and is sent as a brightness command. This does not seem to happen.

### What did you expect to happen?

Brightness should change according to 'b' value of 'hsb' setting (or v value of 'hsv' command).

See #9205 and https://github.com/Koenkk/zigbee2mqtt/issues/9523 

### How to reproduce it (minimal and precise)

Send:
```
mosquitto_pub -h 192.168.100.16 -t "zigbee2mqtt/Hue Bloom/set/color" -m "{\"hsb\":\"359,93,100\"}"
mosquitto_pub -h 192.168.100.16 -t "zigbee2mqtt/Hue Bloom/set/color" -m "{\"hsb\":\"359,93,0\"}"
```
brightness should be set to 100, then 0, but brightness does not change.

### Zigbee2MQTT version

1.27.0 commit: a9b8808

### Adapter firmware version

0x26580700

### Adapter

Deconz Conbee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9373929/log.txt)

relevant log data is:
```
debug 2022-08-18 08:43:32: Received MQTT message on 'zigbee2mqtt/Hue Bloom/set/color' with data '{"hsb":"359,93,100"}'
debug 2022-08-18 08:43:32: Publishing 'set' 'color' to 'Hue Bloom'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/last_seen', payload '2022-08-18T08:43:32-04:00'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/linkquality', payload '95'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/last_seen', payload '2022-08-18T08:43:32-04:00'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/linkquality', payload '103'
debug 2022-08-18 08:43:32: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x0017880100c2d581!
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color_mode', payload 'hs'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color-hue', payload '359'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color-saturation', payload '93'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color-x', payload '0.619'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color-y', payload '0.3007'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/last_seen', payload '2022-08-18T08:43:32-04:00'
info  2022-08-18 08:43:32: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/linkquality', payload '103'
debug 2022-08-18 08:43:37: Received MQTT message on 'zigbee2mqtt/Hue Bloom/set/color' with data '{"hsb":"359,93,0"}'
debug 2022-08-18 08:43:37: Publishing 'set' 'color' to 'Hue Bloom'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/last_seen', payload '2022-08-18T08:43:37-04:00'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/linkquality', payload '95'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/last_seen', payload '2022-08-18T08:43:37-04:00'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/linkquality', payload '103'
debug 2022-08-18 08:43:37: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x0017880100c2d581!
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color_mode', payload 'hs'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color-hue', payload '359'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color-saturation', payload '93'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color-x', payload '0.619'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/color-y', payload '0.3007'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/last_seen', payload '2022-08-18T08:43:37-04:00'
info  2022-08-18 08:43:37: MQTT publish: topic 'zigbee2mqtt/Hue Bloom/linkquality', payload '103'
```

Quautiputzli, 2022-09-21:
Same Problem.
Will there be a solution? I am still using version 1.18. becouse this is the last one which work with HSB an Phillips lamps.

Quautiputzli, 2022-09-23:
Here is also the same problem:
https://github.com/Koenkk/zigbee2mqtt/issues/9523
Ist there a way to use the converter from version 1.18.3 in the new version of zigbee2mqtt.
In version 1.18.3 brightness setting over HSB/HSV works.

closed 2022-11-01 by github-actions[bot]

geforce28, 2023-12-02:
Push here...
@Koenkk  Can you please integrate this ?
Its nessasary fuction to use HSV Value..