Koenkk/zigbee2mqtt #13598: KCTW1Z No battery reading

whakru, 2022-08-18:
### What happened?

https://www.zigbee2mqtt.io/devices/KCTW1Z.html
no indicators:
battery_low
battery

[image]

### Zigbee2MQTT version

 1.27.0

### Adapter firmware version

20220301

### Adapter

zStack3x0

whakru, 2022-09-18:
not fixed

Koenkk, 2022-09-19:
@kirovilya do we know the data points for this?

kirovilya, 2022-09-19:
It is good for me when click on back button:
[image]
[image]
Let's try to add reporting...

whakru, 2022-09-19:
Indeed, after pressing the back button, the battery readings appear.

Koenkk, 2022-09-20:
Should be fixed now!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-20 by Koenkk