Koenkk/zigbee2mqtt #13600: Interview fails for Sonoff SNZB-03 motion sensor

blacknell, 2022-08-19:
### What happened?

Preiodically, I lose connectivity with devices. This has mostly occurred when the battery runs low. However, reconnecting is very problematic.

I can no longer properly pair this SNZB-03. It has a new battery and I'm admitting it via a nearby switch which is a router on the network (with good LQI).

### What did you expect to happen?

I expected it to pair, interview and start working.

### How to reproduce it (minimal and precise)

I first removed the device from the network
Then I selected Permit Join from Dining Room Dimmer
Then I initiated pairing on the motion sensor.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

 Sonoff Zigbee 3.0 USB Dongle Plus × 1

### Debug log

```
info  2022-08-19 08:21:30: Accepting joining not in blocklist device '0x00124b00251973a3'
info  2022-08-19 08:21:30: Accepting joining not in blocklist device '0x00124b00251973a3'
info  2022-08-19 08:21:30: Device '0x00124b00251973a3' joined
info  2022-08-19 08:21:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00251973a3","ieee_address":"0x00124b00251973a3"},"type":"device_joined"}'
info  2022-08-19 08:21:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b00251973a3"},"type":"device_connected"}'
info  2022-08-19 08:21:30: MQTT publish: topic 'zigbee2mqtt/0x00124b00251973a3/availability', payload 'online'
info  2022-08-19 08:21:30: MQTT publish: topic 'zigbee2mqtt/0x00124b00251973a3', payload '{"battery_low":false,"last_seen":"2022-08-19T08:21:30+01:00","linkquality":134,"occupancy":false,"tamper":false}'
info  2022-08-19 08:21:30: Starting interview of '0x00124b00251973a3'
info  2022-08-19 08:21:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00251973a3","ieee_address":"0x00124b00251973a3","status":"started"},"type":"device_interview"}'
info  2022-08-19 08:21:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b00251973a3"},"type":"pairing"}'
info  2022-08-19 08:21:31: MQTT publish: topic 'zigbee2mqtt/0x00124b00251973a3', payload '{"battery_low":false,"last_seen":"2022-08-19T08:21:31+01:00","linkquality":134,"occupancy":false,"tamper":false}'
error 2022-08-19 08:23:49: Failed to interview '0x00124b00251973a3', device has not successfully been paired
info  2022-08-19 08:23:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00251973a3","ieee_address":"0x00124b00251973a3","status":"failed"},"type":"device_interview"}'
info  2022-08-19 08:23:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b00251973a3"},"type":"pairing"}'
```

blacknell, 2022-08-22:
This problem happens a lot with my battery sensors. I often have to delete and re-pair after battery replacement and then frequently experience this behaviour of a failed interview.

Let me know if there's more info needed to help diagnose

blacknell, 2022-09-03:
@Koenkk - any chance we could investigate. I'm happy to try and tests to debug. I'm seeing this a lot on the Sonoff battery devices

Koenkk, 2022-09-06:
> This problem happens a lot with my battery sensors. I often have to delete and re-pair after battery replacement and then frequently experience this behaviour of a failed interview.

Does this only happen with Sonoff devices?

> any chance we could investigate. I'm happy to try and tests to debug.

Can you provide the herdsman debug log when repairing this device?

blacknell, 2022-09-07:
Now, of course pairing is working. I tried a few times and it works and as I said the problem is intermittent.

I will leave herdsman debug logging on and update when i next have to change a battery. For now, the only log I have of failure is the one above without the DEBUG detail

> Does this only happen with Sonoff devices?

Yes, but I only have Sonoff battery devices. But I had it happen with multiple motion and temperature ones.

blacknell, 2022-09-27:
OK. I found a failed interview after installing (trying to) this brand new contact sensor.

I first did the usual allow join and it did see it very quickly. The interview did seem to take a long time and then showed the | triangle warning 'Interview failed'

I then clicked Reconfigure and the device appears to be partially working but still showing interview failed.

```
info  2022-09-27 11:33:38: Accepting joining not in blocklist device '0x00124b00254d1387'
info  2022-09-27 11:33:38: Device '0x00124b00254d1387' joined
info  2022-09-27 11:33:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00254d1387","ieee_address":"0x00124b00254d1387"},"type":"device_joined"}'
info  2022-09-27 11:33:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b00254d1387"},"type":"device_connected"}'
info  2022-09-27 11:33:38: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387/availability', payload 'online'
info  2022-09-27 11:33:38: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:38+01:00"}'
info  2022-09-27 11:33:38: Starting interview of '0x00124b00254d1387'
info  2022-09-27 11:33:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00254d1387","ieee_address":"0x00124b00254d1387","status":"started"},"type":"device_interview"}'
info  2022-09-27 11:33:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b00254d1387"},"type":"pairing"}'
info  2022-09-27 11:33:41: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:41+01:00"}'
debug 2022-09-27 11:33:41: Device '0x00124b00254d1387' announced itself
info  2022-09-27 11:33:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00254d1387","ieee_address":"0x00124b00254d1387"},"type":"device_announce"}'
info  2022-09-27 11:33:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b00254d1387"},"type":"device_announced"}'
debug 2022-09-27 11:33:45: Received Zigbee message from '0x00124b00254d1387', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"eWeLink","modelId":"DS01"}' from endpoint 1 with groupID 0
info  2022-09-27 11:33:45: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:45+01:00","linkquality":21}'
debug 2022-09-27 11:33:49: Received Zigbee message from '0x00124b00254d1387', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
info  2022-09-27 11:33:49: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:49+01:00","linkquality":18}'
debug 2022-09-27 11:33:50: Received Zigbee message from '0x00124b00254d1387', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
info  2022-09-27 11:33:50: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:50+01:00","linkquality":29}'
info  2022-09-27 11:33:55: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:55+01:00","linkquality":29}'
debug 2022-09-27 11:33:55: Device '0x00124b00254d1387' announced itself
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00254d1387","ieee_address":"0x00124b00254d1387"},"type":"device_announce"}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b00254d1387"},"type":"device_announced"}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
info  2022-09-27 11:33:56: MQTT publish: topic 'zigbee2mqtt/0x00124b00254d1387', payload '{"last_seen":"2022-09-27T11:33:56+01:00","linkquality":145}'
[... cut ...]
error 2022-09-27 11:37:45: Failed to configure '0x00124b00254d1387', attempt 1 (Error: Bind 0x00124b00254d1387/1 genPowerCfg from '0x00124b0024c04f51/1' failed (AREQ - ZDO - bindRsp after 10000ms)
info  2022-09-27 11:37:45: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x00124b00254d1387"},"error":"Failed to configure (Bind 0x00124b00254d1387/1 genPowerCfg from '0x00124b0024c04f51/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"opbcv-1"}'
```

Koenkk, 2022-09-28:
> The interview did seem to take a long time and then showed the | triangle warning 'Interview failed'

Can you try to repair the device closer to the coordinator? Could be a temporary hickup.

blacknell, 2022-09-28:
I did and it did pair. Actually I was limiting it to pairing with a specific router not the coordinator

blacknell, 2022-10-19:
After much trial and error and I have found a new reliable way of re-pairing Sonoff SNZB-0x devices including making them live and accepting of reporting update changes

- Get yourself to the Zigbee dashboard Devices page
- Make sure you are not Permitting Join
- Press and hold the reset button (or paperclip switch) for 5s until red light blinks
  - You will see the device disappear and leave the network
- Click Permit Join
  - You will see the previously unblocked device re-join and various status popups flash to the top right of the screen
- After waiting for the device to be set up (but not long after) drill into the device and click on Reporting
  - Here you can change the reporting interval (I have settled on 3600s minimum and 7200s maximum).

I found with the default of 62000s max internal and 25 hours default availability window if one broadcast is missed by the router then the device is flagged offline. With 7200s it would mean 13 broadcasts would have to have been missed.

blacknell, 2022-10-22:
> After much trial and error and I have found a new reliable way of re-pairing Sonoff SNZB-0x devices including making them live and accepting of reporting update changes

This doesn't seem to work very well for SNZB-03 Motion Detectors but does work reliably for SNZB-02 and SNZB-04

ncrmnt, 2022-11-25:
> > After much trial and error and I have found a new reliable way of re-pairing Sonoff SNZB-0x devices including making them live and accepting of reporting update changes
> 
> This doesn't seem to work very well for SNZB-03 Motion Detectors but does work reliably for SNZB-02 and SNZB-04

I've the same issue. But, unfortunately, I can't change the settings which you wrote about, get errors:
Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0x00124b0025177e3d/1 genPowerCfg from '0x0c4314fffe8659f2/1' failed (Route error)'
Bind 0x00124b0025177e3d/1 genPowerCfg from '0x0c4314fffe8659f2/1' failed (Route error)

Maybe I do something wrong? Do you mean this settings? 
[image]

blacknell, 2022-11-25:
> I've the same issue. But, unfortunately, I can't change the settings which you wrote about, get errors:
> 
> Maybe I do something wrong? Do you mean this settings? 

Yes. 

But it only worked for me immediately (within a few seconds) of having pressed the device button for 5s, enabling join and then changing as soon as join pop ups started appearing. It is messy.

ncrmnt, 2022-11-28:
> Yes.
>
> But it only worked for me immediately (within a few seconds) of having pressed the device button for 5s, enabling join and then changing as soon as join pop ups started appearing. It is messy.

Many thanks! Now it works properly

brindille91, 2023-01-01:
> 

@blacknell, thanks for posting this solution.
I have 6x SNZB-04 and was not able to find a stable solution, i was about to send them back all.
With you help and setting max interval to 7200s it has now at least past 48 hours without failing.
You made my start of year 2023 much better :-)

brindille91, 2023-01-01:
@blacknell, thanks for posting this solution.
I have 6x SNZB-04 and was not able to find a stable solution.
With you help and setting max interval to 7200s it seems to be much more stable.

Koenkk, 2023-01-02:
@brindille91 should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

blacknell, 2023-01-02:
Can you advise which PR addresses this? Is it in this repo or the converters?

Regards
Paul

Koenkk, 2023-01-02:
@blacknell whoops, this comment was meant for a different issue. Can someone provide the herdsman debug log when trying to pair this device?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

blacknell, 2023-01-02:
I did add the hersdman debug log see above 27/9

blacknell, 2023-01-03:
I think/suggest this is the same for all the Sonoff SNZB-0x family. 

I'm happy raise a PR for this if it helps?

closed 2023-01-04 by Koenkk

Casper-O, 2023-01-12:
Mine just dropped again after a few days without issues

Friendly name is "vindue soveværelse"
```
Zigbee2MQTT:info  2023-01-11 03:42:25: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":96,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 03:42:25: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":92,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 05:02:57: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":92,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 05:02:57: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":88,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 06:23:28: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":104,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 06:23:28: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":100,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 07:40:21: MQTT publish: topic 'zigbee2mqtt/Loftlampe spisebord', payload '{"brightness":216,"linkquality":196,"power_on_behavior":"on","state":"ON","update":{"installed_version":16782340,"latest_version":16784906,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2023-01-11 07:40:23: MQTT publish: topic 'zigbee2mqtt/Loftlampe spisebord', payload '{"brightness":216,"linkquality":196,"power_on_behavior":"on","state":"OFF","update":{"installed_version":16782340,"latest_version":16784906,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2023-01-11 07:40:26: MQTT publish: topic 'zigbee2mqtt/Loft Lampe Soveværelse', payload '{"brightness":31,"linkquality":136,"power_on_behavior":null,"state":"ON","update":{"installed_version":16782340,"latest_version":16784906,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2023-01-11 07:44:43: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":92,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 07:44:43: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":92,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 09:05:52: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":92,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 09:05:52: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":92,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 09:32:52: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":92,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 10:20:27: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":96,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 10:20:36: MQTT publish: topic 'zigbee2mqtt/Loft Lampe Soveværelse', payload '{"brightness":31,"linkquality":140,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16782340,"latest_version":16784906,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2023-01-11 10:26:32: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":96,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 10:26:32: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":96,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 11:46:36: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":100,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 11:46:36: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":100,"tamper":false,"voltage":3100}'
Zigbee2MQTT:info  2023-01-11 13:07:08: MQTT publish: topic 'zigbee2mqtt/vindue soveværelse', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":92,"tamper":false,"voltage":3000}'
Zigbee2MQTT:info  2023-01-11 15:48:34: MQTT publish: topic 'zigbee2mqtt/Loftlampe spisebord', payload '{"brightness":216,"linkquality":196,"power_on_behavior":"on","state":"ON","update":{"installed_version":16782340,"latest_version":16784906,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2023-01-11 15:48:37: MQTT publish: topic 'zigbee2mqtt/Loftlampe spisebord', payload '{"brightness":254,"linkquality":196,"power_on_behavior":"on","state":"ON","update":{"installed_version":16782340,"latest_version":16784906,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2023-01-11 15:48:42: MQTT publish: topic 'zigbee2mqtt/Loftlampe spisebord', payload '{"brightness":175,"linkquality":196,"power_on_behavior":"on","state":"ON","update":{"installed_version":16782340,"latest_version":167849 [... cut ...]
Zigbee2MQTT:info  2023-01-11 16:58:40: MQTT publish: topic 'zigbee2mqtt/TV bord 1', payload '{"brightness":76,"linkquality":208,"power_on_behavior":null,"state":"ON","update":{"installed_version":16784906,"latest_version":16784906,"state":"idle"},"update_available":null}'
Zigbee2MQTT:warn  2023-01-11 17:22:39: Device 'vindue soveværelse' left the network
Zigbee2MQTT:info  2023-01-11 17:22:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"vindue soveværelse","ieee_address":"0x00124b0029116d6f"},"type":"device_leave"}'
```

majkers, 2023-01-27:
The same for my sonoff-snzb-04.

efolio, 2023-02-15:
I hope this will get fixed and pushed soon… I have 21 sonoff devices (snzb-02 and 04), and they keep disappearing from the network again and again… The only way to get them back online is to pair again, which works for a few hours before dropping out again…
Some of the devices are 2 meters away from the dongle…

nazgul6, 2023-06-04:
I have the same issue with my SNZB-04 too (running latest Z2M from last week). Has anyone come up with any other potential solutions perhaps? Should this issue be-reopened?

mk01, 2023-06-12:
don't know for the others, but for me the devices (despite the fact that interview is not-finished and tab reports is not showing anything set) ARE working, that means SNZB-04 is providing 'contact' updates and SNZB-03 is providing 'occupancy'. Apart from those, both types are also properly providing 'battery_low'. 

out from 10 of those, one SNZB-04 managed to finish interview, no clue what was different. 
btw - none of the devices is leaving the network.

maybe it is worth saying that for 1 year I was running version 1.26, there was no issue with those. this behavior started after I upgraded to 1.29 and 1.30 for EZSP adapter support (tubeszb-efr32 POE adapter).
as I see that @blacknell is having sonoff zigbee usb adapter (as well EZSP stack?), maybe the issue is not about z2m directly, but about the EZSP stack. no clue.

but again, I don't care to see the battery % , so (for me) definitely critical.

xking3, 2023-08-24:
I have the same problem here. Two SNZB-3 and they disconnect after one to two weeks.
Are there any solutions to this problem. Or should I buy other sensors? Any suggestion is welcome.

nazgul6, 2023-08-24:
I got a cheap Zigbee lightbulb, which acts as a router at the same, and is plugged very close to my SNZB-04 to ‘expand/extend’ the network. No more drops. I guess the zigbee sensors in SNZB-04 must be quite weak and need some sort of a repeater/router nearby to ensure they stay in the network.

xking3, 2023-08-25:
I have a lot of Zigbee devices in the same room, but it still disconnects. If the SNZB-03 & SNZB-04 is as week as you describe it, I will replace them immediately.

deece, 2023-09-26:
I have all manner of devices (not just Sonoff) dropping randomly, with both Ti coordinators and EFR ones.

I asked ChatGPT to write an extension to automatically apply the max report interval for all devices as they are interviewed, you can add it in the 'extensions' tab.

```
class BatteryPercentageReportingExtension {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.zigbee = zigbee;
        this.logger = logger;
        this.eventBus = eventBus;
        this.eventBus.on('deviceInterview', this.onDeviceInterview.bind(this), this.constructor.name);
    }

    async onDeviceInterview(data) {
        const { device, status } = data;
        if (status === 'successful') {
            const endpoint = device.getEndpoint(1);
            if (endpoint && endpoint.supportsInputCluster('genPowerCfg')) {
                try {
                    const cluster = 'genPowerCfg';
                    const attribute = 'batteryPercentageRemaining';
                    const minReportInterval = 0;
                    const maxReportInterval = 3600;
                    const reportableChange = 0;

                    await endpoint.bind(cluster);
                    await endpoint.configureReporting(cluster, [{
                        attribute: { ID: 0x0021, type: 32 }, // batteryPercentageRemaining
                        minimumReportInterval: minReportInterval,
                        maximumReportInterval: maxReportInterval,
                        reportableChange: reportableChange
                    }]);
                    this.logger.info(`Configured reporting for ${device.ieeeAddr} ${cluster} ${attribute}`);
                } catch (e) {
                    this.logger.error(`Failed to configure reporting for ${device.ieeeAddr} genPowerCfg batteryPercentageRemaining`, e);
                }
            }
        }
    }

    async stop() {
        this.eventBus.removeListeners(this.constructor.name);
    }
}

module.exports = BatteryPercentageReportingExtension;
```