Koenkk/zigbee2mqtt #13604: MOES Tuya Wall Switch TS0012 Activates in Tandem

shuufi, 2022-08-19:
### What happened?

Hello, I bought a few [Moes branded wall switches](https://shopee.com.my/Moes%F0%9F%92%A5-Smart-Switch-Tuya-ZigBee%F0%9F%92%A5-No-Neutral-Wire-No-Capacitor-Needed-Smart-Life-2-3-Way-Works-with-Alexa-Google-Home-2mqtt-i.759938586.19100866057?xptdk=60f8feb1-dca7-44cc-a3a5-ac82853d8b9d) and managed to get them installed and paired in Zigbee2MQTT successfully. Using Zigbee2MQTT, they were recognised as TS0012. If registered and managed via Zibgee2MQTT, the switches always go in tandem, i.e., if the left switch is being switched on (via lovelace or backend), the right switch is also being switched off, and vice versa.

This doesn't happen if I connect the wall switch using the Tuya integration. I would like to get some assistance if I missed out on any steps or configurations after successfully pairing the devices.

Exposes section
[image]
Turn on the left switch, which will also activate the right switch and vice versa.

Binding section
[image]

State
```
{
    "last_seen": "2022-08-19T21:37:28+08:00",
    "linkquality": 25,
    "state_left": "OFF",
    "state_right": "OFF"
}
```
Thanks in advance.

### What did you expect to happen?

Turn on the left switch, will only activate the left switch, i.e., no side-effect. Vice versa.

### How to reproduce it (minimal and precise)

1. Pair the wall switch
2. Create lovelace button card, or from the Zigbee2MQTT device page, turn on/off either switch.

### Zigbee2MQTT version

 1.27.0

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus Stick TI CC2652P 

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9382858/log.txt)

Koenkk, 2022-08-19:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0012'],
    model: 'TS0012',
    vendor: 'TuYa',
    description: 'Smart light switch - 2 gang',
    whiteLabel: [{vendor: 'Vrey', model: 'VR-X712U-0013'}, {vendor: 'TUYATEC', model: 'GDKES-02TZXD'},
        {vendor: 'Earda', model: 'ESW-2ZAA-EU'}],
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right')],
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

shuufi, 2022-08-20:
Excellent, it does fix the issues. Is it supposed to be added via extension and not work out of the box?

I have additional switches under TS0013 and managed to get them working by doing the same. Pasting the codes here as it could be useful for others. What's important here is the module must export either an object or an array of objects, like so.

```
const definition = [
    {
        zigbeeModel: ['TS0012'],
        model: 'TS0012',
        vendor: 'TuYa',
        description: 'Smart light switch - 2 gang',
        whiteLabel: [{ vendor: 'Vrey', model: 'VR-X712U-0013' }, { vendor: 'TUYATEC', model: 'GDKES-02TZXD' },
        { vendor: 'Earda', model: 'ESW-2ZAA-EU' }],
        extend: extend.switch(),
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right')],
        endpoint: (device) => {
            return { 'left': 1, 'right': 2 };
        },
        meta: { multiEndpoint: true },
        configure: async (device, coordinatorEndpoint, logger) => {
            await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
    },
    {
        zigbeeModel: ['TS0013'],
        model: 'TS0013',
        vendor: 'TuYa',
        description: 'Smart light switch - 3 gang without neutral wire',
        extend: extend.switch(),
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right')],
        endpoint: (device) => {
            return { 'left': 1, 'center': 2, 'right': 3 };
        },
        whiteLabel: [{ vendor: 'TUYATEC', model: 'GDKES-03TZXD' }],
        meta: { multiEndpoint: true },
        configure: async (device, coordinatorEndpoint, logger) => {
            await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
            try {
                for (const ID of [1, 2, 3]) {
                    const endpoint = device.getEndpoint(ID);
                    await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
                }
            } catch (e) {
                // Fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/4872
            }
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
    }
];

module.exports = definition;
```

closed 2022-08-20 by shuufi