Koenkk/zigbee2mqtt #13607: SONOFF Zigbee 3.0 USB Dongle Plus V2 (EFR32MG21) - Route Error during device join

alexberdnik, 2022-08-19:
### What happened?

Device is not connected to the network; log has the following error message:
```
Error: Route error
    at /app/node_modules/zigbee-herdsman/src/utils/waitress.ts:33:70
    at Waitress.forEachMatching (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:83:17)                                               at Waitress.reject (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:33:21)
    at Driver.handleRouteError (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:371:23)
    at Driver.handleFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:300:18)
    at Ezsp.emit (node:events:527:28)
    at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:294:28)
    at SerialDriver.emit (node:events:527:28)
    at SerialDriver.handleDATA (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:209:14)
    at SerialDriver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:161:22)
```
Docker container is restarted after that.

### What did you expect to happen?

Normal operation.

### Zigbee2MQTT version

1.27.0 commit: a9b8808

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2 (EFR32MG21)

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9383716/log.txt)

alexberdnik, 2022-08-19:
The device in question is https://www.zigbee2mqtt.io/devices/TS0201.html, works perfectly with my older dongles.

kirovilya, 2022-08-28:
try again on the dev-version (edge) of z2m and tell if it works or not.

alexberdnik, 2022-08-30:
I've set up a separate zigbee network and paired two sensors and a button successfully. https://www.zigbee2mqtt.io/devices/TS0201.html#tuya-ts0201 and https://www.zigbee2mqtt.io/devices/LCZ030.html#tuya-lcz030 and https://www.zigbee2mqtt.io/devices/WXKG01LM.html#xiaomi-wxkg01lm. The button didn't report battery level (blinking red battery icon in the UI), but after several repairings I managed to fix that.
I'll leave it like that for a day or two (to make sure there are no random disconnects) and let you know the result.

Thanks!

alexberdnik, 2022-08-30:
The button is not reacting to clicks after an hour. I guess this should be reported as a separate issue.

alexberdnik, 2022-09-04:
The issue with button persists after re-pairing. However the joining issue seems to be resolved.

closed 2022-09-04 by alexberdnik

evelant, 2022-11-21:
FWIW I just got a zbdongle-e, ensured it was on latest firmware, and tried to use it with zigbee2mqtt edge addon in home assistant. It crashes with the same route error when trying to pair some devices so I think this is not resolved.

patchpump, 2022-11-25:
z2m dev version 1.28.2, dongle firmware 6.10.3.0 build 297 successfully pairs with TuYa Door sensor (TS0203) and works well for a few minutes. Crash with Error: No route when device leaves and joins: [log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10094822/log.txt)