Koenkk/zigbee2mqtt #13609: TS0505b

Prince3k, 2022-08-19:
### Link

https://shop.retoura.de/Livarno-Home-LED-Bodenlichterkette-5er-dimmbar-Zigbee-Smart-Home-Kugel-B-Ware-einwandfrei?utm_source=googleads&gclid=CjwKCAjw6fyXBhBgEiwAhhiZssqRQonGbaeCQ-0TZ17vpaC_oN0MADigw54s42MEN4-H8NR70SPXNhoCRycQAvD_BwE

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0x540f57fffe975009","nwkAddr":28783,"manufId":4098,"manufName":"_TZ3000_bwlvyjwk","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3000_bwlvyjwk","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1660932986962,"defaultSendRequestWhen":"immediate"}

### Comments

`
[image]
`

### Supported color modes

16Mio Color + different white colors

Prince3k, 2022-08-19:
`<html><body>

2022-08-19 19:56:52.189  - info: zigbee.0 (2235) Zigbee: allowing new devices to join.
--
2022-08-19 19:57:53.455  - info: zigbee.0 (2235) Zigbee: stop joining
2022-08-19 20:00:33.828  - info: zigbee.0 (2235) Zigbee: allowing new devices to join.
2022-08-19 20:01:16.020  - info: zigbee.0 (2235) Starting interview of '0x540f57fffe975009'
2022-08-19 20:01:16.129  - warn: zigbee.0 (2235) Device '0x540f57fffe975009' announced itself
2022-08-19 20:01:18.771  - info: zigbee.0 (2235) Zigbee: allowing new devices to join.
2022-08-19 20:01:18.772  - info: zigbee.0 (2235) Joining already permitted
2022-08-19 20:01:26.743  - info: zigbee.0 (2235) Successfully interviewed '0x540f57fffe975009', device has succesfully been paired
2022-08-19 20:01:26.786  - warn:  zigbee.0 (2235) download icon from  https://www.zigbee2mqtt.io/images/devices/TS0505B.jpg saved into  /opt/iobroker/node_modules/iobroker.zigbee/admin/img/TS0505B.png
2022-08-19 20:01:26.938  - warn:  zigbee.0 (2235) Object 540f57fffe975009.effect is invalid:  obj.common.states has an invalid type! Expected "object", received  "string"
2022-08-19 20:01:26.939  - warn: zigbee.0 (2235) This object will not be created in future versions. Please report this to the developer.
2022-08-19 20:01:26.960  - warn:  zigbee.0 (2235) Object zigbee.0.540f57fffe975009.effect is invalid:  obj.common.states has an invalid type! Expected "object", received  "string"
2022-08-19 20:01:26.961  - warn: zigbee.0 (2235) This object will not be created in future versions. Please report this to the developer.
2022-08-19 20:01:35.167  - info: zigbee.0 (2235) Zigbee: stop joining
2022-08-19 20:01:37.456  - info: zigbee.0 (2235) Zigbee: allowing new devices to join.
2022-08-19 20:02:38.719  - info: zigbee.0 (2235) Zigbee: stop joining

</body>
</html>`

Koenkk, 2022-08-19:
`_TZ3000_bwlvyjwk` is already supported in the dev branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-19 by Koenkk