Koenkk/zigbee2mqtt #13612: Confusing dashbord items Blitzwolf TS11F_plug_3

dykst002, 2022-08-19:
### What happened?

[image]

### What did you expect to happen?

See my picture

### How to reproduce it (minimal and precise)

The shows the solution

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

2022

### Adapter

Sonoff zigbee 3.0

### Debug log

No logging

dykst002, 2022-08-19:
Sorry did not know the firmware version of my sonoff stick. I found it “20210708” so an old version. I think this nothing to do with my issue

Koenkk, 2022-08-20:
Please create issues related to the frontend here: https://github.com/nurikk/zigbee2mqtt-frontend/issues

dykst002, 2022-08-20:
OK

closed 2022-08-21 by Koenkk