Koenkk/zigbee2mqtt #13613: Xiaomi GZCGQ01LM iluminance sensor only lasts 1 week

kbfifi, 2022-08-19:
### What happened?

I've replaced the battery of this sensor 3 times. Everytime the battery drains down in 1 week.  
zigbee2mqtt: 
- version 1.27.0
GZCGQ01LM:
- Sensor version: 2019www.
- Build date: 20191118
- 

### What did you expect to happen?

The sensor to work for a few months at least

### How to reproduce it (minimal and precise)

Just replace battery, pair and leave everything default

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 plus

kbfifi, 2022-09-08:
Anyone other experience?

closed 2022-10-16 by github-actions[bot]

m42j, 2022-11-20:
I'm kind of seeing the same thing but I'm not replacing any battery, the battery seems to recover. But on days with dips I'm seeing a lot of reporting so I'm going to try changing the min reporting settings. 
[image]