Koenkk/zigbee2mqtt #13620: SASWELL, Zigbee smart water valve

pganansia, 2022-08-20:
### What happened?

I bought a SASWELL, Zigbee smart water valve. I found in the supported devices that it is the same than RTX ZVG1.
The integration wit Home Assistant is working well.
I found In the Zigbee2Mqtt Frontend that a warning is fired : 
"zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}"

### What did you expect to happen?

I expected that the case DP #6 was handled by the converters fromZigbee.js.
I think that there is a special DP for the SASWELL comparing to RTX/ZVG1, but I d'on't how to modify the converter to avoid the warning.
Model Zigbee is TS0601
Manufacturer is _TZE200_81isopgh

I have also noticed that the Model Zigbee and the manufacturer are handled in the devices file rtx.js.

### How to reproduce it (minimal and precise)

Just switch on the valve and then switch off.
The warning is fired 2 or three time when switching off.

### Zigbee2MQTT version

1.27.0 commit: a9b8808

### Adapter firmware version

20211217

### Adapter

zStack3x0 Sonoff

### Debug log

Warning 2022-08-20 11:11:45zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Warning 2022-08-20 11:12:02zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}

Koenkk, 2022-08-21:
To find out what this dp means, please follow: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

pganansia, 2022-09-01:
Thanks for the answer and for the link.
Unfortunatly, the procedure is too technical for me and I don't a Tuya Bridge.
Don't worry, probably some one more technical than me will have the problem and will be able to apply the procedure.

closed 2022-09-01 by pganansia