Koenkk/zigbee2mqtt #13623: Tuya 2CH neutral switch _TZ3000_fisb3ajo wrongly discovered as TS0002

ArJay60, 2022-08-20:
### What happened?

Tuya 2CH neutral switch _TZ3000_fisb3ajo is discovered as TS0002

### What did you expect to happen?

Should be discovered as:
https://www.zigbee2mqtt.io/devices/QS-Zigbee-S04-2C-LN.html

### How to reproduce it (minimal and precise)

Discovered as such once paired

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

20201026

### Adapter

zzh! stick / zStack3x0

Koenkk, 2022-08-21:
It should currently be discovered as [`TS0002_switch_module_2`](https://www.zigbee2mqtt.io/devices/TS0002_switch_module_2.html), is that not the case?

ArJay60, 2022-08-21:
Yes. Discovered as such.

Koenkk, 2022-08-21:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0003', manufacturerName: '_TYZB01_zsl6z0pw'}, {modelID: 'TS0002', manufacturerName: '_TZ3000_fisb3ajo'}],
    model: 'QS-Zigbee-S04-2C-LN',
    vendor: 'Lonsonho',
    description: '2 gang switch module with neutral wire',
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    toZigbee: [tz.TYZB01_on_off],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

ArJay60, 2022-08-21:
Tried this, but doesn't solve it. Result is the same.

Koenkk, 2022-08-21:
Did you add it to the z2m configuration.yaml? It must show up as `QS-Zigbee-S04-2C-LN` in the frontend.

ArJay60, 2022-08-21:
It does now. I am using Hassos and changed configuration.yaml manually with a copy-paste of the external_converters part from the mail send as reaction on your comment in github. And then restarted. This didn't work (probably some strange/hidden characters in the string copied??). I now changed configuration.yaml through the frontend and restarted. It now shows QS-Zigbee-S04-2C-LN.

Koenkk, 2022-08-22:
Great, can you control it? (via the exposes tab on the frontend)

ArJay60, 2022-08-22:
After pairing I could control it once. After that error 205.
 failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Tuinkamer Fan"},"type":"zigbee_publish_error"}'.

After reset and repair the same.

Koenkk, 2022-08-22:
When stopping z2m, removing the ext converter and starting z2m, does it work again?

ArJay60, 2022-08-22:
The same. Works briefly after removing power and adding power the switch device then error 205.

Koenkk, 2022-08-23:
Can you try pairing the device close to the coordinator and see if it keeps working?

ArJay60, 2022-08-23:
Then it does work.

Koenkk, 2022-08-23:
Then I expect something goes wrong with one of your router. To debug this further I would need to have a sniff off:
- Pair the device
- control it (OK)
- wait some time
- control it (failure)

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

ArJay60, 2022-08-29:
Hi. Had a sniffer prepared but after removal of the device and pairing it again on the spot where it should be working everything is working fine right now. In the error situation I paired it close to the coordinator and then moved it to its location where it started having problems. I will investigate if problems re-occur. For now this item can be closed.

closed 2022-08-29 by Koenkk

Alex2zl, 2023-01-11:
Hello. I have the same problem. There are 2 of the same relays from Lonsonho, which are not defined correctly in z2mqtt (just like it was here TS0002). And the worst thing is that after a while (usually every other day), it is these devices that stop responding to commands, they cannot be controlled, an error appears: No network route' (205). After a while, the connection itself can be restored again and they can be controlled again, but then it is lost again, and so on in a circle.
With an external adapter, devices are detected correctly as lonsohno, but this does not solve the No network route (205) error.
The devices are not far from the coordinator (SONOFF ZIGBEE 3.0 USB DONGLE PLUS), there are other devices in the room (including routers) that work correctly.
What can I do to try and fix this error?