Koenkk/zigbee2mqtt #13629: HA Status topic defaults to "hass/status" vs. "homeassistant/status"

ptr727, 2022-08-20:
### What happened?

Birth and will messages for HA did not seem to work.

Code:  
https://github.com/Koenkk/zigbee2mqtt/blob/a9b8808237554316c5c97e9d9c990048733779e8/lib/util/settings.ts#L121

Docs:  
https://www.zigbee2mqtt.io/guide/configuration/homeassistant.html#advanced-configuration

Config defaults to "hass/status" per code.  
Docs say default should be "homeassistant/status".

HA [docs](https://www.home-assistant.io/docs/mqtt/birth_will/) say to use the MQTT GUI config, and default is "homeassistant/status".

Must manually change config else HA reports entities offline when restarting.

### How to reproduce it (minimal and precise)

Use simple HA integration:

```
homeassistant: true
```

Open advanced settings UI, UI reports the HA status topic is "hass".

Change HA config to advanced:

```
homeassistant:
  discovery_topic: "homeassistant"
  status_topic: "homeassistant/status"
  legacy_entity_attributes: false
  legacy_triggers: false
```

Settings UI now matches documentation.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

tubez

Koenkk, 2022-08-21:
Z2M always matches on both `homeassistant/status` and the configured `status_topic`:

- https://github.com/Koenkk/zigbee2mqtt/blob/e0a9323d6efd0bee5547e537588010aece93e0c9/lib/extension/homeassistant.ts#L27
- https://github.com/Koenkk/zigbee2mqtt/blob/e0a9323d6efd0bee5547e537588010aece93e0c9/lib/extension/homeassistant.ts#L1340

ptr727, 2022-08-22:
I see, so this was not the cause of the problem I observed.
But would you agree the default shown in UI is different to the documentation stated default?

Koenkk, 2022-08-22:
> But would you agree the default shown in UI is different to the documentation stated default?

Yes, I've updated the docs!

closed 2022-08-22 by ptr727