Koenkk/zigbee2mqtt #13641: Xiaomi RTCGQ11LM Precision options not working

sjchickey, 2022-08-22:
### What happened?

None of the Xiaomi RTCGQ11LM (Motion Sensors) honour the precision for temp, lux or illuminance. Regardless of the settings in these option nothing other than whole numbers is reported.

### What did you expect to happen?

setting temperature_precision = 2 would result in a value such as 20.21 reported

### How to reproduce it (minimal and precise)

Adjust any precision settings with no effect

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20210708

### Adapter

sonoff zigbee 3.0 usb dongle

Koenkk, 2022-08-22:
The RTCGQ11LM reports the `device_temperature` in whole numbers, same goes for the illuminance. Javscript drops the decimals if they are `00` so you won't see them.

sjchickey, 2022-08-22:
I can understand that when it reports a .00 but I've had nothing but whole numbers for over 6 months, a decimal number has never been reported with any setting.

sjchickey, 2022-08-22:
If this device can't support it then then the options need to be updated so isn't available to be set and the documentation updated to reflect this isn't supported.

Koenkk, 2022-08-22:
>I can understand that when it reports a .00 but I've had nothing but whole numbers for over 6 months, a decimal number has never been reported with any setting.

The reported number is an `int` (no decimals), I've removed these options now. Docs will be updated with the next release

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-22 by Koenkk