Koenkk/zigbee2mqtt #13644: Ikea Motion sensor "Illuminance_above_threshold" is never true

JasonBSteele, 2022-08-22:
### What happened?

I have an Ikea Tradfri Motion Sensor as described here: https://www.zigbee2mqtt.io/devices/E1525_E1745.html 

One of the values reported is "illuminace_above_threshold" and is described as "Indicates whether the device detected bright light (works only in night mode)."

I definitely have night mode enabled (this is a phyisical push button switch on the device), however no matter how bright it is this value is always false - even in full sunlight,

### What did you expect to happen?

In night mode I would expect  "illuminace_above_threshold" to return true when the device is in bright light.

### How to reproduce it (minimal and precise)

Pair the device. 
Put it in Night Mode if necessary (this is the default).
Place in brightly lit room
Trigger it with movement.
Read the value of "illuminace_above_threshold" (it should be true)

### Zigbee2MQTT version

 1.27.0 commit: e0a9323d

### Adapter firmware version

20190608

### Adapter

CC2561

JasonBSteele, 2022-09-22:
I hope "stale" isn't meant to indicate that this may no longer be an issue, because it is :)

closed 2022-10-31 by github-actions[bot]

ebkraus, 2023-02-12:
I have the same problem, is there any solution for this?

JasonBSteele, 2023-02-14:
None that I have found :(

duceduc, 2023-07-09:
Yes, I have noticed this binary sensor in the past but never used it until now. I have use for it, but the state always stays in the off. Though I have read somewhere that this only works connected to the ikea hub. If that is the case, why have this binary sensor documented in the zigbee1mqtt docs?

Krazubu, 2023-08-16:
Same problem here. This worked fine in the past but it doesn't anymore for unknown reason.
However I changed from CC2521 to SONOFF ZBDongle-E, I can't really tell if the problem started at this time but it's not impossible.

watermarkhu, 2023-12-29:
Changing the sensor to night mode (click the day/night switch on the sensor) enables the reporting of `Illuminance_above_threshold`.