Koenkk/zigbee2mqtt #13645: IH012-RT01 TuYa PIR not clearing after timeout

smartroad, 2022-08-22:
### What happened?

Occasionally the IH012-RT01 TuYa PIR sensor I am using will not clear and reports occupancy as "detected". It will stay that way until the PIR is activated again, which most of the time will transmit occupancy (LED flashes) and then clears after timeout. As shown below the PIR was shown activated for nearly 2.5 hours and only reset as my partner got up in the night and walked by it making it cycle again. To the left you can also see another large red block of about 30 minutes.

[image]

### What did you expect to happen?

After timeout it should reset to "Clear".

Is there anyway to make Zigbee2MQTT assume it is clear after a given timeout?

### How to reproduce it (minimal and precise)

Hard to say as it happens erratically.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

smartroad, 2022-08-22:
Just to clarify, the PIR is resetting itself (because it flashes its LED after it's internal timeout when it detects more motion), it just seems like Zigbee2MQTT/Home Assistant do not notice.

closed 2022-09-29 by github-actions[bot]

WildeRNS, 2022-10-15:
Same issue...

tchlyah, 2023-01-10:
I have exact same issue. I have to reboot the devices each time to make them work, for few hours only. 

Hi @smartroad, @WildeRNS, did you find any fix or workaround ?

smartroad, 2023-01-10:
> I have exact same issue. I have to reboot the devices each time to make them work, for few hours only.
> 
> Hi @smartroad, @WildeRNS, did you find any fix or workaround ?

I did, but it was to buy another PIR 😔 

[image]

smartroad, 2023-01-10:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 7 days

Hadn't noticed this, how is this stale when we were waiting for a response, how would I keep it "alive" just do a pointless "bump" post?

Dash, 2023-08-17:
Found this issue after experiencing similar problems.  Though I removed the units from the device list, and then held the reset button on the device for 10 seconds to trigger a fresh join.  This _seems_ to have resolved it.  Posting here just in case somebody has a similar problem.