Koenkk/zigbee2mqtt #13649: Woox R7060 battery % don’t shown

acaonweb, 2022-08-22:
### What happened?

https://www.zigbee2mqtt.io/devices/R7060.html#woox-r7060
battery isn’t exposed, just “?” Char
Last firmware into zzh coordinator, last Zigbee2mqtt version running on raspbian buster

### What did you expect to happen?

See battery level 

### How to reproduce it (minimal and precise)

Installed as instructions says

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

Zzh!

acaonweb, 2022-08-23:
ciao, 
i've googled in the meantime and i've created this ext converter

`pi@zigbee2mqtt:/opt/zigbee2mqtt/data $ cat ext_converter.js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    woox_R7060: {
        cluster: 'manuSpecificTuya',
        type: ['commandActiveStatusReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'woox_R7060');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);

            switch (dp) {
            case tuya.dataPoints.wooxSwitch:
                return {state: value === 2 ? 'OFF' : 'ON'};
            case 101:
                return {battery: value};
            default:
                meta.logger.warn(`zigbee-herdsman-converters:WooxR7060: Unrecognized DP #${
                    dp} with data ${JSON.stringify(dpValue)}`);
            }
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS0101', manufacturerName: '_TZ3210_eymunffl'}],
    model: 'R7060',
    vendor: 'Woox',
    description: 'Smart garden irrigation control',
    fromZigbee: [fz.on_off, fz.ignore_tuya_set_time, fz.ignore_basic_report, fzLocal.woox_R7060, fz.battery],
    toZigbee: [tz.on_off],
    onEvent: tuya.onEventSetTime,
    exposes: [e.switch(), e.battery(), e.battery_voltage()],
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;
`
i'm testing now, let me know if i procede with this or yours, i'll wait your response :)

acaonweb, 2022-08-23:
ok, in the meantime i've checked my converter and now i can have percentuage of battery but not voltage.
[image]

Koenkk, 2022-08-23:
Great then it works, the device does not support voltage so I've removed that.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-23 by Koenkk

acaonweb, 2022-08-23:
ok so, if i want wait the official release, i stay with the external converter, right? and when it will be in the official one i can remove it?
when the official one will be out?

acaonweb, 2022-08-23:
[image]

Koenkk, 2022-08-23:
Yes you can use the external converter until the next release. Next release is on 1 September (every 1st of the month)

acaonweb, 2022-08-24:
Than I can remove external, right?

Koenkk, 2022-08-24:
Yes!