Koenkk/zigbee2mqtt #13651: Eglo Fueva5-Z mis-identified as Awox 33943_33944_33946

MrFouFou, 2022-08-23:
### What happened?

Wrong recognition of this light recognise as Awox 33943_33944_33946
Light is EGLO 33991, brought in France in "Le Roy Merlin" store. It's probaly a rebrand or custom made.
Not working as expected: 
ON/OFF state is broken trough the frontend, can't toggle it.
Got 4-5 colors with RGB

### What did you expect to happen?

Working good.

### How to reproduce it (minimal and precise)

Pair it and toggle ON/OFF

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Debug log

{"id":30,"type":"Router","ieeeAddr":"0xa4c1389d575c4145","nwkAddr":28095,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx",
"epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],
"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},
"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorMode":2,"currentX":0,"currentY":0,"enhancedCurrentHue":256,"currentSaturation":77,"colorTemperature":268}},
"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":8}}},"binds":[],"configuredReportings":[],
"meta":{"scenes":{"0_0":{"name":"Scene 0","state":{"color_mode":"color_temp","state":"ON","brightness":86,"color_temp":370,
"color":{"x":0.4599,"y":0.4106,"hue":32,"saturation":82,"h":32,"s":82}}}}}},"3":
{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],
"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,
"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,
"meta":{"configured":88764544},"lastSeen":1661241744395,"defaultSendRequestWhen":"immediate"}

MrFouFou, 2022-08-23:
Similar issue as: https://github.com/Koenkk/zigbee2mqtt/issues/11915

Koenkk, 2022-08-23:
This looks to be a whitelabel indeed (`,"manufacturerName":"AwoX"`) 

What is logged in the debug log when trying to turn on/off?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

MrFouFou, 2022-08-24:
Hi Koenkk,

This is the log when I toggle the light on the frontend:
[image]
Light is not working in this case (won't toggle state)

This is the log when I force the state (with On or OFF) on the frontend:
[image]
Light is working.

Many thanks for your work !

Koenkk, 2022-08-24:
> This is the log when I toggle the light on the frontend:

Are you clicking the refresh icon? This will not toggle the state but read the state from the device. You can click the on/off button to toggle it:

[image]

MrFouFou, 2022-08-24:
Sorry for this bad explanation; what I mean is the state is always off, even when the light is on. If I'm clicking the refesh icon, it does not update the state, it's always off.

Koenkk, 2022-08-25:
So if I understand correctly, click on ON will turn the bulb ON but the state stays OFF? If yes, can you provide me the debug log of this?

MrFouFou, 2022-09-04:
Yes exactly.
Debug is: 
[image]

Koenkk, 2022-09-06:
Can you provide me your `configuration.yaml` file? (without password/usernames in it)

closed 2022-10-15 by github-actions[bot]