Koenkk/zigbee2mqtt #13656: TuYa TS0012 turns on both loads when only one switch is pressed.

Hull9k, 2022-08-23:
### What happened?

TuYa TS0012 Smart light switch - 2 gang
When I click on one of the two switches on the dashboard (State Left or State Right) that refer to that switch, both loads turn on, not just one. Similarly, then clicking on one of the switches turns off both loads at the same time.

### What did you expect to happen?

Each of the switches (Left or Right) will only control its own load.

### How to reproduce it (minimal and precise)

It worked correctly before. This problem arose when I connected the switches to the Smart Life app through the Zigbee gateway which I received the other day and the application updated the firmware in the switches to version 1.1.0.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

zStack3x0, SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9410328/log.txt)

Hull9k, 2022-08-23:
Found a solution here:
https://community.home-assistant.io/t/zigbee2mqtt-moes-switch-problem/451248

Koenkk, 2022-08-24:
Thanks for the link, integrated the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-24 by Koenkk

fvillena, 2022-10-18:
I think my device is TS0012_switch_module, so I created a new issue #14541

fjpg1970, 2023-01-19:
Tuya TS0012 double light switch: cannot toggle one switch only both from,
The same thing happens to me, any solution for newbies in zigbee2mqtt?