Koenkk/zigbee2mqtt #13658: TuYa TS0601 smart air house keeper payload spam and counter roll-over

meeb, 2022-08-24:
### What happened?

I have a TuYa TS0601 smart air house keeper which identifies as `_TZE200_dwcarsat`. The PM2.5 (the `pm25` value in the message) counter rolls over a couple of times a day and makes the data useless as readings spike from the usual range (5-10 ish) to 43,521 then back down again. This may be related to the high number of updates the device outputs (multiple times per second).

Here's an example standard status message from the device

```
Zigbee2MQTT:info  2022-08-24 13:07:55: MQTT publish: topic 'zigbee2mqtt/air_sensor', payload '{"co2":360,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"air_sensor","hardwareVersion":1,"ieeeAddr":"0xa4c138997a43a811","manufacturerID":4417,"manufacturerName":"_TZE200_dwcarsat","model":"TS0601_smart_air_house_keeper","networkAddress":52364,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"formaldehyd":5,"humidity":59.1,"linkquality":94,"pm25":6,"temperature":19.7,"voc":15}'
```

Here's a message from the device when spiking:

```
Zigbee2MQTT:info  2022-08-24 01:04:47: MQTT publish: topic 'zigbee2mqtt/air_sensor', payload '{"co2":358,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"air_sensor","hardwareVersion":1,"ieeeAddr":"0xa4c138997a43a811","manufacturerID":4417,"manufacturerName":"_TZE200_dwcarsat","model":"TS0601_smart_air_house_keeper","networkAddress":52364,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"formaldehyd":2,"humidity":56,"linkquality":51,"pm25":43521,"temperature":20.6,"voc":13}'
```

Attached is an example of the data the device records in the Home Assistant graph showing the spikes.

[image]

I'm not entirely sure if this is the device outputting the errant value or if this is some counter in zigbee2mqtt parsing the value incorrectly.

### What did you expect to happen?

The `pm25` value to not spike irregularly to 43,521.

### How to reproduce it (minimal and precise)

Purchase a `_TZE200_dwcarsat` device and link it to zigbee2mqtt.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0

Payou6994, 2022-09-23:
hello, 
I have the same problem. with the same setup. 
[image]

meeb, 2022-10-24:
Do you need any additional info from me to assist with this? I can just submit a PR that ignores readings for this device that have a value of 43521 if that would be helpful?

closed 2022-12-03 by github-actions[bot]