Koenkk/zigbee2mqtt #13679: [New device support]: New version of Philips Hue Iris copper limited edition (generation 4)

3mta3, 2022-08-25:
### Link

https://www.philips-hue.com/en-us/p/hue-white-and-color-ambiance-iris-copper-special-edition/046677577179

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x0017880109c32fd4","nwkAddr":34843,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"929002401201","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"zclVersion":2,"dateCode":"20211210","swBuildId":"1.93.7"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20211210","swBuildId":"1.93.7","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1661412114350,"defaultSendRequestWhen":"immediate"}

### Comments

I added new model to devices/philips.js and it is working:

```
    {
        zigbeeModel: ['929002401201'],
        model: '929002401201',
        vendor: 'Philips',
        description: 'Hue Iris copper limited edition (generation 4) ',
        extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    },
```
Lamp image from product page:

[image]

Koenkk, 2022-08-25:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-25 by Koenkk

3mta3, 2022-08-28:
fyi added [pull request #1534](https://github.com/Koenkk/zigbee2mqtt.io/pull/1534) with supporting doc/image on zigbee2mqtt.io.