Koenkk/zigbee2mqtt #13682: MoesGo EDM-1ZBB-EU reported "Received unexpected Tuya DataPoint"

artrmnov, 2022-08-25:
### What happened?

Device is `MoesGo EDM-1ZBB-EU.`
Url: https://www.moeshouse.com/en-gb/products/new-zigbee-smart-touch-light-dimmer-switch-timer-brightness-memory-smart-life-tuya-app-remote-control-work-with-alexa-google?_pos=11&_sid=64e9f8bed&_ss=r

identified by Z2M as:
```
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_w4cryh2i
```

Warnings:
```
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,228]}}': type='commandDataResponse', datatype='value', value='740', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,43]}}': type='commandDataReport', datatype='value', value='43', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
Warning zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x60a423fffe61bfe5 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,75]}}': type='commandDataReport', datatype='value', value='331', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
```

Solved problem with another manufacturer: https://github.com/Koenkk/zigbee2mqtt/issues/12313

Any help please

And please add device to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L982-L991

### What did you expect to happen?

Errors will gone and devices will behave as it should.

### How to reproduce it (minimal and precise)

Via UI. - switch on/off. Change brightness level.

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

20210708

### Adapter

CC2652P

Koenkk, 2022-08-25:
First we need to know what this data points means, to figure that out you can follow: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

artrmnov, 2022-08-26:
I don't have Tuya Zigbee Gateway

But if you override `meta.device.manufacturerName`, then the errors disappear and the device works correctly

`manufacturerName` taken from previous issue

```
fromZigbee: [
    {
        ...fz.tuya_dimmer,
        convert: (model, msg, publish, options, meta) => {
            meta.device.manufacturerName = '_TZE200_dfxkcots';
            
            return fz.tuya_dimmer.convert(model, msg, publish, options, meta);
        }
    },
    fz.ignore_basic_report
],
```

Koenkk, 2022-08-26:
Great, integrated the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)