Koenkk/zigbee2mqtt #13685: Xaomi WXKG11LM Switches trigger without

isujosh, 2022-08-25:
### What happened?

I have installed two Xaomi WXKG11LM switches and setup a simple automation to toggle two lights on/off with a single button press. This works fine. However, if I leave the switches alone - they both trigger 50 minutes later - without any button press. 

The logs clearly show the button presses I made but then show that Zigbee2MQTT received a message 50 minutes later to switch the units again - without any intervention by me.

Curiously, this appears to be 3000 seconds after the first press - not sure if that is significant. 

### What did you expect to happen?

I expected that there would be no change to the light status without a press of the switch

### How to reproduce it (minimal and precise)

Setup HA automation to set button press to toggle light on/off.

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

26780700

### Adapter

Conbee II

### Debug log

[First Button Press.log](https://github.com/Koenkk/zigbee2mqtt/files/9425034/First.Button.Press.log)
[Phantom Button Press.log](https://github.com/Koenkk/zigbee2mqtt/files/9425036/Phantom.Button.Press.log)

Koenkk, 2022-08-25:
I don't see any actions being published  in `Phantom Button Press.log`. e.g. 

```
Zigbee2MQTT:info  2022-08-25 12:39:23: MQTT publish: topic 'zigbee2mqtt/Switch 2', payload '{"action":null,"battery":100,"device_temperature":30,"linkquality":255,"power_outage_count":12,"voltage":3225}'
```

You see that `action` is `null` which means that no `action` has been triggered.

isujosh, 2022-08-25:
I can see that - but why is the light ("Living Room Colour 2") state being changed to "ON"? 

Something is clearly triggering it?

isujosh, 2022-08-25:
This issue seems to be like `https://github.com/Koenkk/zigbee2mqtt/issues/12018` - (and Issue #1967).

There is a comment there about the 50 minute cycle and the device sending a '**genBasic**' message to itself. `https://github.com/Koenkk/zigbee2mqtt/issues/12018#issuecomment-1093799243`

This sounds like a very similar issue to me

Koenkk, 2022-08-26:
> I can see that - but why then is the light ("Living Room Colour 2") state being changed to "ON" off the back of that 'null' action?

Because you probably have an automation which triggers on this. See https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#responding-to-button-clicks how to properly setup your automation.

isujosh, 2022-08-26:
> Because you probably have an automation which triggers on this. See https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#responding-to-button-clicks how to properly setup your automation.

No, I don't have any other automation triggering this light.

As a test, I amended [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters)/[devices](https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/devices)/xiaomi.js as follows:

`{
        zigbeeModel: ['lumi.sensor_switch.aq2', 'lumi.remote.b1acn01'],
        model: 'WXKG11LM',
        vendor: 'Xiaomi',
        description: 'Aqara wireless switch',
        meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
        exposes: [e.battery(), e.battery_voltage(), e.action(['single', 'double', 'triple', 'quadruple', 'hold', 'release']),
            e.device_temperature(), e.power_outage_count()],
        fromZigbee: [fz.xiaomi_multistate_action, fz.xiaomi_WXKG11LM_action, fz.ignore_basic_report,
            fz.legacy.WXKG11LM_click, fz.legacy.xiaomi_action_click_multistate],
        toZigbee: [],
    },`

I removed: **fz.xiaomi_basic** and added: **fz.ignore_basic_report**

and this solved the issue completely. The button actions still work and of course, if I had any other automations triggering these lights then this change obviously wouldn't have made any difference because it is specific to only one device: WXKG11LM.

For whatever reason, the WXKG11LM genBasic status update triggers the automation despite the fact that the action is set to **"null"**

Koenkk, 2022-08-26:
> For whatever reason, the WXKG11LM genBasic status update triggers the automation despite the fact that the action is set to "null"

That is a problem in your automation, not in zigbee2mqtt. Your automation should not do anything when action null is published. https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended shows how to properly set up an automation which does not trigger on `action: null`.

isujosh, 2022-09-02:
> That is a problem in your automation, not in zigbee2mqtt. Your automation should not do anything when action null is published. [...]

I'll revisit that to see where this is going wrong. Thanks for the pointer

closed 2022-10-11 by github-actions[bot]