Koenkk/zigbee2mqtt #13688: [New device support]: Zigbee smart energy TS0601

hovbjo99, 2022-08-25:
### Link

TS0610 Zigbee smart meter

### Database entry

.

### Comments

Is it possible to get the modelID: manufature name: _TZE200_bkkmqmyo to this Device?

part form a file i do some debugging on....

const definition = {
    
        {modelID: 'TS0601', manufacturerName: '_TZE200_bkkmqmyo'},		// added new fingerprint
    ],
    model: 'TS0601_din',
    vendor: 'TuYa',
    description: 'Zigbee smart energy meter DDS238-2 Zigbee',
    fromZigbee: [fz.tuya_dinrail_switch],

jrosier, 2022-08-31:
I used the Tuya ioT platform to find the data points. Now, they have to be mapped into converters.
1: total forward energy
6: Phase B (empty data in my device)
9: fault
12: clear energy
16: state (reading of the state of the on/off relay)
17: alarm set 1 
18: alarm set 2
101: total active energy (I think they mean the balance of forward and reverse energy)
102: reverse energy
103: power
105: frequency
108: breaker (maybe you can activate the relay with this dp?)
109: reactive energy (KVArh)
110: reactive power (kVAr)
111: power factor

The strange thing so far: I couldn't find values for voltage and current.

dirkjanfaber, 2022-09-29:
This issue should be closed as it is picked up in #13767

closed 2022-09-29 by Koenkk