Koenkk/zigbee2mqtt #13693: [New device support]: Climax technology door sensor

jerome-angeloni, 2022-08-25:
### Link

https://www.climax.com.tw/new/dc16sl-zb.php

### Database entry

{"id":39,"type":"EndDevice","ieeeAddr":"0x00124b0008b7282e","nwkAddr":47016,"manufId":0,"manufName":"ClimaxTechnology","powerSource":"Battery","modelId":"DC_00.00.03.07TC","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"DC_00.00.03.07TC","manufacturerName":"ClimaxTechnology","powerSource":3,"zclVersion":1,"appVersion":16,"hwVersion":0,"dateCode":"20160902        "}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00238db3b2","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"hwVersion":0,"dateCode":"20160902        ","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1661452917029,"defaultSendRequestWhen":"immediate"}

### Comments

I'm using Z2M addon v1.27.0-1.
I've created a DC_00.00.03.07TC.js file for my device and I've placed it in Share/zigbee2mqtt.
In the add-on, I edit the configuration.yaml file and add:
external_converters:
  - DC_00.00.03.07TC.js
I restart, but it seems Z2M completely ignores my file. Even if the file contains a big syntax error on first line, I don't get any complaint in the logs.
When I play with the device, I just get :
"Zigbee2MQTT:warn  2022-08-25 22:08:15: Received message from unsupported device with Zigbee model 'DC_00.00.03.07TC' and manufacturer name 'ClimaxTechnology'
Zigbee2MQTT:warn  2022-08-25 22:08:15: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html"

I've spent 2 evenings trying to understand how to make it work but I'm stuck. Do I miss something obvious?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	zigbeeModel: ['DC_00.00.03.07TC'],
	model: 'DC_00.00.03.07TC',
	vendor: 'ClimaxTechnology',
	description: 'contact de porte ZigBee qui detecte et signale ouverture-fermeture des portes et des fenetres',
	fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
	toZigbee: [],
	exposes: [e.contact(), e.battery_low(), e.tamper()],
};

module.exports = definition;
```

Koenkk, 2022-08-26:
> In the add-on, I edit the configuration.yaml file and add:

Are you sure you added it in the z2m configuration.yaml file and not the Home Assistant one?

jerome-angeloni, 2022-08-26:
Hello,

I go through HA settings > Addons > zigbee2mqtt > configuration > then click on the 3 dots to edit yaml. There I can add or see my: 
external_converters:
  - DC_00.00.03.07TC.js
Then save, restart add-on.

But what I see in the yaml file in the addon is not what I see when I open the file in share/zigbee2mqtt/configuration.yaml. I have a disconnect somewhere. probably something very basic, but I can't get it.

Koenkk, 2022-08-26:
Can you try adding it directly in the `configuration.yaml` file?

jerome-angeloni, 2022-08-31:
Bingo! It was as simple as that. I've copied the yaml from the addon UI in the configuration.yaml file, restarted, then it was working. Thanks for pointing to this.

closed 2022-08-31 by jerome-angeloni