Koenkk/zigbee2mqtt #13694: Xiaomi Aqara sensor battery drain

wasim7715, 2022-08-25:
### What happened?

Xiaomi Aqara sensor battery drain  lot faster then before with release 1.27.1

### Zigbee2MQTT version

1.27.1

### Adapter firmware version

 20220726

### Adapter

cc2652p

Shred99, 2022-08-26:
Which sensor is it?

I have around 28 Aqara sensors (various types) across two sites.  I haven't seen an increase in battery consumption since upgrading.  The time that power consumption is highest is during transmit.  Generally the sensors determine how frequently they transmit an update and the Zigbee2MQTT controller/coordinator has no say in it.

groenmarsmannetje, 2022-08-26:
Be aware that in latest release they have changed the method to calculate the battery percentage. There is a lot discussion about that, because it doesn't seem to reflect a realistic value. Earlier the problem was that the percentage dropped from almost 100% to 0% or devices that had empty batteries while just before that it reported high percentage which was also not correct.

So they are trying to improve the calculation which we can expect in next months release.

Anka001, 2022-08-26:
Maybe the previous way to calculate battery percentage was not perfect, but the current one is worse. All aqara sensors using CR2032s that displayed <40% battery before updating now show 0%, while still running fine. 
I replaced one battery in a temperature sensor a week ago, and battery status is already down to 59%. I expect it to be down to 0% in another week, and run for 3 more years.

Koenkk, 2022-08-26:
This is likely because the battery % calculation changed, it will be changed again with the next release (on 1 September).

Anka001, 2022-08-26:
Not a big issue for me. But at least some guys -or gals?- seem to think, zigbee2mqtt has changed the behaviour of their sensors.

Koenkk, 2022-08-26:
@Anka001 yes, so this will be addressed in the next release.

wasim7715, 2022-08-26:
![Capture](https://user-images.githubusercontent.com/63366326/186998687-00dda6a3-0264-44b2-8d6a-c342d624b561.PNG

WSDCGQ11LM   Tem Sensor
RTCGQ14LM    Aqara E1     motion sensor
MCCGQ14LM  Aqara E1 door & window contact sensor
battery drain withen 4days almost 40%

Anka001, 2022-09-10:
After updating to 1.27.2, aqara sensors with a high battery level seem to report plausible values. Sensors with low battery status all report 0%, while running fine.
From my point of view, things have gotten better in 1.27.2, but the calculated battery status in the lower ranges still lacks accuracy.

closed 2022-10-19 by github-actions[bot]