Koenkk/zigbee2mqtt #13700: Devices stop reporting/Can't join new devices

brhahlen, 2022-08-26:
### What happened?

After updating to 1.27 (I think), devices are not able to join the network anymore, and some devices will stop reporting statuses altogether.
Now, I'm not entirely sure _when_ this started happening exactly, but I was able to join devices before.

The "troublesome" devices are an Aqara Sensor ([WSDCGQ11LM](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#xiaomi-wsdcgq11lm)) that I have had for quite a while, and a motion sensor ([WSDCGQ11LM](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#xiaomi-wsdcgq11lm)) that joins, but then just stops reporting.

I've tried downgrading to 1.26, and that kinda works, sometimes.

I've tried switching batteries as well, and have tried multiple, also brand new ones.

Device breakdown:
Total 67
By device type

- Router: 41
- End devices: 26

By power source

- Mains (single phase): 41
- Battery: 26

### What did you expect to happen?

I expect that my devices keep reporting status

### How to reproduce it (minimal and precise)

Run Z2M version 1.27.

### Zigbee2MQTT version

1.27.0-dev commit: c5d54a8 (also with regular 1.27)

### Adapter firmware version

20211217 (also with other firmware)

### Adapter

zzh

Koenkk, 2022-08-26:
In case you have a CC2531, could you make a sniff when the sensor should report data but nothing shows up in z2m?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

brhahlen, 2022-08-26:
I don't have one of those, unfortunately...

ocso, 2022-08-30:
@brhahlen I noticed the same or similar after upgrading to 1.27. IKEA Symfonisk button stopped working, I changed the battery and it could not join to the network again. I have 70+ zigbee devices and all of them have been working since 2 years without major issues. After 1.27 upgrade some of them (Xiaomi motion and door sensors) left the network and I had to re-pair again, 2 of them became unresponsible after 1 hour again and again, did not report anymore.

I decided to downgrade to 1.26. No unresponsible devices anymore and the mentioned IKEA Symfonisk button paired again with one try. Looks strange to me.

CC: @Koenkk

brhahlen, 2022-08-31:
I had to check, but I am actually now on 1.26 and am still suffering from some unresponsive devices.

brhahlen, 2022-09-05:
@ocso I'm back to 1.26 as well and haven't had any issues so far.

closed 2022-10-14 by github-actions[bot]