Koenkk/zigbee2mqtt #13702: Legrand Valena Life Netatmo Dimmer shows as compatible but i can only turn it on or off

wladkolc, 2022-08-26:
### What happened?

I have paired Legrand Valena Life Netatmo Dimmer and it shows that interview completed OK, but not work as described.

### What did you expect to happen?

working dimmer, using builtin buttons as actions for another devices, using knock

### How to reproduce it (minimal and precise)

pair and not working as needed

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle

### Debug log

Changing brightness:
`Debug 2022-08-26 19:14:27Received MQTT message on 'zigbee2mqtt/0x00047400008a371e/set' with data '{"state":"ON","brightness":137}'
Debug 2022-08-26 19:14:27Publishing 'set' 'brightness' to 'Living Room Celing Lamp'
Info 2022-08-26 19:14:27MQTT publish: topic 'zigbee2mqtt/Living Room Celing Lamp', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":137,"device_mode":"dimmer_on","led_when_on":"ON","linkquality":0,"permanent_led":"ON","state":"ON"}'
Debug 2022-08-26 19:14:43Received MQTT message on 'zigbee2mqtt/0x00047400008a371e/set' with data '{"state":"ON","brightness":94}'
Debug 2022-08-26 19:14:43Publishing 'set' 'brightness' to 'Living Room Celing Lamp'
Info 2022-08-26 19:14:43MQTT publish: topic 'zigbee2mqtt/Living Room Celing Lamp', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":94,"device_mode":"dimmer_on","led_when_on":"ON","linkquality":0,"permanent_led":"ON","state":"ON"}'
Debug 2022-08-26 19:14:44Received MQTT message on 'zigbee2mqtt/0x00047400008a371e/set' with data '{"state":"ON","brightness":142}'
Debug 2022-08-26 19:14:44Publishing 'set' 'brightness' to 'Living Room Celing Lamp'
Info 2022-08-26 19:14:44MQTT publish: topic 'zigbee2mqtt/Living Room Celing Lamp', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":142,"device_mode":"dimmer_on","led_when_on":"ON","linkquality":0,"permanent_led":"ON","state":"ON"}'
Debug 2022-08-26 19:14:45Received MQTT message on 'zigbee2mqtt/0x00047400008a371e/set' with data '{"state":"ON","brightness":183}'
Debug 2022-08-26 19:14:45Publishing 'set' 'brightness' to 'Living Room Celing Lamp'
Info 2022-08-26 19:14:45MQTT publish: topic 'zigbee2mqtt/Living Room Celing Lamp', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":183,"device_mode":"dimmer_on","led_when_on":"ON","linkquality":0,"permanent_led":"ON","state":"ON"}'`

Knocking on button:
`Debug 2022-08-26 19:15:43Received Zigbee message from 'Living Room Celing Lamp', type 'attributeReport', cluster 'genIdentify', data '{"identifyTime":180}' from endpoint 1 with groupID 0
Info 2022-08-26 19:15:43MQTT publish: topic 'zigbee2mqtt/Living Room Celing Lamp', payload '{"action":"identify","ballast_lamp_alarm_lamp_burn_hours":false,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":183,"device_mode":"dimmer_on","led_when_on":"ON","linkquality":0,"permanent_led":"ON","state":"ON"}'
Debug 2022-08-26 19:15:45Received Zigbee message from 'Living Room Celing Lamp', type 'attributeReport', cluster 'genIdentify', data '{"identifyTime":60}' from endpoint 1 with groupID 0
Info 2022-08-26 19:15:45MQTT publish: topic 'zigbee2mqtt/Living Room Celing Lamp', payload '{"action":"identify","ballast_lamp_alarm_lamp_burn_hours":false,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":183,"device_mode":"dimmer_on","led_when_on":"ON","linkquality":0,"permanent_led":"ON","state":"ON"}'`

nfnvsc, 2022-09-07:
I got the same device with the same problem.

wladkolc, 2022-09-28:
Hi, i have tried home assistant. It shows me more all inputs and configuration. When i will have more time i will try it and comment here. It looks like there is no support for things like this.

closed 2022-11-05 by github-actions[bot]