Koenkk/zigbee2mqtt #13711: [Feature request]: Support Aqara FP1 Regions

johnsturgeon, 2022-08-27:
### Is your feature request related to a problem? Please describe

Region support in the Aqara FP1 sensor would be great, there are folks who have issues with ceiling fans, and other areas of a monitored room that require ignoring for the sensor to function properly

### Describe the solution you'd like

Aqara FP1 device itself supports regions, but that feature is not in zigbee2mqtt, I'd like to be able to define regions, and have region information in the detection event so that I can filter in / out events that I care about.

### Describe alternatives you've considered

Positioning the sensor in a way so that it's not seeing the region, but of course, that means there are other regions that I miss.

### Additional context

I'm honestly not sure exactly how regions work or are implemented in the FP1, so I'm kind of guessing at what's possible here.

ucsbricks, 2022-09-04:
FP1 is a game changer either way, but if we could get regions support, it would make the device perfect.

ucsbricks, 2022-09-06:
deconz-rest-plugin community did a deep dive in FP1 regions:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5928

mKeRix, 2022-10-07:
I’d still be interested!

mKeRix, 2022-11-08:
/unstale

johnsturgeon, 2022-11-08:
-unstale

patmalcolm91, 2022-12-17:
I started playing around with implementing an external converter to add support for regions, and have managed to create a region and detect events for that region in the debug log, but for some reason, the events are not shown in the web UI or published to MQTT. I'm not a Javascript/Typescript developer, so I'm sure it's a minor issue. I'm posting the code here. If anyone can spot the problem, I would be very grateful and could then proceed to implement the rest of the features and make it more user-friendly.

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: 0x115f, disableDefaultResponse: true},
};

const getKey = (object, value) => {
    for (const key in object) {
        if (object[key] == value) return key;
    }
};
const bind = async (endpoint, target, clusters) => {
    for (const cluster of clusters) {
        await endpoint.bind(cluster, target);
    }
};
const octet_as_hex = (octet) => {
    var hex = ""
    for (const num of octet) {
        hex += `:${num.toString(16).padStart(2, 0)}`
    }
    return hex.substring(1);
};

const array_to_byte = (arr) => {
    var byte = 0;
    for (const x of arr) {
        byte += 2**(x-1)
    }
    return byte;
}

const fzLocal = {
    fp1_regions_events: {
        cluster: 'aqaraOpple',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            const event_lookup = {
                1: 'enter',
                2: 'leave',
                4: 'occupied',
                8: 'unoccupied'
            };
            Object.entries(msg.data).forEach(([key, value]) => {
                switch (parseInt(key)) {
                case 0x0151:
                    meta.logger.debug(`zigbee-herdsman-converters:aqara_fp1: Received key ${key} = ${value}`)
                    meta.logger.debug(`zigbee-herdsman-converters:aqara_fp1: value interpreted ${value[0]} ${value[1]}`)
                    const event = event_lookup[value[1]];
                    const region_id = value[0];
                    meta.logger.debug(`zigbee-herdsman-converters:aqara_fp1: event: region ${region_id} ${event}`)
                    return {region_event: `region_${region_id}_${event}`};
                    break;
                case 0xf7:
                    meta.logger.debug(`zigbee-herdsman-converters:aqara_fp1: Unhandled key ${key} = ${value}`);
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:aqara_fp1: Unknown key ${key} = ${value}`);
                }
            });
[... 54 lines cut ...]
            'sensor detects approaching'),
        exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']).withDescription('Different sensitivities ' +
            'means different static human body recognition rate and response speed of occupied'),
        exposes.enum('reset_nopresence_status', ea.SET, ['']).withDescription('Reset the status of no presence'),
        e.device_temperature(), e.power_outage_count(), exposes.text('regions_config', ea.SET),
        exposes.enum('region_event', ea.STATE, ['region_*_enter', 'region_*_leave', 'region_*_occupied', 'region_*_unoccupied']).withDescription('Region events'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
        await endpoint.read('aqaraOpple', [0x0142], {manufacturerCode: 0x115f});
        await endpoint.read('aqaraOpple', [0x0144], {manufacturerCode: 0x115f});
        await endpoint.read('aqaraOpple', [0x0146], {manufacturerCode: 0x115f});
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

This results in events being printed to the log:
[image]

But not being shown in the UI:
[image]

maxi1134, 2023-01-04:
Any chance of seeing this included in the next Z2M version?

patmalcolm91, 2023-01-06:
> Any chance of seeing this included in the next Z2M version?

Unfortunately I don't have much time currently to finalize the features and prepare a pull request and won't for a few more weeks. Also, one major thing that still should be changed before inclusion in a release is to make the region configuration part of the device-specific settings rather than the normal "exposes" tab. After looking through many examples, I still can't figure out how to do this. If anyone has any tips on that, let me know.

mdziekon, 2023-01-07:
@patmalcolm91 are you OK with me taking your code and submitting a PR based on your solution (with some code refactoring), mostly as is? I was not able to figure out how to move the newly added configs to "Device specific settings" as well, but the rest works just fine, so I should be able to prepare a merge for basic support.

patmalcolm91, 2023-01-07:
> @patmalcolm91 are you OK with me taking your code and submitting a PR based on your solution (with some code refactoring), mostly as is? I was not able to figure out how to move the newly added [...]

Yes, feel free. That would be great!

laplacacm, 2023-01-25:
This is exciting!  It looks like it is getting close.  Thank you to all that helped with this!

mdziekon, 2023-02-02:
> Nice to see some progress on this topic. But, it does not work...Setting a region works after some attempts but when i reload the page, the settings are gone. Somehow not saved....

That does not necessarily mean the settings were not saved. Have you tried to actually trigger a region event? To verify whether the setting upload (to the device) was successful, you can try to trigger a region event, and verify if it's present in the logs or in momentary `action` field.

I'm unaware of any option to retrieve these settings from the device, so settings are uploaded to the device, but not persisted anywhere in Z2M. Maybe there's a way to persist this in Z2M's state as a secondary settings storage, but I'm not that proficient with Z2M's architecture to know that.

chrigu1981, 2023-02-02:
@mdziekon thx for your answer.

i tried setting regions over MQTT as well in the z2m UI and also tried to manually add them in the state.json file, non of them seems to work. Also when i check the state.json file in z2m/data/state.json

Btw: I walked around with my Macbook but never got an action :)

TzachiGuetta, 2023-02-02:
I have this in the homebridge log:
[02/02/2023, 19:40:34] [zigbee2mqtt] Failed to setup stateless programmable switch for accessory Aqara FP1  from expose "{"access":1,"description":"Most recent region event. Event template is \"region_<REGION_ID>_<EVENT_TYPE>\", where <REGION_ID> is region number (1-10), <EVENT_TYPE> is one of \"enter\", \"leave\", \"occupied\", \"unoccupied\". \"enter\" / \"leave\" events are usually triggered first, followed by \"occupied\" / \"unoccupied\" after a couple of seconds.","name":"action","property":"action","type":"enum","values":["region_*_enter","region_*_leave","region_*_occupied","region_*_unoccupied"]}", error: Error: Device found with a wildcard in the exposed possible values for the action, which cannot be mapped: region_*_enter

what does it means?

mdziekon, 2023-02-02:
@chrigu1981 

Setting it in `state.json` directly won't work, for the reason mentioned above. You either have to use the GUI, or issue a command in device's dev console manually.

I've just tried out the new update (as I usually wait a couple of days for the rest of the stuff to stabilise), tested it on two FP1s I have currently wired up (one in the kitchen, one in the bathroom).

I was able to add a region (y = 1, x = 1,2,3,4, so quite a wide area of detection) at my Kitchen's sensor, and after that it properly detects the region events:
```
debug 2023-02-02 19:07:51zigbee-herdsman-converters:xiaomi:aqara_fp1: action: Triggered event (region "1", type "enter")
debug 2023-02-02 19:07:53zigbee-herdsman-converters:xiaomi:aqara_fp1: action: Triggered event (region "1", type "leave")
debug 2023-02-02 19:08:01zigbee-herdsman-converters:xiaomi:aqara_fp1: Unhandled key 0x0143 = 7
debug 2023-02-02 19:08:07zigbee-herdsman-converters:xiaomi:aqara_fp1: Unhandled key 0x0143 = 6
debug 2023-02-02 19:08:10zigbee-herdsman-converters:xiaomi:aqara_fp1: action: Triggered event (region "1", type "enter")
debug 2023-02-02 19:08:13zigbee-herdsman-converters:xiaomi:aqara_fp1: action: Triggered event (region "1", type "occupied")
debug 2023-02-02 19:08:35zigbee-herdsman-converters:xiaomi:aqara_fp1: action: Triggered event (region "1", type "leave")
debug 2023-02-02 19:08:42zigbee-herdsman-converters:xiaomi:aqara_fp1: Unhandled key 0x0143 = 7
debug 2023-02-02 19:08:45zigbee-herdsman-converters:xiaomi:aqara_fp1: action: Triggered event (region "1", type "unoccupied")
```

I was also able to trigger a simpe mobile notification using this HomeAssistant automation:
```yaml
alias: test
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.kitchen_presence_sensor_action
    to: region_1_enter
condition: []
action:
  - service: notify.grouped_bygroup_admin_mobile
    data:
      message: "Triggered: Kitchen, region 1, event enter"
mode: single
```

[image]

---

However, for some weird reason, my other sensor (in a Bathroom) doesn't pick up the update, and after some time there's an error log with a timeout from that device. However, it's not the first time I'm seeing this one acting up, it's either broken or I have to pair it again.

chrigu1981, 2023-02-02:
@mdziekon i have 3 FP1‘s but not 1 is working. Are you using the Sonoff Zigbee Dongle (2652P)? With which firmware?

mdziekon, 2023-02-02:
@chrigu1981 no, I'm using conbee II, with the latest firmware, and all my FP1's have been updated to the latest firmware version.

chrigu1981, 2023-02-02:
@mdziekon Just to clarify, all are working but not with regions. Not sure if i have the latest FW on the sensors but i will try to update them….

johnsturgeon, 2023-03-05:
I'm closing this issue since it's merged, any future issues with regions should probably just be their own.

closed 2023-03-05 by johnsturgeon

amaisano, 2024-10-20:
about this... I just spent 2 hours setting up regions for my office (which work), but as soon as my wife sat down on her desk, my own desk region stopped getting detected.... do regions only work with 1 single person? if so, what is the limitation here?