Koenkk/zigbee2mqtt #13713: Gledopto GL-C-008-2ID can not be turned off using scenes

EuleMitKeule, 2022-08-27:
### What happened?

When trying to create a scene for this device (Gledopto Zigbee LED Controller RGB+CCT (2 ID)), an error message gets displayed showing `Publish 'set' 'scene_add' to '<name_of_the_device>' failed: 'TypeError: Cannot read properties of undefined (reading 'getDevice')'`.

Also when creating a scene for a group that contains this device, it can not be turned off by activating the scene.

### What did you expect to happen?

A scene should be able to be created and the device should be able to be turned off using group scenes.

### How to reproduce it (minimal and precise)

* Add this device
* Create a scene using the frontend
* Error popup will be displayed

### Zigbee2MQTT version

1.26.0 commit: cd23aea

### Adapter firmware version

0x26660700

### Adapter

ConBee II

### Debug log

[relevant_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9437995/relevant_log.txt)

Koenkk, 2022-08-27:
I don't see `Publish 'set' 'scene_add' to '<name_of_the_device>' failed: 'TypeError: Cannot read properties of undefined (reading 'getDevice')'` error in your log, can you post the full log?

EuleMitKeule, 2022-08-27:
@Koenkk Sorry, that line was indeed missing. This is all that gets logged after I try to create the scene.
```
debug 2022-08-28 00:53:36: Received MQTT message on 'zigbee2mqtt/Büro Schreibtisch Strip/set' with data '{"scene_store":{"ID":1,"name":"test"}}'
debug 2022-08-28 00:53:36: Publishing 'set' 'scene_store' to 'Büro Schreibtisch Strip'
error 2022-08-28 00:53:36: Publish 'set' 'scene_store' to 'Büro Schreibtisch Strip' failed: 'TypeError: Cannot read properties of undefined (reading 'getDevice')'
debug 2022-08-28 00:53:36: TypeError: Cannot read properties of undefined (reading 'getDevice')
    at disableDefaultResponse (/app/node_modules/zigbee-herdsman-converters/devices/gledopto.js:225:61)
    at getMetaValues (/app/node_modules/zigbee-herdsman-converters/lib/utils.js:368:61)
    at Object.getOptions (/app/node_modules/zigbee-herdsman-converters/lib/utils.js:359:12)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:5407:99)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:122:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27)
    at WebSocket.emit (node:events:527:28)
info  2022-08-28 00:53:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'scene_store' to 'Büro Schreibtisch Strip' failed: 'TypeError: Cannot read properties of undefined (reading 'getDevice')'","meta":{"friendly_name":"Büro Schreibtisch Strip"},"type":"zigbee_publish_error"}'
```

Koenkk, 2022-08-28:
This issue may already be fixed in the latest-dev, can you check and if not provide me the debug log again?  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

EuleMitKeule, 2022-08-28:
@Koenkk Same problem with latest-dev docker image.
```
debug 2022-08-28 15:38:57: Received MQTT message on 'zigbee2mqtt/Büro Schreibtisch Strip/set' with data '{"scene_store":{"ID":1,"name":"Büro Abend"}}'
debug 2022-08-28 15:38:57: Publishing 'set' 'scene_store' to 'Büro Schreibtisch Strip'
error 2022-08-28 15:38:57: Publish 'set' 'scene_store' to 'Büro Schreibtisch Strip' failed: 'TypeError: Cannot read properties of undefined (reading 'getDevice')'
debug 2022-08-28 15:38:57: TypeError: Cannot read properties of undefined (reading 'getDevice')
    at disableDefaultResponse (/app/node_modules/zigbee-herdsman-converters/devices/gledopto.js:233:61)
    at getMetaValues (/app/node_modules/zigbee-herdsman-converters/lib/utils.js:371:61)
    at Object.getOptions (/app/node_modules/zigbee-herdsman-converters/lib/utils.js:362:12)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:5548:99)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:137:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27)
    at WebSocket.emit (node:events:513:28)
info  2022-08-28 15:38:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'scene_store' to 'Büro Schreibtisch Strip' failed: 'TypeError: Cannot read properties of undefined (reading 'getDevice')'","meta":{"friendly_name":"Büro Schreibtisch Strip"},"type":"zigbee_publish_error"}'
```

EuleMitKeule, 2022-08-28:
Could the problem be that I don't have anything plugged in to the CCT part of the controller?

Koenkk, 2022-08-28:
Pushed a fix, can you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

EuleMitKeule, 2022-08-28:
Creating a scene for the device does work now, but the device can still not be turned off using either the device scene or the group scene. Turning it on or changing brightness and color works fine, it just won't turn off.

Koenkk, 2022-08-29:
Just to confirm; before you do a scene store, you turn the device off, right? If yes, then the FW of this device is bugged.

EuleMitKeule, 2022-08-29:
@Koenkk Yes, of course I turn it off before doing the scene store.
When I turn it off via UI and then load a scene that is supposed to turn it on, it does turn on. But the visible state still shows that it's off, even though it did turn on. When loading a scene that should turn it off, it does not turn off and also shows in UI that the device is still on. Maybe this has something to do with the 2 ID stuff (the rgb and cct state being two separate attributes)?

EuleMitKeule, 2022-08-29:
Correction: It does actually turn off now after I created a scene using a `scene_add` message like this:
```json
{
  "scene_add": {
    "ID": 2,
    "name": "Aus",
    "state": "OFF",
    "state_rgb": "OFF",
    "state_cct": "OFF"
  }
}
```
It still does not work when creating the scene from the UI and also the state change is not reflected by the `state_rgb` and `state_cct` attributes in the UI. I will close this issue and create a new one for those problems.

closed 2022-08-29 by EuleMitKeule