Koenkk/zigbee2mqtt #13714: Slow messages from Symfonisk

michalk-k, 2022-08-27:
### What happened?

The Symfonisk, when turning it right or left, generates the following sequence in the MQTT topic (for turning left):

`brightness_move_down, "", None, brightness_stop, ""`

Some time ago that sequence was generated fast enough to resend it to light changing its brightness gradually. But now
it doesn't generate this sequence repeatedly. When I turn the wheel at normal speed it starts with:

`brightness_move_down, "", None`

and hangs here, until I stop turning the wheel. Then it throws:

 `brightness_stop, ""`

Is it a bug? Some hardcoded throttling? Or how can I retrieve a sequence of messages, which is dependent on turn angle rather than turn speed? is this behavior somewhere described to the detail? I would like to understand that and program my automation properly. 
Even better if I could change this behavior to make it possible to respond on turn speed too (so a number of messages depends only on turn angle).

I'm using HomeAssistant. But I do automate in NodeRed, using MQTT as a message source. I have legacy_triggers disabled

The settings of the Symfonisk are:
Optimistic: true
No debounce
[image]

### What did you expect to happen?

Number of messages proportional to turned angle (ie 1 per degree, or 1 per 5 degree) allowing gradual control of other devices based on this input.

### How to reproduce it (minimal and precise)

Turn the Symfonisk wheel fast enough (with normal speed you expect from the dimmer)the 

### Zigbee2MQTT version

 1.27.0-1

### Adapter firmware version

0x26700700

### Adapter

ConBee2

michalk-k, 2022-09-17:
Any response?

Is it possible that Symfonisk doesn't report the angle but the time of movement instead? it seems that it "emulates" pressing buttons rather than a rotation dimer. 

Anyway Z2M reports movement as long as the knob is being rotated. Not being proportional to the angle. So turning it slowly by 20 degrees may have a greater impact than turning by 180 degrees rapidly.

Is it correct?

Koenkk, 2022-09-19:
This is expected behaviour, the device generates 2 messages when turning, on start and on the stop.

michalk-k, 2022-09-20:
Thank you for your response.
Pretty bad design of the device then.

Currently, I'm simulating multiple-pressing in NodeRed because without it I may get a single trigger while rotating for 180 degree with moderate speed. Not perfect but at least something closer to reality.

Koenkk, 2022-09-20:
> Pretty bad design of the device then.

it is good design actually, otherwise it would be spamming the network. The brightenss move down command indicates that a bulb should dim until the command stop is received. 

I assume this can be closed now.

closed 2022-09-20 by Koenkk