Koenkk/zigbee2mqtt #13717: Pls add Moes MS-105B modelID: 'TS0601', manufacturerName: '_TZE200_fjjbhx9d'

louismulder, 2022-08-27:
### Link

https://nl.aliexpress.com/item/1005004355713229.html?spm=a2g0o.order_list.0.0.21ef79d2tC9N7R&gatewayAdapt=glo2nld

### Database entry

"description":"Smart dimmer module (2 gang)"

### Comments

I have already tried it in the moes.js and works correct.

    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_e3oitdyu'},
					  **{modelID: 'TS0601', manufacturerName: '_TZE200_fjjbhx9d'}**],
        model: 'MS-105B',
        vendor: 'Moes',
        description: 'Smart dimmer module (2 gang)',
        fromZigbee: [fz.moes_105_dimmer, fz.ignore_basic_report],
        toZigbee: [tz.moes_105_dimmer],
        meta: {turnsOffAtBrightness1: true, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        },
        exposes: [e.light_brightness().withEndpoint('l1').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),
            e.light_brightness().withEndpoint('l2').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET)],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 1};
        },
    },

Koenkk, 2022-08-27:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-27 by Koenkk

louismulder, 2022-08-28:
Hi Koen,

Sorry I was too enthusiastic. Only On/Off is working
Pls revert
regards Louis

louismulder, 2022-08-28:
I have made a custom converter, which works ok.
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    parse: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: async (model, msg, publish, options, meta) => {
            const result = {};
            meta.logger.debug(`MOES FZ msg.data=${JSON.stringify(msg.data)} meta.state=${JSON.stringify(meta.state)}`);

            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp; // First we get the data point ID
                const value = tuya.getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                switch (dp) {
                case 1:
                    // {"seq":52224,"dpValues":[{"dp":1,"datatype":1,"data":{"type":"Buffer","data":[1]}}]}
                    result.state_l1 = value ? "ON" : "OFF"; // value is already converted
                    // meta.logger.debug(`MOES FZ Button 1: state=${value}`);
                    // if (value && meta.state[`brightness_l1`]) {                        
                    //     const brightness = utils.mapNumberRange(meta.state[`brightness_l1`], 0, 254, 0, 1000);
                    //     meta.logger.debug(`MOES FZ Brightness 1: brightness=${brightness}`);
                    //     await tuya.sendDataPointValue(entity, dp+1, brightness, 'dataRequest', 1);
                    // }
                    break;
                case 2:
                    // {"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,64]}}
                    // max: 1000 [0,0,3,197]
                    // min: 10 [0,0,0,10]
                    result.brightness_l1 = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    //meta.logger.debug(`MOES FZ Button 1: brightesss=${value}`);
                    break;
                case 7:
                    result.state_l2 = value ? "ON" : "OFF"; // value is already converted
                    //meta.logger.debug(`MOES FZ Button 2: state=${value}`);
                    break;
                case 8:
                    // {"dp":8,"datatype":2,"data":{"type":"Buffer","data":[0,0,3,102]}}
                    result.brightness_l2 = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    //meta.logger.debug(`MOES FZ Button 2: brightesss=${value}`);
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #${
                        dp} with data ${JSON.stringify(dpValue)}`); // This will cause zigbee2mqtt to print similar data to what is dumped in tuya.dump.txt
                }
            }
            return result;
        },
    }
}
const tzLocal = {
    tuya_switch_state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            meta.logger.debug(`MOES TZ key=${key} value=${value} meta.endpoint_name=${meta.endpoint_name} meta.state=${JSON.stringify(meta.state)}`);
            // key=state value=OFF
            const lookup = {l1: 1, l2: 7};
            const dp = lookup[meta.endpoint_name];            
            await tuya.sendDataPointBool(entity, dp, value === 'ON');
        },
    },
    tuya_dimmer_level: {
[... 101 lines cut ...]
dimmerMaxLevel 11 = 1000
dimmerMinLevel 9 = 10

dimmerMaxLevel 19 = 1000
dimmerMinLevel 17 = 10

moesSwitchPowerOnBehavior: 14, = 0

arning 2022-07-16 19:12:45zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #5 with data {"dp":5,"datatype":2,"data":{"type":"Buffer","data":[0,0,3,232]}}
Warning 2022-07-16 19:12:45zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #3 with data {"dp":3,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,10]}}
Warning 2022-07-16 19:12:45zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Warning 2022-07-16 19:12:45zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #4 with data {"dp":4,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Warning 2022-07-16 19:12:46zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #11 with data {"dp":11,"datatype":2,"data":{"type":"Buffer","data":[0,0,3,232]}}
Warning 2022-07-16 19:12:46zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #9 with data {"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,10]}}
Warning 2022-07-16 19:12:46zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #12 with data {"dp":12,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Warning 2022-07-16 19:12:46zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #10 with data {"dp":10,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Warning 2022-07-16 19:12:46zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #19 with data {"dp":19,"datatype":2,"data":{"type":"Buffer","data":[0,0,3,232]}}
Warning 2022-07-16 19:12:46zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #17 with data {"dp":17,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,10]}}
[... cut ...]
Warning 2022-07-16 19:12:47zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Warning 2022-07-16 19:12:47zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #18 with data {"dp":18,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Warning 2022-07-16 19:12:47zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #21 with data {"dp":21,"datatype":4,"data":{"type":"Buffer","data":[1]}}
Warning 2022-07-16 19:12:47zigbee-herdsman-converters:MOES: NOT RECOGNIZED DP #14 with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}

*/