Koenkk/zigbee2mqtt #13723: TS0207_repeater is recoginsed as TS0207_water_leak_detector

SikaiGGMU, 2022-08-28:
### What happened?

My threeTS0207_repeater are recognised as TS0207_water_leak_detector in my zigbee2mqtt home assistant add-on.

### What did you expect to happen?

TS0207_repeater will not be recoginsed as another model

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

Z-Stack 3.x.0

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

No debug log, if need, I will manage to upload some

Koenkk, 2022-08-28:
Can you provide the data/database.db entry of this device?

SikaiGGMU, 2022-08-28:
after I add the following external converter, the device is recognised as what it should be:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =[
    {
        fingerprint: [{modelID: 'TS0207', manufacturerName: '_TZ3000_nkkl7uzv'}],
        model: 'TS0207_repeater',
        vendor: 'TuYa',
        description: 'Repeater',
        fromZigbee: [fz.linkquality_from_basic],
        toZigbee: [],
        exposes: [],
    },
]

module.exports = definition;

Koenkk, 2022-08-28:
`_TZ3000_nkkl7uzv` has already been added in the dev branch, you can use your external converter until the next release on 1 September.

closed 2022-08-28 by Koenkk

Elkillor, 2022-09-22:
is the change implmented in version 1.27.2 because i also have the issue?

SikaiGGMU, 2022-10-02:
yes