Koenkk/zigbee2mqtt #13725: Problem of capital letters in the documentation

nixilb, 2022-08-28:
### What happened?

All documentation is wrong.
It says that "Temperature" is exposed.
This is false: "temperature" is.
This is very important since MQTT topics are case sensitiv.
(and people copy paste from the documentation as I did today, wondering why my MQTT was so silent!)

### Zigbee2MQTT version

NA

### Adapter firmware version

NA

### Adapter

noone

Koenkk, 2022-08-28:
Can you provide a link of this? If I look at e.g. https://www.zigbee2mqtt.io/devices/TS0201.html#temperature-numeric it is correct:

> Value can be found in the published state on the `temperature` property.

nixilb, 2022-08-28:
I understand what happened: as the title is the same as the property, I used the title as the property.
Thus, the doc is factually Ok (but may be error prone).

closed 2022-08-29 by Koenkk