Koenkk/zigbee2mqtt #13727: Saswell SEA801-Zigbee/SEA802-Zigbee, Additional manufacturer name[New device support]:

SimonSezKossel, 2022-08-28:
### Link

https://nl.aliexpress.com/item/1005003952150617.html?spm=a2g0o.order_list.0.0.773979d2HwK8Cy&gatewayAdapt=glo2nld

### Database entry

{"id":50,"type":"EndDevice","ieeeAddr":"0x0c4314fffe61c76e","nwkAddr":21015,"manufId":4098,"manufName":"_TZE200_bvu2wnxz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�fp*\u0013\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\f\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u00126\u0010�*\u00127\u0010�*\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_bvu2wnxz","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d9966d","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1661679562548,"defaultSendRequestWhen":"immediate"}

### Comments

Previously I tried to do a shortcut with this device hoping it would work: https://github.com/Koenkk/zigbee2mqtt/issues/12315

Conclusion: this device is not compatible with Saswell SEA801. Unfortunately it`s now recognized as such, giving a lot of errors caused by my previous request.
No idea how to proceed, but would love to help cleaning up the mess I caused and would like to get this Radiator valve up and running.

Koenkk, 2022-08-28:
`_TZE200_bvu2wnxz` is already supported in z2m 1.27.0

Koenkk, 2022-08-28:
Then the data points will be different, meaning it has to be reverse engineered via the TuYa gateway: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

I will remove the `_TZE200_bvu2wnxz` from `SEA801-Zigbee/SEA802-Zigbee`

raketenemo, 2022-09-05:
Here you have a connector with some basic functionality for the thermostat:

````
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocal = {
  dataPoints: {
    me167Mode: 2,
    me167HeatingSetpoint: 4,
    me167LocalTemp: 5,
    me167ChildLock: 7,
    me167Heating: 3,
    me167ScheduleMon: 28,
    me167ScheduleTue: 29,
    me167ScheduleWed: 30,
    me167ScheduleThu: 31,
    me167ScheduleFri: 32,
    me167ScheduleSat: 33,
    me167ScheduleSun: 34,
  },
};

const fzLocal = {
  me167_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};

        // ToDo - currently not sure of the format
        // function weeklySchedule(day, value) {
        //   // byte 0 - Day of Week (0~7 = Mon ~ Sun???)
        //   // byte 1 - hour ???
        //   // byte 2 - minute ???
        //   // byte 3 - second ???
        //   // byte 4 - Temperature (temp = value / 10)

        //   const weekDays=['mon', 'tue', 'wed', 'thu', 'fri', 'sat', 'sun'];
        //   // we get supplied in value only a weekday schedule, so we must add it to
        //   // the weekly schedule from meta.state, if it exists
        //   const weeklySchedule= meta.state.hasOwnProperty('weekly_schedule') ? meta.state.weekly_schedule : {};
        //   meta.logger.info(JSON.stringify({'received day': day, 'received values': value}));
        //   let daySchedule = []; // result array
        //   for (let i=1; i<16 && value[i]; ++i) {
        //     const aHour=value[i];
        //     ++i;
        //     const aMinute=value[i];
        //     ++i;
        //     const aSecond=value[i];
        //     ++i;
        //     const aTemp=value[i];
        //     daySchedule=[...daySchedule, {
        //       temperature: Math.floor(aTemp/10),
        //       hour: aHour,
        //       minute: aMinute,
        //       second: aSecond,
        //     }];
        //   }
        //   meta.logger.info(JSON.stringify({'returned weekly schedule: ': daySchedule}));
        //   return {'weekly-schedule': {...weeklySchedule, [weekDays[day]]: daySchedule}};
        //}

        for (const dpValue of msg.data.dpValues) {
[... 167 lines cut ...]
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
      e.child_lock(),
      exposes.binary('heating', ea.STATE, 'ON', 'OFF').withDescription('Device valve is open or closed (heating or not)'),
      exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 1)
                     .withLocalTemperature()
                     .withSystemMode(['auto','heat','off'])

        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
````
Working functions:
- local temperature
- heating setpoint
- valve position
- mode
- child lock

I'll try to add some functionality, when I get my tuya bridge...

Pewidot, 2022-09-06:
@raketenemo I can confirm your code absolutely works. You have to set all thermostats to heating or off mode. The Time mode overrides all changes to 16 degree after a few minutes. You can also add the battery functionality just by copying the battery code from TS0601_thermostat_1. I had some issues getting the local temperature from 2 of my thermostats but after removing and reconnecting everythign works now!

raketenemo, 2022-09-07:
I was able to map all functions of the thermostat with the help of the tuya brige.

**Connector:**
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocal = {
  dataPoints: {
    me167Mode: 2,
    me167HeatingSetpoint: 4,
    me167LocalTemp: 5,
    me167ChildLock: 7,
    me167Heating: 3,
    me167Schedule1: 28,
    me167Schedule2: 29,
    me167Schedule3: 30,
    me167Schedule4: 31,
    me167Schedule5: 32,
    me167Schedule6: 33,
    me167Schedule7: 34,
    me167ErrorCode: 35,
    me167FrostGuard: 36,
    me167AntiScaling: 39,
    me167TempCalibration: 47,
  },
};

const fzLocal = {
  me167_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};

        function weeklySchedule(day, value) {
          // byte 0 - Day of Week (0~7 = Wed ~ Tue) ???
          // byte 1 - hour ???
          // byte 2 - minute ???
          // byte 3 - Temp (temp = value )
          // byte 4 - Temperature (temp = value / 10)

          const weekDays=[ 'wed', 'thu', 'fri', 'sat', 'sun','mon', 'tue'];
          // we get supplied in value only a weekday schedule, so we must add it to
          // the weekly schedule from meta.state, if it exists
          const weeklySchedule= meta.state.hasOwnProperty('weekly_schedule') ? meta.state.weekly_schedule : {};
          meta.logger.info(JSON.stringify({'received day': day, 'received values': value}));
          let daySchedule = []; // result array
          for (let i=1; i<16 && value[i]; ++i) {
            const aHour=value[i];
            ++i;
            const aMinute=value[i];
            ++i;
            const aTemp2=value[i];
            ++i;
            const aTemp=value[i];
            daySchedule=[...daySchedule, {
              temperature: Math.floor((aTemp+aTemp2*256)/10),
              hour: aHour,
              minute: aMinute,
            }];
          }
          meta.logger.info(JSON.stringify({'returned weekly schedule: ': daySchedule}));
          return {'weekly-schedule': {...weeklySchedule, [weekDays[day]]: daySchedule}};
[... 257 lines cut ...]
    "sat":[
          {"hour":8,"minute":0,"temperature":25},
          {"hour":12,"minute":0,"temperature":26},
          {"hour":18,"minute":0,"temperature":27}
        ],
    "sun":[
          {"hour":8,"minute":0,"temperature":28},
          {"hour":12,"minute":0,"temperature":29},
          {"hour":18,"minute":0,"temperature":30}
        ]
  }
```

**Available Features:**
- local temperature
- heating setpoint
- valve position
- mode
- child lock
- weekly schedule
- local temeperature calibration
- anti scaling (untested - should trigger some weekly valve movements)
- frost protection (untested)

I was not able to trigger a status update via Zigbee. Maybe someone with more experience has an idea on how to implement this...

SimonSezKossel, 2022-09-08:
@raketenemo: Great work!

Tested your converter and it`s looking good. This is what my unit is reporting back:

    "anti_scaling": "OFF",
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 27,
    "frost_guard": "OFF",
    "heating": "ON",
    "linkquality": 112,
    "local_temperature": 23,
    "local_temperature_calibration": 0,
    "system_mode": "heat"

Only I cannot change "frost_guard" and "anti_scaling" "
All others settings can be changed

Stov1k, 2022-09-12:
What about battery status? Is battery info available and could it also be added to the mapping? Thanks for the code shared earlier!

wirtsi, 2022-09-20:
I also got one of those valves. Can confirm that temperature adjustments work with the code from @raketenemo , on/off switch (aka heating mode), window_detection and away_mode still seem wonky. Also current temperature and battery is not present, but it's a start! Thanks guys!

raketenemo, 2022-09-27:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocal = {
  dataPoints: {
    me167Mode: 2,
    me167HeatingSetpoint: 4,
    me167LocalTemp: 5,
    me167ChildLock: 7,
    me167Heating: 3,
    me167Schedule1: 28,
    me167Schedule2: 29,
    me167Schedule3: 30,
    me167Schedule4: 31,
    me167Schedule5: 32,
    me167Schedule6: 33,
    me167Schedule7: 34,
    me167ErrorCode: 35,
    me167FrostGuard: 36,
    me167AntiScaling: 39,
    me167TempCalibration: 47,
  },
};

const fzLocal = {
  me167_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};

        function weeklySchedule(day, value) {
          // byte 0 - Day of Week (0~7 = Wed ~ Tue) ???
          // byte 1 - hour ???
          // byte 2 - minute ???
          // byte 3 - Temp (temp = value )
          // byte 4 - Temperature (temp = value / 10)

          const weekDays=[ 'wed', 'thu', 'fri', 'sat', 'sun','mon', 'tue'];
          // we get supplied in value only a weekday schedule, so we must add it to
          // the weekly schedule from meta.state, if it exists
          const weeklySchedule= meta.state.hasOwnProperty('weekly_schedule') ? meta.state.weekly_schedule : {};
          meta.logger.info(JSON.stringify({'received day': day, 'received values': value}));
          let daySchedule = []; // result array
          for (let i=1; i<16 && value[i]; ++i) {
            const aHour=value[i];
            ++i;
            const aMinute=value[i];
            ++i;
            const aTemp2=value[i];
            ++i;
            const aTemp=value[i];
            daySchedule=[...daySchedule, {
              temperature: Math.floor((aTemp+aTemp2*256)/10),
              hour: aHour,
              minute: aMinute,
            }];
          }
          meta.logger.info(JSON.stringify({'returned weekly schedule: ': daySchedule}));
          return {'weekly-schedule': {...weeklySchedule, [weekDays[day]]: daySchedule}};
        }

        for (const dpValue of msg.data.dpValues) {
[... 217 lines cut ...]
        tzLocal.me167_thermostat_frost_guard,
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
      e.child_lock(),
      exposes.binary('heating', ea.STATE, 'ON', 'OFF').withDescription('Device valve is open or closed (heating or not)'),
      exposes.switch().withState('anti_scaling', true).withDescription('Anti Scaling feature is ON or OFF'),
      exposes.switch().withState('frost_guard', true).withDescription('Frost Protection feature is ON or OFF'),
      exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 1)
                     .withLocalTemperature()
                     .withSystemMode(['auto','heat','off'])
                     .withLocalTemperatureCalibration(-3, 3, 1, ea.STATE_SET)
    ],
};

module.exports = definition;

```

I added the **low battery detection**.
This makes the thermostat sufficiently usable for my purposes. But please feel welcome to further improve the connector ;)

wirtsi, 2022-09-28:
So I managed to get the code working. I am dropping the code above into `node_modules/zigbee-herdsman-converters/devices/` as saswell_buv.js.

To make it work I need to wrap definition in the last line into an array (so `[definition]`), I have no idea why that's needed

twhittock, 2022-09-28:
Thank you @raketenemo - I have made some minor changes and pushed a repo here https://github.com/twhittock/avatto_me167 - I needed the heating state to be a part of the climate entity in home assistant, so I re-exposed `heating` as `running_state`. I had to delete the state.json entry to make it upgrade properly, though...

@wirtsi the above repo has a very quick rundown on how to install this as an external converter.

HTH.

wirtsi, 2022-09-28:
@twhittock can you elaborate what you mean by `config/zigbee2mqtt` ... so the installation folder zigbee2mqtt is installed? When I follow the steps in your repo, the web ui only gives me an option to copy paste a new converter. Pasting the me167.js file gives errors about it being in an incompatible format

twhittock, 2022-09-28:
Sounds like you're using a different version of zigbee2mqtt than me, or not using the HA add-on... https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters may be helpful?

augard, 2022-10-05:
@twhittock Battery status will be shown only when it's on lower power? It's possible to see state of anti_scalling or frost_guard. It's reporting just 'null' value.

pratsenka, 2022-10-20:
I see that anti_scaling and frost_guard cannot be changed neither by web interface nor by direct mqtt /set message

pratsenka, 2022-10-21:
> I see that anti_scaling and frost_guard cannot be changed neither by web interface nor by direct mqtt /set message

The right converters for anti_scaling and frost_guard are:
  me167_thermostat_anti_scaling: {
    key: ['anti_scaling'],
    convertSet: async (entity, key, value, meta) => {
      await tuya.sendDataPointBool(entity, tuyaLocal.dataPoints.me167AntiScaling, value === 'ON');
    },
  },
  me167_thermostat_frost_guard: {
    key: ['frost_guard'],
    convertSet: async (entity, key, value, meta) => {
      await tuya.sendDataPointBool(entity, tuyaLocal.dataPoints.me167FrostGuard, value === 'ON');
    },

raketenemo, 2022-10-23:
Here some screenshots from the "Smart Life" app...

**Main menu with low battery error:**
[image]

**Mode selection:**
[image]

**Time schedule:**
[image]

**Settings:**
[image]

[image]

rickx, 2022-11-09:
is this going to be integrated? What are the missing steps?

pkoretic, 2022-11-09:
@rickx it's a least missing schedule support and there are errors on getting values in Z2M (refresh button press) so wouldn't say it's ready yet but we can all try to finish it :)

rickx, 2022-11-10:
ok, will give it a try next week when the hub arrives. Actually schedule support is there?

timderspieler, 2022-11-13:
> ok, will give it a try next week when the hub arrives. Actually schedule support is there?

I am currently using the scheduler addon to accomplish this. Is there someone out there that knows how the schedule normally can be setup using the trv build in week schedule functionality?

rickx, 2022-11-15:
from what I see in my first tests here, raketenemo's first raft was correct, while the latest version is wrong: the commands for the daily schedules (one for each day) start on Monday = 28 and end on Sunday = 34. It might be locale dependent but which locale starts the week on Wednesday?

hhoang308, 2022-11-28:
> from what I see in my first tests here, raketenemo's first draft was correct, while the latest version is wrong: the commands for the daily schedules (one for each day) start on Monday = 28 and end on Sunday = 34. It might be locale dependent but which locale starts the week on Wednesday?

can you give me a link to the first draft, because as far as I can see all the versions are the same. Also, do you have a raw byte schedule, please let me know.

rickx, 2022-11-28:
this is the change I was referring to:
`const weekDays=['mon', 'tue', 'wed', 'thu', 'fri', 'sat', 'sun'];`
and which was later changed to start from Wednesday.

IIChrisII, 2022-11-30:
this external converter works already pretty fine but in home assistant i don't see the frost_guard and anti_scaling entities. Instead the "main switch" for the device is used for controlling frost_guard and is not working. When i press this switch in home assistant, Zigbee2mqtt is showing "No converter available for 'state' ("ON")". anti_scaling is completly missing in HA. Can somebody confirm?

I use the mosquitto addon as broker inside home assistant and a external zigbee2mqtt server.

vladyspavlov, 2022-12-05:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5081

closed 2022-12-06 by Koenkk

Evgeka07, 2022-12-09:
@raketenemo:
"Weekly Schedule:
The schedule can be set with <friendly_name>/weekly_schedule/set command with payload:"

where is this stated? in which file?

raketenemo, 2022-12-09:
Hi @Evgeka07,
I have described the schedule mechanism in the [code above the comment](https://github.com/Koenkk/zigbee2mqtt/issues/13727#issuecomment-1239928984) or in my external converter
[here](https://gist.github.com/raketenemo/a0088a2ba529f84dfe428a0c117e5d1a) in line 208-237.

When using this external converter, you are able to use the <friendly_name>/weekly_schedule/set command to set up the schedule. When sending the command, you have to send the json encoded payload as shown [here](https://gist.github.com/raketenemo/6291624a3c292ad98b008c99a47ca65a).

As mentioned before in here, the days might be out of order. The order of the weekdays is set in line 211.
My code may be outdated by now.

derpuma, 2022-12-09:
> this is the change I was referring to: `const weekDays=['mon', 'tue', 'wed', 'thu', 'fri', 'sat', 'sun'];` and which was later changed to start from Wednesday.

I think it is more important how you setup your week in HA rather then the sorting of the days there?
[image]

cgarciaq, 2023-02-01:
Hi, I do not know why but my SEA801-Zigbee thermostat started opening and immediatly closing the valve every 5 minutes, no matter if it is on or off, showing 'AdAP' on the screen, which means anti-scaling is being triggered. It was working well until yesterday. I can see anti-scaling set in the state, but I do not know how to turn it off. Is there a way to do it already?

`
{
    "anti_scaling": "ON",
    "away_mode": "OFF",
    "battery_low": false,
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 10,
    "frost_detection": "ON",
    "linkquality": 116,
    "local_temperature": 19,
    "local_temperature_calibration": -1,
    "preset_mode": "none",
    "system_mode": "off",
    "window_detection": "OFF",
    "device": {
        "applicationVersion": 85,
        "dateCode": "",
        "friendlyName": "Radiador de la habitación 3",
        "hardwareVersion": 1,
        "ieeeAddr": "0x8cf681fffec58437",
        "manufacturerID": 4098,
        "manufacturerName": "_TZE200_c88teujp",
        "model": "SEA801-Zigbee/SEA802-Zigbee",
        "networkAddress": 30541,
        "powerSource": "Battery",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "heating": null
}
'

randykuntz, 2026-10-16:
I can confirm that the Hama head variant `_TZE200_yw7cahqs` (Zigbee module `1.1.5`, MCU module `1.0.0`) sadly does not send the `running_state` datapoint at all.