Koenkk/zigbee2mqtt #13737: Niko 552-721X2 endpoint actions not working

sephiroth1395, 2022-08-28:
### What happened?

In both control_relay and decoupled modes, the actions remains null on both endpoints L1 and L2 when I activate the switches physically.  When in control_relay mode, the relays do operate when pressing the button so I have every reason to believe the device is not broken.  It is brand new.

I can see a single "single, hold or release" action exposed in Home Assistant at device level instead of the l1 and l2 endpoints.

### What did you expect to happen?

See the single, hold and release actions exposed.

### How to reproduce it (minimal and precise)

Push any switch of the device.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20210708

### Adapter

Sonoff USB Dongle

sjorge, 2022-08-29:
Is it just home assistant not seeing the actions because it might be looking in the wrong property, or are you not getting them via mqtt either?

sephiroth1395, 2022-08-30:
Okay, just had a better look at this to answer your question.

Current configuration of the device:
`{
    "linkquality": 60,
    "operation_mode": "decoupled",
    "operation_mode_l1_l1": "control_relay",
    "operation_mode_l2_l2": "decoupled",
    "state_l1": "OFF",
    "state_l2": "OFF"
}`

When pressing the left button, I see these events in the Z2M logs:

`Zigbee2MQTT:info  2022-08-30 11:40:52: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":63,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"ON","state_l2":"OFF"}'

`Zigbee2MQTT:info  2022-08-30 11:40:54: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":54,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'`

`Zigbee2MQTT:info  2022-08-30 11:40:55: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":63,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"ON","state_l2":"OFF"}'`

`Zigbee2MQTT:info  2022-08-30 11:40:56: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":60,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'`

You can see the action_l1 attribute remains to null.  Z2M appears to see the relay event, but not the button click themselves.

What's even weirder is this.  When I press the right button, I see these events in the Z2M log:

`Zigbee2MQTT:info  2022-08-30 11:45:28: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":27,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"ON"}'`

`Zigbee2MQTT:info  2022-08-30 11:45:29: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":33,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'`

The l2 state shouldn't change as it is in decoupled operation mode.  I can confirm that the L2 relay DOES activate.

If you provide me directions I can provide other logs if it can help.  I'm really at loss here..  I can't really tell if the device is doing something wrong or if it's a Z2M issue.

sjorge, 2022-08-30:
Can you set the log level to debug and press the buttons again, that should show what the device is sending. And hopefully that is enough for me to figure out what is wrong. 🤞

sephiroth1395, 2022-08-30:
Starting from both relays off, nothing changed in the device config.

L1 button first press:

`Zigbee2MQTT:debug 2022-08-30 13:48:56: Received Zigbee message from 'Interrupteurs toit cuisine', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0`

`Zigbee2MQTT:info  2022-08-30 13:48:56: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":75,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"ON","state_l2":"OFF"}'`

L1 second press:

`Zigbee2MQTT:debug 2022-08-30 13:50:35: Received Zigbee message from 'Interrupteurs toit cuisine', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0`

`Zigbee2MQTT:info  2022-08-30 13:50:35: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":54,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'`

L2 first press:

`Zigbee2MQTT:debug 2022-08-30 13:51:49: Received Zigbee message from 'Interrupteurs toit cuisine', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0`

`Zigbee2MQTT:info  2022-08-30 13:51:49: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":48,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"ON"}'`

L2 second press:

`Zigbee2MQTT:debug 2022-08-30 13:52:15: Received Zigbee message from 'Interrupteurs toit cuisine', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0`

`Zigbee2MQTT:info  2022-08-30 13:52:15: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":30,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'`

I also looked into what the feedback is when I try to set the switch behaviour to decoupled, as obviously L2 behaves as control_relay although it is configured as decoupled.

Here is what I see in the debug logs :

`Zigbee2MQTT:debug 2022-08-30 13:53:37: Received MQTT message on 'zigbee2mqtt/Interrupteurs toit cuisine/set' with data '{"operation_mode_l2":"decoupled"}'`

`Zigbee2MQTT:debug 2022-08-30 13:53:37: Publishing 'set' 'operation_mode' to 'Interrupteurs toit cuisine'`

`Zigbee2MQTT:info  2022-08-30 13:53:37: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":36,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'`

When I do a get after that:

`Zigbee2MQTT:debug 2022-08-30 13:54:32: Received MQTT message on 'zigbee2mqtt/Interrupteurs toit cuisine/get' with data '{"operation_mode_l2":""}'`

`Zigbee2MQTT:debug 2022-08-30 13:54:32: Publishing get 'get' 'operation_mode' to 'Interrupteurs toit cuisine'`

`Zigbee2MQTT:debug 2022-08-30 13:54:32: Received Zigbee message from 'Interrupteurs toit cuisine', type 'readResponse', cluster 'manuSpecificNiko1', data '{"switchOperationMode":1}' from endpoint 2 with groupID 0`

`Zigbee2MQTT:info  2022-08-30 13:54:32: MQTT publish: topic 'zigbee2mqtt/Interrupteurs toit cuisine', payload '{"action_l1":null,"action_l2":null,"linkquality":39,"operation_mode":"decoupled","operation_mode_l1":null,"operation_mode_l1_l1":"control_relay","operation_mode_l2":null,"operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'`

I'm not a Z2M expert, but I think the whole issue is excess/missing _l1 and _l2 things here and there.  In particular, this:

`"operation_mode_l2":null,"operation_mode_l2_l2":"decoupled"`

Sounds extremely weird, and as control_relay is the default behaviour, I think it's our clue.  I will be away from home for the coming 48 hours but happy to test any fix that might come up :)

sjorge, 2022-08-30:
Yes the double _l2_l2 and _l1_l1 are not correct for sure, but it at least looks to be reading the info from the correct endpoint. 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/5ad7be2d69f371b170b5b0741e98d2e4579ea255/devices/niko.js#L193-L221

Looks like I did set multiEndpoint correctly so as of now i am puzzled why it is not working. I think the actions are probably missing because the device never wrote the decoupled config for the 2nd endpoint correctly so it's not generating the actions, as I am still seeing the genOnOff reporting coming in.

sephiroth1395, 2022-08-30:
Not really how to do this with Z2M running as a HASS add-on..  Looking into it.

Koenkk, 2022-08-30:
@sephiroth1395 issue should be fixed now, updated to the latest-dev, clear out `data/state.json` (or just `operation_mode_l2_l2`/`operation_mode_l1_l1`) and start z2m. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@sjorge for toZigbee converters, the endpoint is added by z2m thus it doesn't have to be added in the converter.

sjorge, 2022-08-30:
@Koenkk ok so as my diff showed, I think this is also needed for the action one as you missed that.

Koenkk, 2022-08-31:
@sjorge that should be fine, this is a fromZigbee converter so no endpoint will appended twice.

sjorge, 2022-08-31:
OK, lets wait until we hear back form sephiroth1395 then to make sure everything is working.

sephiroth1395, 2022-09-06:
Sorry for the late feedback.  Life happened.
I wasn't comfortable with switching to the dev branch, so I just fetched the niko.js file from master, cleaned state file and restarted Z2M.

It's better!  The duplicate state is still there but so is the "normal" one:

```
{
    "linkquality": 90,
    "operation_mode": "decoupled",
    "operation_mode_l1": "decoupled",
    "operation_mode_l1_l1": "control_relay",
    "operation_mode_l2": "decoupled",
    "operation_mode_l2_l2": "decoupled",
    "state_l1": "OFF",
    "state_l2": "OFF",
    "action_l1": null,
    "action_l2": null
}
```

And it appears to be obeyed.  The relays are indeed decoupled from the switches !  Changing the operation_mode to control_relay also works as intended.

What I'm still missing is actions.  Things are still happening on a single action attribute, that appears when I press a button, instead on action_l1 and action_l2.

sjorge, 2022-09-06:
Can you try applying the additional change I had in my diff for the switch_action function?

https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...sjorge:zigbee-herdsman-converters:patch-2

sephiroth1395, 2022-09-06:
No difference, but I noticed only the left (l1) button triggers events to HA.  The second one, not so much.

sjorge, 2022-09-06:
Can you grab another debug log with both of l1 and l2 in decoupled mode when you click each of the buttons?

closed 2022-10-15 by github-actions[bot]

nlspln, 2022-10-20:
Hi all, 
I am having the same problem with these devices. Found this thread and so I tested now with the Edge build to see if there's improvement. 

### Symptoms (on edge build)
- When at least 1 of the two switches is set as `decoupled`, neither physical button doesn't respond. 
- HASS shows Null as the status, I don't get a `single`, `hold`, and `release` action.
- I can control the relays using z2m / HASS now, which wasn't the case with the stable build. 

### Debug logs 
Below are the logs I get when pressing the physical switches. Both are set to decoupled mode:

```
Debug 2022-10-20 11:24:23Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16}' from endpoint 1 with groupID 0
Info 2022-10-20 11:24:23MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action":null,"action_l1":null,"action_l2":null,"linkquality":120,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
Debug 2022-10-20 11:24:23Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":64}' from endpoint 1 with groupID 0
Info 2022-10-20 11:24:23MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action":"single","action_l1":null,"action_l2":null,"linkquality":120,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
Info 2022-10-20 11:24:23MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action":"","action_l1":null,"action_l2":null,"linkquality":120,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
Info 2022-10-20 11:24:23MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights/action', payload 'single'
Debug 2022-10-20 11:24:24Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":4096}' from endpoint 1 with groupID 0
Info 2022-10-20 11:24:24MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action_l1":null,"action_l2":null,"linkquality":127,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
Debug 2022-10-20 11:24:24Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16384}' from endpoint 1 with groupID 0
Info 2022-10-20 11:24:24MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action_l1":null,"action_l2":null,"linkquality":127,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
```

Hope this helps.

sjorge, 2022-10-29:
I no longer have the dual rocker one so I can't double check myself, so some follow up questions:

1. when setting to decoupled on operation_mode_l1, what does a /get for operation_mode_l2 return? Is it also updated to decoupled?

2. can you post a debug log in decoupled mode while doing the following: 
- press left button
- press and hold left button  (5 sec or so should do fine I think)
- press right button
- press and hold right button (5sec or should should do)

The value in your snipper above for 4096 16384 are things I have not see with my brief time with the device, so it probably means I did not capture all the data correctly before I send it back. I only had like 3 days to poke and collect data.

My initial messing around (pre having any support) indicated that it should be possible to toggle the mode individually as the difference between single and dual seemed to be an extra endpoint mirroring all the clusters. This might have been wrong. Hopefully 1 above should that they are either in sync or not, if the former it's probably an all or nothing option.

That would also explain why all actions arrive on endpoint 1 even when using the other button. If that's the case the action map probably also needs to be extended. Currently it is:
```
const actionMap = {16: null, 64: 'single', 32: 'hold', 48: 'release'};
```

But it might be that there is another set of 4 values for the 2nd button.

nlspln, 2022-10-31:
> 1. when setting to decoupled on operation_mode_l1, what does a /get for operation_mode_l2 return? Is it also updated to decoupled?

These are the logs after setting operation_mode_l1 to decoupled and pressing the physical button for L2: 
```
Debug 2022-10-31 07:13:59Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":4096}' from endpoint 1 with groupID 0
Info 2022-10-31 07:13:59MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action_l1":null,"action_l2":null,"linkquality":76,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"OFF"}'
Debug 2022-10-31 07:13:59Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16384}' from endpoint 1 with groupID 0
Info 2022-10-31 07:13:59MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action_l1":null,"action_l2":null,"linkquality":72,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"OFF"}'
Debug 2022-10-31 07:13:59Received Zigbee message from 'Master Bedroom Heater', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1223362]}' from endpoint 1 with groupID 0
Info 2022-10-31 07:13:59MQTT publish: topic 'zigbee2mqtt/Master Bedroom Heater', payload '{"child_lock":"LOCK","current":0,"energy":122.34,"led_enable":true,"linkquality":87,"power":0,"state":"OFF","voltage":230.4}'
Debug 2022-10-31 07:13:59Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2022-10-31 07:13:59MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action_l1":null,"action_l2":null,"linkquality":51,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"OFF"}'
```

> 2. can you post a debug log in decoupled mode while doing the following:
> 
> * press left button
> * press and hold left button  (5 sec or so should do fine I think)
> * press right button
> * press and hold right button (5sec or should should do)
**Left button (L1 & L2 decoupled)** 
```
Debug 2022-10-31 07:16:34Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16}' from endpoint 1 with groupID 0
Info 2022-10-31 07:16:34MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action":null,"action_l1":null,"action_l2":null,"linkquality":94,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"OFF"}'
Debug 2022-10-31 07:16:35Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":32}' from endpoint 1 with groupID 0
Info 2022-10-31 07:16:35MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action":"hold","action_l1":null,"action_l2":null,"linkquality":91,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"OFF"}'
Info 2022-10-31 07:16:35MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action":"","action_l1":null,"action_l2":null,"linkquality":91,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"OFF"}'
Info 2022-10-31 07:16:35MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights/action', payload 'hold'
Debug 2022-10-31 07:16:38Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":48}' from endpoint 1 with groupID 0
Info 2022-10-31 07:16:38MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action":"release","action_l1":null,"action_l2":null,"linkquality":91,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"OFF"}'
Info 2022-10-31 07:16:38MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action":"","action_l1":null,"action_l2":null,"linkquality":91,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"OFF"}'
Info 2022-10-31 07:16:38MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights/action', payload 'release'
```

**Right button (L2 & L1 decoupled)**
```
Debug 2022-10-31 07:18:41Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":4096}' from endpoint 1 with groupID 0
Info 2022-10-31 07:18:41MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action_l1":null,"action_l2":null,"linkquality":40,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"ON","state_l2":"OFF"}'
Debug 2022-10-31 07:18:42Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":8192}' from endpoint 1 with groupID 0
[... cut ...]
**Right button (L1 decoupled, L2 control_relay)**
Debug 2022-10-31 07:31:20Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2022-10-31 07:31:20MQTT publish: topic 'zigbee2mqtt/Living Room / Kitchen Lights', payload '{"action_l1":null,"action_l2":null,"linkquality":83,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"control_relay","state_l1":"ON","state_l2":"ON"}'

Hope this helps

Herwigvd, 2022-11-01:
Hello, I am also struggling with this double Niko switch and recognize the problems described in this topic. I want to use it in decoupled mode and detect the individual switches. Not sure if I can help in debugging as I am not that acquainted with github and install a beta version to do some testing. Nevertheless, if I can do a test with the released version then let me know.

aocaoc, 2022-12-19:
I'm having the same problem.
When the switches are set to decoupled, neither action is recognized.

sjorge, 2022-12-21:
Can you try with this external converter? 

I modified the fz.switch_action, I took a guess based on the data in this thread but I did not test it, as no longer have any of the Niko switches.

```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const local = {
    fz: {
        switch_operation_mode: {
            cluster: 'manuSpecificNiko1',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const state = {};
                if (msg.data.hasOwnProperty('switchOperationMode')) {
                    const operationModeProperty = `operation_mode${meta.endpoint_name ? `_${meta.endpoint_name}` : ''}`;
                    const operationModeMap = {0x02: 'control_relay', 0x01: 'decoupled', 0x00: 'unknown'};
                    state[operationModeProperty] = operationModeMap[msg.data.switchOperationMode];
                }
                return state;
            },
        },
        switch_action: {
            cluster: 'manuSpecificNiko2',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const state = {};

                if (msg.data.hasOwnProperty('switchAction')) {
                    // NOTE: a single press = two seperate values reported, 16 followed by 64
                    //       a hold/release cyle = three seperate values, 16, 32, and 48
                    const actionMap = (mode.model == '552-721X1') ?
                                {16: null, 64: 'single', 32: 'hold', 48: 'release'} :
                                {16: null, 64: 'left_single', 32: 'left_hold', 48: 'left_release', 4096: 'right_single', 8192: 'right_hold', 12288: 'right_release'};
                    state['action'] = actionMap[msg.data.switchAction];
                }
                return state;
            },
        },
    },
    tz: {
        switch_operation_mode: {
            key: ['operation_mode'],
            convertSet: async (entity, key, value, meta) => {
                // WARN: while we can technically write 0x00 to the operationMode attribute
                //       this seems to brick the device and it will need to be rejoined
                const operationModeLookup = {control_relay: 0x02, decoupled: 0x01};
                if (!operationModeLookup.hasOwnProperty(value)) {
                    throw new Error(`operation_mode was called with an invalid value (${value})`);
                } else {
                    await entity.write('manuSpecificNiko1', {'switchOperationMode': operationModeLookup[value]});
                    return {state: {operation_mode: value.toLowerCase()}};
                }
            },
            convertGet: async (entity, key, meta) => {
                await entity.read('manuSpecificNiko1', ['switchOperationMode']);
            },
        },
    },
};

const definition = {
        zigbeeModel: ['Double connectable switch,10A'],
        model: '552-721X2',
        vendor: 'Niko',
        description: 'Double connectable switch',
        fromZigbee: [fz.on_off, local.fz.switch_operation_mode, local.fz.switch_action],
        toZigbee: [tz.on_off, local.tz.switch_operation_mode],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep1 = device.getEndpoint(1);
            const ep2 = device.getEndpoint(2);
            await reporting.bind(ep1, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(ep2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(ep1);
            await reporting.onOff(ep2);
            await ep1.read('manuSpecificNiko1', ['switchOperationMode']);
            await ep2.read('manuSpecificNiko1', ['switchOperationMode']);
        };
        exposes: [
            e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
            e.action(['single', 'hold', 'release']).withEndpoint('l1'),
            e.action(['single', 'hold', 'release']).withEndpoint('l2'),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withEndpoint('l1'),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withEndpoint('l2'),
        ],
};
module.exports = definition;

```

bartpijnaert, 2022-12-26:
helaas heb ook ik hetzelfde probleem.

Unfortunately, I too have the same problem.

Homergy, 2022-12-31:
Same issue here.
@sjorge I've tested with you updated external converter however the behavior is exactly the same with or without the converter.
added the following to configuration in zigbee2mqtt configuration.yaml:
"
advanced:
  log_level: debug
external_converters:
  - nikodebug.js
"
and added the nikodebug.js in the same folder with your code pasted.

Behavior is the following:
Everything seems to work in relay mode. (however I don't see any actions)
Logs from both buttons in relay mode:
[image]

In decoupled mode I don't recieve any info in the logs. (I've bought 9 different double folds tested on 5 of the so it's not a defective unit)
Logs from first setting it to decoupled and then controlling the relays. as I said pressing the bottons both long and short doesn't provide any logs at all.
[image]

I'm using the standrad zigbee build (not the dev)

Really hope someone could solve this, I tested on single fold and then bought a bunch of double folds...
I could do it via Niko smarthub but would prefer to use it directly.

Herwigvd, 2022-12-31:
@sjorge
I tested your code in decoupled mode
I just pressed the right button (l1?) and got two error messages.
I did no further tests.
Below the messages I got in the log

By the way, I had to correct line 83 for syntax (was semi colon).
I am not a js-programmer, so I hope I did the right thing

           await ep2.read('manuSpecificNiko1', ['switchOperationMode']);
        **},**
        exposes: [

Kind regards,
Herwig

debug 2022-12-31 09:54:29: Received Zigbee message from 'SwitchSalon', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":4096}' from endpoint 1 with groupID 0
error 2022-12-31 09:54:29: Exception while calling fromZigbee converter: mode is not defined}
debug 2022-12-31 09:54:29: ReferenceError: mode is not defined
    at Object.convert (/opt/zigbee2mqtt/dist/util/externally-loaded.js:33:39)
    at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitDeviceMessage (/opt/zigbee2mqtt/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
    at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:495:14)
    at Controller.onZclOrRawData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:706:18)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:143:70)
    at ZStackAdapter.emit (node:events:527:28)
debug 2022-12-31 09:54:29: Received Zigbee message from 'SwitchSalon', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16384}' from endpoint 1 with groupID 0
error 2022-12-31 09:54:29: Exception while calling fromZigbee converter: mode is not defined}
debug 2022-12-31 09:54:29: ReferenceError: mode is not defined
    at Object.convert (/opt/zigbee2mqtt/dist/util/externally-loaded.js:33:39)
    at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitDeviceMessage (/opt/zigbee2mqtt/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
    at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:495:14)
    at Controller.onZclOrRawData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:706:18)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:143:70)
    at ZStackAdapter.emit (node:events:527:28)

sjorge, 2022-12-31:
@Homergy you will not get an action in when coupled to the relays, it will just toggle the state_l1 and state_l2.

However decoupled mode should show actions, but it looks like on the double operation_mode is somehow linked for both so setting either as decoupled will set both. (Not sure if it's a device limitation or not).

@Herwigvd I probably still have an error somewhere then. I'll try and give the code pass when I have more time.

Homergy, 2023-01-07:
@sjorge absolutely don't want to push anything. Just wanted to let you know that we probably live close to each other and I can supply you with the 552-72102 double fold. That might be easier to work with?

I've send you an e-mail with more details.

Love your great work ;)

sjorge, 2023-01-07:
@Homergy yes I got your e-mail but outlook.com server were rejecting the netblock where my mail server is located in. (Pending ticket open with my hosting provider -_-)

I'll reply here then:

> Thanks for the offer of lending some hardware but I'm not super comfortable with the idea. I think this should be fixable with the information provided in the github issue.
It looks like there are 2 main issues:
> - `operation_mode` seems to be linked if you change it for 1 endpoint both change, don't think we can do much here except cleanup the converters
> - the value for action are different for the single vs dual button one, I think this is fixable with the info here.

The switch does not send the action message when it's set to 'control_relay', so unless you are in 'decoupled' mode you won't see actions as you mentioned above.

Can you try the converter again but with the operation_mode set to decoupled? I also fixed the syntax bug @Herwigvd  found, but did not manage to find out where the `mode` undefined error is coming from. I'm not getting it when I use this external converter with my single version.

Homergy, 2023-01-07:
@sjorge no problem I fully understand.
Off course all was without any strings attached what-so-ever. I fully respect it is your free time and would never dare to demand anything or stalk you. Your a Hero for helping people out in your free time ;)

Anyway I tested the new external convertor and I now do see that the decoupled is moved to the top and is a general setting. (no longer per endpoint)
However in my case after decoupling I still don't see any actions coming in when decoupled. Logs just stay empty :(

PS: Is there any way I can support/sponser you? Again no strings attached. I've been looking both on your site and github but can't seem to find anything?

sjorge, 2023-01-08:
Do you have a debug log (just log level set to debug for now) when you press a button in decoupled mode?
I'd expect to at least see entries like:

```
Debug 2022-10-31 07:16:34 Received Zigbee message from 'Living Room / Kitchen Lights', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16}' from endpoint 1 with groupID 0
```

Homergy, 2023-01-08:
I have set my log level to debug and that works fine for other messages or devices however all my double folds don't seem to send any actions.
I have linked some of them to Niko hub afterwards and there they work fine in "decoupled mode".

I did also notice that in decoupled mode the led still follows the relay output. In the decoupled mode of single fold it does not do that.

I have no idea why I have no actions whatsoever even in debug logging. Bit worried tbh. Could it be that the decoupled command somehow does not fully work? My buttons are brand new.

sjorge, 2023-01-08:
Indeed the led still following the relay instead of always being of would indeed indicate they are not in decoupled mode.

Can you do a read in the frontend like this:
[image]

For each of the endpoints on the double?

sjorge, 2023-01-08:
I would expect it to be 1 in decoupled or 2 in control relay

Homergy, 2023-01-08:
I ran it on all endpoints:
[image]
All returned 1. Except the 242 where I got the feedback cannot read properties but if I'm not mistaken that's the link quality so to be expected?

sjorge, 2023-01-12:
Hmmm very weird so it is set correctly according to the device. What is the firmware version these are running on?

Homergy, 2023-01-12:
I just did the check, these buttons were launched by Niko 03/2022.
Firmware is: 20211012
So this means still the first firmware officially released.

It is indeed very strange and I'm not sure how we could solve this. Maybe I'll try with the ubiqua scanner this week to see if it even sends anything.

sjorge, 2023-01-12:
That indeed looks like the original firmware ... I guess it could be possible there are other values than 1 and 2 for switchOperationMode on the dual variant 🤔 

You could try other values using the frontend by setting the int and hitting write instead of read.

Homergy, 2023-01-12:
I've tried 0 up to 128. (Don't think it's really useful to test values even higher)
0 throws a fatal error removing it from the zigbee network, reboot and repair solves this.
It does accept 1 and 2, all the other value's return the error:
"
Publish 'set' 'write' to 'Niko double fold' failed: 'Error: Write 0x6c5cb1fffe1096bc/2 manuSpecificNiko1({"switchOperationMode":122}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4703,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'
"
Changing it for one endpoint changes it for all of them.
I'm wondering why it does work for the other users like Herwigvd.

I sniffed using ubiqua, it is really the button which is sending nothing in the zigbee network.
I even noticed that I soft bricked the double fold after sending 0 to the SwitchOperationMode. Even in relay mode it stopped sending actions... Remove, factory reset and repair solved it however. I was hoping the decoupled mode would work then too but no luck.

sjorge, 2023-01-12:
Yeah i’ve soft bricked them a ton while adding support.

Superweird it’s not sending anything in decoupled mode, i got the extended values to take a stap at a new map from earlier in the issues so for some people it is sending stuff :/

Thanks for confirming it is linked though, so af least the one operation_mode attribute of the new convertor is correct.

aocaoc, 2023-01-13:
Not sure I can be of much assistance but my Niko switches return:
Firmware build date 20210224
Firmware version 313-688-00

Homergy, 2023-01-13:
@aocaoc do you recieve the actions in debug logging when in decoupled mode?

sjorge, 2023-01-13:
> @aocaoc do you recieve the actions in debug logging when in decoupled mode? I have the exact same firmware. But I believe there is only 1 firmware.

Preferably with the external converter from https://github.com/Koenkk/zigbee2mqtt/issues/13737#issuecomment-1362030574 in place.

aocaoc, 2023-01-16:
> @aocaoc do you recieve the actions in debug logging when in decoupled mode? I have the exact same firmware. But I believe there is only 1 firmware.

Unfortunately not.

sdp74, 2023-01-16:
Zigbee2MQTT 1.29.2-1

no external converter

NIKO 
firmware build date 20210224
Firmware version 313-688-00

decoupled mode, Log level debug show all

#### single press button 1 (left) ####

debug 2023-01-16 11:35:32Received Zigbee message from 'NIKO 552-721X2', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16}' from endpoint 1 with groupID 0
info 2023-01-16 11:35:32MQTT publish: topic 'zigbee2mqtt/NIKO 552-721X2', payload '{"action":null,"action_l1":null,"action_l2":null,"linkquality":68,"operation_mode":"decoupled","operation_mode_l1":"decoupled","operation_mode_l1_l1":"decoupled","operation_mode_l2":"decoupled","operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
debug 2023-01-16 11:35:33Received Zigbee message from 'NIKO 552-721X2', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":64}' from endpoint 1 with groupID 0
info 2023-01-16 11:35:33MQTT publish: topic 'zigbee2mqtt/NIKO 552-721X2', payload '{"action":"single","action_l1":null,"action_l2":null,"linkquality":65,"operation_mode":"decoupled","operation_mode_l1":"decoupled","operation_mode_l1_l1":"decoupled","operation_mode_l2":"decoupled","operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-16 11:35:33MQTT publish: topic 'zigbee2mqtt/NIKO 552-721X2', payload '{"action":"","action_l1":null,"action_l2":null,"linkquality":65,"operation_mode":"decoupled","operation_mode_l1":"decoupled","operation_mode_l1_l1":"decoupled","operation_mode_l2":"decoupled","operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
info 2023-01-16 11:35:33MQTT publish: topic 'zigbee2mqtt/NIKO 552-721X2/action', payload 'single'

#### single press button 2 (right) ####

debug 2023-01-16 11:36:06Received Zigbee message from 'NIKO 552-721X2', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":4096}' from endpoint 1 with groupID 0
info 2023-01-16 11:36:06MQTT publish: topic 'zigbee2mqtt/NIKO 552-721X2', payload '{"action_l1":null,"action_l2":null,"linkquality":65,"operation_mode":"decoupled","operation_mode_l1":"decoupled","operation_mode_l1_l1":"decoupled","operation_mode_l2":"decoupled","operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
debug 2023-01-16 11:36:06Received Zigbee message from 'NIKO 552-721X2', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16384}' from endpoint 1 with groupID 0
info 2023-01-16 11:36:06MQTT publish: topic 'zigbee2mqtt/NIKO 552-721X2', payload '{"action_l1":null,"action_l2":null,"linkquality":70,"operation_mode":"decoupled","operation_mode_l1":"decoupled","operation_mode_l1_l1":"decoupled","operation_mode_l2":"decoupled","operation_mode_l2_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'

yawaweb, 2023-01-25:
I had the same problem with:

- Niko 552-721X2
- Firmware version: 313-688-00
- Firmware build date: 20210224

and also 

- Niko 552-721X1
- Firmware version: 313-687-00
- Firmware build date: 20210224

aocaoc, 2023-01-31:
Will this be implemented in a z2m update or does it need confirming that it works? 
I don't have much knownledge regarding z2m besides the basics but I'd like to see this work.
I'm running HA supervised with z2m as add-on so don't really know what I can contribute but I'm willing to help.

svenjochems, 2023-04-17:
@sjorge 
I tested your proposal, replaced following code in niko.js
```
const actionMap = {16: null, 64: 'single', 32: 'hold', 48: 'release'};
```
with 
```
const actionMap = (model.model == '552-721X1') ?
       {16: null, 64: 'single', 32: 'hold', 48: 'release'} :
       {16: null, 64: 'left_single', 32: 'left_hold', 48: 'left_release', 4096: 'right_single', 8192: 'right_hold', 12288: 'right_release'};
```

Now I get following data published:
```
zigbee2mqtt/myswitch/action left_single
zigbee2mqtt/myswitch/action left_hold
zigbee2mqtt/myswitch/action left_release
zigbee2mqtt/myswitch/action right_single
zigbee2mqtt/myswitch/action right_hold
zigbee2mqtt/myswitch/action right_release
```

Homeassistant however does not receive the actions as it is expecting `action_l1` and `action_l2`..

sjorge, 2023-04-17:
@svenjochems Excellent! I'll clean this up a bit and put that into a PR (that will also fix the homeassistant issue)

svenjochems, 2023-04-17:
@sjorge thanks!
Just one small remark. My Aqara switch exposes actions like `single_left` and your proposed solution was `left_single`. Maybe it's best to keep action naming consistent. 😉

sjorge, 2023-04-17:
> @sjorge thanks! Just one small remark. My Aqara switch exposes actions like `single_left` and your proposed solution was `left_single`. Maybe it's best to keep action naming consistent. 😉

I based those on https://github.com/Koenkk/zigbee-herdsman-converters/blob/57dc2ef63138e0208ec43264720df9fa3c93788e/devices/xiaomi.js#L1480-L1484

Homergy, 2023-04-17:
Great to see some action again, really eager to test it. However I'm unsure how to.
In your external converter from January is seems that this adaption was already made then?
Is the dev build from Z2M the way to go? I'm sorry for the noob question.

sjorge, 2023-04-17:
@Homergy it’s not very different at all but in an other issue by @svenjochems was seeing the same data, they were able to verify that the change worked for them.

As of 2h ago dev now has a slightly cleaned up version with fixed expose data.

To be on the safe side you should probably remove the device, restart z2m, add the device after upgrading so all the cached state is gone.

sjorge, 2023-04-18:
@Koenkk how often does the edge image get rebuild?

Koenkk, 2023-04-18:
edge gets rebuild after every commit on zigbee2mqtt/dev

svenjochems, 2023-04-18:
@sjorge @Homergy I also pulled the latest dev and tested the changes.
- action works as expected. All the different actions are properly exposed and received by homeassistant
- switch_led_state and switch_led_enable indeed seem to control both left and right simultaneously like @Homergy observed
- operation mode is now only possible for both endpoints simultaneously and not per endpoint like before?

sjorge, 2023-04-18:
@svenjochems operation_mode used to change for both when you set either of them… i guess that’s true for switch_led_* too… that would make sense.

i’ll do a pr for that tomorrow, if i dont forget

Homergy, 2023-04-18:
@sjorge in NHC it is possible to enable the led for only one of the 2 leds. Shouldn't we keep this functionality if it's working?

@svenjochems good to know it is working for some but I can't explain why it doesn't work me. Could you check the production date of yours? Mine are all from w45 of 2022
[image]

svenjochems, 2023-04-18:
> operation_mode used to change for both when you set either of them

@sjorge I tested the previous build and this seems correct.
@Homergy do you know if the operation mode can be changed for left and right separately in NHC? I don't have this so I cannot test this myself.

[image]
[image]

sjorge, 2023-04-18:
@Homergy what is NHC? The Niko gateway?

If the led stuff is working for some people we should def keep it, but if it always toggles both like operation_mode we should change it to match.

I suspect there, that like with the action there might he other values that we don't know about to set both, left only, right only.

if NHC is indeed the Niko gateway, if we can somehow sniff the traffic that could help us figure it out. But we’d need to know the key Niko uses, I don't think we do.

sjorge, 2023-04-19:
@Homergy does this ubiqua tool support capering to pcapng ?

I usually (when the tc is know) capture with wireshark and have been able to manually get the info I needed from them in the past.

svenjochems, 2023-04-24:
@sjorge Upon further testing I encounted a few more issues

1. In decoupled mode, when I long press the right switch, I receive a `single_right`, `hold_right` and a `release_right` action as I only expect the last 2.
2. I have a dumb toggle switch connected to the external input of the switch. When using the switch in decoupled mode and pressing the external input, no actions are forwarded.

The current action map is defined as:

```
                    const actionMap = (model.model == '552-721X1') ? {
                        16: null,
                        64: 'single',
                        32: 'hold',
                        48: 'release',
                    } : {
                        16: null,
                        64: 'single_left',
                        32: 'hold_left',
                        48: 'release_left',
                        4096: 'single_right',
                        8192: 'hold_right',
                        12288: 'release_right',
                    };
```
I extracted the following actions from the debug log:
|       | left | right |
| ---- | ---- | ------ |
| null | 16 | 4096 |
| single | 64 | 46384 |
| hold | 32 | 8192 |
| release | 48 | 12288 |
| ext null | _256_ | 65536 |
| ext single | _1024_ | 262144 |
| ext hold | _512_ | 131072 |
| ext release | _768_ | 196608 |

I did not test the ext for the left endpoint as I have not wired this in my setup currently but this table totally makes sense to me because the `right = left * 256` and `ext = normal * 16`.

sephiroth1395, 2023-05-02:
I just tested again on Z2M 1.30.4 and **everything works as expected**.  My plan was to use both relays in decoupled mode so I'm happy.  I guess we can close this one?

[69 more replies]