Koenkk/zigbee2mqtt #13738: no network route to a connected device

icherfas, 2022-08-28:
### What happened?

trying to change a state of a switch return 'no network route' while link quality in the map is '27' to another router

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

Z-Stack_Home_1.2 20211115/20211116

### Adapter

Texas Instruments CC2531

### Debug log

`Publish 'set' 'state' to 'Living room' failed: 'Error: Command 0xa4c138f2686805cd/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

aoberoi, 2022-08-28:
Sounds similar to #13547

Koenkk, 2022-08-29:
@icherfas how many devices do you have in your network?

icherfas, 2022-08-30:
> @icherfas how many devices do you have in your network?

Hi,
I have 7 devices.
Out of that 3 are currently offline (not in use)

Koenkk, 2022-08-30:
> Out of that 3 are currently offline (not in use)

Which devices are these?

icherfas, 2022-08-30:
Two of them are smart plugs: 
Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_ps3dmato

The other one is a 2 gang switch:
Zigbee Model
TS0002
Zigbee Manufacturer
_TZ3000_bvrlqyj7

Koenkk, 2022-08-30:
Interesting, does repairing it solve it? If not I would need a sniff of when the command fails: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

wrichter, 2022-09-10:
sounds similar to #12774

JacekKac, 2022-10-09:
i have also this plug... they are imho poor as a routers:
1/ First plug was in very good range in with coordinator and router, but was disconnecting from the network - don't know the reason. 

2/ Second plug is not accepting any end devices or end devices are not choosing this router - and they should (devices very close to plug). 

both plugs are close to other routers/coordinators - showing nice LQI but sending commands ON/OFF often fails - don't know why. I use them with sonoof temp sensors and sonoff repeater wall outlets - which are working much better.

closed 2022-11-17 by github-actions[bot]