Koenkk/zigbee2mqtt #13741: Custom converter values listed as diagnostic and not shown

jsbrich, 2022-08-28:
### What happened?

Creating a sensor using the E18-MS1PA2 device (cc2530, cc2592) with an MQ-5 gas sensor attached.  I was able to write the external converter and am seeing the MQTT messages being sent

But in Home Assistant, the link quality and voltage values are listed as Diagnostic and not showing any values.  When I add the device to the network, I see this in the debug
````
debug 2022-08-28 18:23:28: Received MQTT message on 'homeassistant/sensor/0x00124b000710b5f1/voltage/config' with data '{"availability":[{"topic":"z
igbee2mqtt/bridge/state"}],"device":{"configuration_url":"http://zigbee2mqtt.jnetinc.local/#/device/0x00124b000710b5f1/info","identifiers":["zigbee2
mqtt_0x00124b000710b5f1"],"manufacturer":"Custom devices (DiY)","model":"JENTINC Gas Sensor (CC2530.ROUTER)","name":"Gas_Sensor_1","sw_version":"202
1-07-05"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Gas_Sensor_1","nam
e":"Gas_Sensor_1_voltage","state_class":"measurement","state_topic":"zigbee2mqtt/Gas_Sensor_1","unique_id":"0x00124b000710b5f1_voltage_zigbee2mqtt",
"unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
````
I get a similar message for the linkquality.

Here is the external converter:
````
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');

const exposes = zigbeeHerdsmanConverters.exposes;
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const {repInterval} = require('zigbee-herdsman-converters/lib/constants');

const fzLocal = {
    diy_gas_adc: {
        cluster: 'genAnalogInput',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            return {
                voltage: msg.data.presentValue
            };
        },
    }
};

const reportingLocal = {
    diy_gas_adc: async (endpoint, overrides) => {
        const p = reporting.payload('presentValue', 10, repInterval.HOUR, 0, overrides);
        await endpoint.configureReporting('genAnalogInput', p);
    }  
};

const device = {
    zigbeeModel: ['lumi.router'],
    model: 'CC2530.ROUTER',    toZigbee: [],
    exposes: [e.voltage(),e.linkquality()],
    configure: async(device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genAnalogInput']);
        await reportingLocal.diy_gas_adc(device.getEndpoint(1));
    },
    meta: {
        multiEndpoint: true,

    },
};

module.exports = device;
````

### What did you expect to happen?

I expected that the voltage reading would show as a Sensor in Home Assistant and that the values would be reported

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20211217

### Adapter

ZIGBEE 3.0 USB DONGLE PLUS

Koenkk, 2022-08-29:
What shows up in the z2m frontend -> exposes tab? That these sensor are listed under diagnostics is expected behaviour.

jsbrich, 2022-08-29:
I see both fields with values in the exposes tab
[image]

What would I need to do differently to have the sensor voltage reading show as a Sensor?  I understand voltage is often for the battery attached to the device, but in this instance it is an ADC reading which is the primary information.

Koenkk, 2022-08-30:
These sensors are disabled by default in HA (therefore they appear greyed out). Click on them and enable them first.

jsbrich, 2022-08-30:
That worked! Thanks!
Is there any way to have a voltage listed as a Sensor?

Also, the MQ-5 has a digital output that goes high when the level reaches a set point.  What would I add to the converter to get that true/false value?

Koenkk, 2022-08-31:
> Is there any way to have a voltage listed as a Sensor?

You have to override the HA MQTT discovery in the z2m `configuration.yaml`, e.g. 

```
devices:
  "0x12345678":
    friendly_name: my_switch
    homeassistant:
      voltage:
        entity_category: null
```

> Also, the MQ-5 has a digital output that goes high when the level reaches a set point. What would I add to the converter to get that true/false value?

If your sensor sends a message when this happens yes. You can see all message by enabling the z2m debug logging, then you can create a converter like explained in https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

I will close this since this is not an issue in z2m.

closed 2022-08-31 by Koenkk