Koenkk/zigbee2mqtt #13745: Friends of hue switch (Niko, Senic)

stefanuytterhoeven, 2022-08-29:
### What happened?

https://www.niko.eu/nl-be/artikel/120-91004
https://www.senic.com/products/friends-of-hue-smart-switch

This switch should support 'dimming'. Though, when pressing a button, the only  events I get is 'press_3' and when I release the button, a 'release_3' event. So starting to press the button, keep pressing on it, and release, only generates those two events.
According to the support pages of Friends of Hue switchs (like the niko one and the senic one) should support dimming.
Any news on this? Can this be added?

### What did you expect to happen?

Dimming events. Is supported by the device itself, but not by Zigbee2mqtt

### How to reproduce it (minimal and precise)

Press and hold a button

### Zigbee2MQTT version

 1.27.0 

### Adapter firmware version

 20220219

### Adapter

Sonoff zigbee 3.0 usb dongle plus

sjorge, 2022-08-29:
I don't think this is supported by the device as this is a greenpower device.
This is probably handled by the hue bridge, my guess it it triggered a `genLevelCtrl.moveWithOnOff `on press_X and a `genLevelCtrl.stop` on the corresponding release_X.

This is probably something that can be handled by hass or node-red

Danit2, 2022-08-31:
This os not a function of the switch. This is done in the gateway from hue. 

Look at this:
https://github.com/Koenkk/zigbee2mqtt/issues/13361#issuecomment-1202159610

I make it like this. 

Hope it helps.

closed 2022-10-09 by github-actions[bot]