Koenkk/zigbee2mqtt #13750: [Feature request]: store groups in separate config (or devices.yaml)

jficz, 2022-08-29:
### Is your feature request related to a problem? Please describe

On NixOS, as described in #2071, the configuration file is generated by the system and is supposed to be read-only. This is a common best practice for managed systems as configuration and state should be separate. Zigbee2mqtt does not follow this paradigm storing both config and (some of) state in `config.yaml`.

Groups break on NixOS easily as the `config.yaml` file is rewritten to the known (i.e. _defined_) state on each service start.

This leads to serious problems with group as zigbee2mqtt is aware of them (due to them being in `database.db` and devices announcing them) but missing them from `config.yaml` - like losing group names, devices being removed from groups and losing the ability to manage groups ("group cannot be deleted because it doesn't exist" kind of problems).

### Describe the solution you'd like

Clearly separate configuration and state store. Groups are (imho) part of the system state, not configuration, and as such should not be written to `config.yaml` but rather kept in a separate file (like `devices.yaml` does with devices).

### Describe alternatives you've considered

Static group configuration in `configuration.nix` - not feasible as groups are part of the device states and configuring groups through configuration.nix would require multiple-step approach (first config the service, then scan for devices, then configure the devices,...) and each group reconfiguration would require service restart.

### Additional context

In my opinion (as a experienced systems and automation engineer) configuration should basically be immutable and anything requiring runtime changes should be considered a state and treated as such (i.e. separate from immutable config).

jficz, 2022-08-29:
Or, even better, the group config should not exist at all and should be derived from the actual infrastructure state (i.e. from `database.db`) which is partially happening already anyway. Only the group id<->name pairing will probably need to be stored somewhere.

Koenkk, 2022-08-29:
> (like devices.yaml does with devices).

Just like devices, the same can be done for groups see: https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#extract-config-to-separate-files