Koenkk/zigbee2mqtt #13755: [New device support]: Aubess ZigBee Socket Plug EU 16A Adapter Power Monitor Smart Socket APP Remote Control Tuya Smart For Alexa Google Home Assistant

sdalee, 2022-08-29:
### Link

https://www.aliexpress.com/item/1005003698857421.html?spm=a2g0o.order_list.0.0.21ef18025roUSg

### Database entry

{"id":104,"type":"Router","ieeeAddr":"0xa4c138f4c8ecdfcb","nwkAddr":36112,"manufId":4417,"manufName":"_TZ3000_hdopuwv6","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"seMetering":{"attributes":{"currentSummDelivered":[0,10]}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":228,"rmsCurrent":117,"activePower":30}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2972188,"53251":122,"53252":10338,"53253":1211,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00216004eb","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00216004eb","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00216004eb","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00216004eb","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":76,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1661795325861,"defaultSendRequestWhen":"immediate"}

### Comments

Unfortunately, I don't know how can I do Tuya support in Home Assistant.

Koenkk, 2022-08-30:
Added! Device will be supported in the next release on 1 September.

closed 2022-08-30 by Koenkk

chartinger, 2022-09-01:
@Koenkk  I bought almost the same plug (20A Version)
https://www.aliexpress.com/item/1005003970217316.html
and it identified itself as `_TZ3000_gvn91tmx`

I did not see it in the changed file, so just wanted to note this.

Koenkk, 2022-09-05:
@chartinger  added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

sdalee, 2022-09-14:
Thank you for Koenkk this support. 
I used it for a few days. I created a logging script that logs instantaneous performance. 
Every few minutes (5-25 minutes, 5 minutes on average), the measurement drops out for 5 seconds:

09/14/2022   21:42:38:  58 W
09/14/2022   21:42:39:  34 W
09/14/2022   21:42:41:  34 W
09/14/2022   21:43:18:  34 W
09/14/2022   21:43:19:  0 W
09/14/2022   21:43:19:  0 W
09/14/2022   21:43:23:  0 W
09/14/2022   21:43:24:  37 W
09/14/2022   21:43:24:  37 W
09/14/2022   21:43:28:  37 W
09/14/2022   21:43:33:  37 W
09/14/2022   21:44:25:  37 W
....
09/14/2022   21:47:38:  39 W
09/14/2022   21:47:39:  39 W
09/14/2022   21:47:48:  39 W
09/14/2022   21:47:49:  0 W
09/14/2022   21:47:49:  0 W
09/14/2022   21:47:53:  0 W
09/14/2022   21:47:54:  34 W
09/14/2022   21:47:54:  34 W
09/14/2022   21:48:24:  34 W

What could be the problem?

Koenkk, 2022-09-15:
This is a bug in the device, there is a long discussion about issues with these devices here: https://github.com/Koenkk/zigbee2mqtt/discussions/13343

pyloutom, 2022-09-23:
Bonjour, j'ai acherter la même prise en version 20A
_TZ3000_gvn91tmx.

Je ne sais pas comment changer les paramêtre plus haut dans HA. Pourriez vous m'aider svp.